prusa3d/PrusaSlicer #139: [Request] Add option to specify layer step in layer height limits

lonelymyp, 2017-02-17:
### Version
1.3.4
I propose to add a new parameter to the layer height limits.

In setting the minimum and maximum height of the layer must be added the setting "step."
eg
minimum 0.2
maximum 0.5
step 0.1

bubnikv, 2017-02-17:
@lonelymyp Why do you need that?

lonelymyp, 2017-02-17:
For me 0.2 acceptable quality.
but if the minimum height is 0.2 and maximum of 0.4, then it turns out not too many options.
if I understand correctly, step height = minimum height, so if I put a range of 0.2-0.4, really I do not get the intermediate numbers, only 0.2 and 0.4.
It would be great to put a step 0.05, and get intermediate height for a smooth transition.

bubnikv, 2017-02-17:
> if I understand correctly, step height = minimum height, so if I put a range of 0.2-0.4, really I do not get the intermediate numbers, only 0.2 and 0.4.

I don't understand. The current Slic3r changes the layer height continuously. This works well with the Prusa i3 MK2 with 200 steps per mm and 16 micro steps on the Z lead screws.

I would expect you to ask for fixing the layer steps to full steps of the Z stepper motor.

lonelymyp, 2017-02-17:
are two reasons =)
1. Sync with the parameters of the screw / belt
2. The use of a smooth transition between the thick layers

bubnikv, 2017-02-17:
> Sync with the parameters of the screw / belt

You mean the full steps of the stepper motors?

> The use of a smooth transition between the thick layers

I don't understand. Would you please be more specific?

lonelymyp, 2017-02-17:
Just try to put a range of 0.2 and 0.4
and make a smooth transition to the model =)
model will slice just two heights, either 0.2 or 0.4
Z14.000
Z14.400
Z14.800
Z15.000
Z15.200
Z15.400

If it were possible to set the step = 0.05, then:
Z14.000
Z14.400
Z14.800   and here begins a smooth decrease height 
Z15.150
Z15.450

bubnikv, 2017-02-17:
> Just try to put a range of 0.2 and 0.4 and make a smooth transition to the model =)
> model will slice just two heights, either 0.2 or 0.4

There must be some misunderstanding. I am certainly getting layers with smoothly varying layer thickness. I don't know what you are doing and I have no idea how to help you.

lonelymyp, 2017-02-17:
> varying layer thickness.

which specific thickness?
where does this magic number which height is reduced?

0.4 layer height:
[image]
your printer can produce layers with accuracy of 0.001 mm?? it's impossible

I propose to give the opportunity to choose the minimum height to which the next layer will decrease

josefprusa, 2017-02-17:
It doesn't matter if your Z axis has that resolution. Firmware will round it to the steps. At the end of the day this will ensure that the gradient will be as smooth as possible on every hardware. Try our implementation and yours with filling in the layer thicknesses manually and you will see better results with ours. We tried :-)

closed 2017-02-24 by lonelymyp