prusa3d/PrusaSlicer #140: Can't run appImage on Ubuntu 16.04.2 LTS without Try::Tiny module

burbilog, 2017-02-17:
### Version
Slic3r 1.33.5 Prusa

### Operating system type + version
Ubuntu 16.04.2 LTS

### Behavior
Download Slic3r-1.33.5-prusa3d-linux64-full-201702171509.AppImage and try to run

Expected result: running Slic3r

Actual result: Slic3r fails with following message:

~~~
Can't locate Try/Tiny.pm in @INC (you may need to install the Try::Tiny module) (@INC contains: /tmp/.mount_I7GhlJ/usr/bin/bin/lib /home/rm/perl5/lib/perl5 /tmp/.mount_I7GhlJ/usr/bin/lib/site_perl/5.22.0/x86_64-linux-thread-multi /tmp/.mount_I7GhlJ/usr/bin/lib/site_perl/5.22.0 /tmp/.mount_I7GhlJ/usr/bin/lib/5.22.0/x86_64-linux-thread-multi /tmp/.mount_I7GhlJ/usr/bin/lib/5.22.0 .) at /home/rm/perl5/lib/perl5/LWP/Protocol.pm line 11.
BEGIN failed--compilation aborted at /home/rm/perl5/lib/perl5/LWP/Protocol.pm line 11.
Compilation failed in require at /home/rm/perl5/lib/perl5/LWP/UserAgent.pm line 13.
BEGIN failed--compilation aborted at /home/rm/perl5/lib/perl5/LWP/UserAgent.pm line 13.
Compilation failed in require at /tmp/.mount_I7GhlJ/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI/Plater.pm line 11.
BEGIN failed--compilation aborted at /tmp/.mount_I7GhlJ/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI/Plater.pm line 11.
Compilation failed in require at /tmp/.mount_I7GhlJ/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI.pm line 20.
BEGIN failed--compilation aborted at /tmp/.mount_I7GhlJ/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI.pm line 20.
Compilation failed in require at (eval 90) line 1.
~~~

cpanm install Try::Tiny  fixes the problem, but hey, it's appImage, eh?

#### STL/Config (.ZIP) where problem occurs
Config does not matter I suppose...

bubnikv, 2017-02-18:
It looks like there is another problem going on: The appimage shall not pull the system perl modules. I think what you are seeing is, the system perl modules are pulled before the ones from the AppImage, and seemingly your Perl modules have different dependencies.

bubnikv, 2017-02-18:
Looking inside the AppRun script generated by the AppImage suite, it starts an application defined in Slic3r.desktop. It looks like the AppRun is flawed, as it first sets up the paths to point inside the AppImage, but then it starts the system installed Slic3r script.

I am not sure where shall I fix it. @probonopd any idea?

probonopd, 2017-02-18:
On Ubuntu 16.04 without a system-installed Slic3r, the AppImage _launches_ with the following output on the console:

```
me@host:~$ /home/me/Downloads/Slic3r-1.33.5-prusa3d-linux64-full-201702171509.AppImage 
Use of uninitialized value in subroutine entry at /tmp/.mount_isAMZR/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI/Plater.pm line 170.
```

So I suspect as soon as there is also a Slic3r installed in the system, it somehow picks up the system one/mixes up the two. More investigation needed.

bubnikv, 2017-02-18:
@probonopd 

Thanks. The issue is in how the AppRun pulls the name of the executable from the *.desktop file.
It simply gets the file name and it starts it, without prefixing it with the "local" path.

probonopd, 2017-02-18:
I do not get the "Use of uninitialized value in subroutine entry" with the AppImage I had generated earlier, https://bintray.com/probono/AppImages/download_file?file_path=Slic3r-1.31.6.prusa3d.glibc2.14-x86_64.AppImage. Can you confirm that the issue reported in this ticket is also present in that version?

probonopd, 2017-02-18:
> Thanks. The issue is in how the AppRun pulls the name of the executable from the *.desktop file.
It simply gets the file name and it starts it, without prefixing it with the "local" path.

AppRun exports `$PATH` to include `usr/bin` in the AppImage. 

If you look at the AppImage, it has `Exec=slic3r.wrapper --gui` in the desktop file, which launches `usr/bin/slic3r.wrapper` inside the AppImage.

probonopd, 2017-02-18:
@bubnikv is the script that you are using to produce the AppImage available on GitHub?

bubnikv, 2017-02-18:
@probonopd 

> AppRun exports $PATH to include usr/bin in the AppImage.
> If you look at the AppImage, it has Exec=slic3r.wrapper --gui in the desktop file, which launches usr/bin/slic3r.wrapper inside the AppImage.

You are right.

What does the 
HERE="$(dirname "$(readlink -f "${0}")")")
do?
Namely, why is readling -f necessary? What is special about ${0}, that readlink -f is needed?

slic3r is a bash script, containing the following three lines:
DIR=$(dirname $0)
export LD_LIBRARY_PATH="$DIR/bin"
exec "$DIR/bin/perl5.22.0" "$DIR/bin/slic3r.pl" "$@"

I think the problem lies in my slic3r script. I wonder, whether I shall modify the script to call readlink -f on $0 as well?

@burbilog @probonopd I think the AppImage generates some debug info if started with DEBUG=1. @burbilog  would you please provide that log? Thanks.

bubnikv, 2017-02-18:
> @probonopd  is the script that you are using to produce the AppImage available on GitHub?

It is part of our build suite to build and pack on all platforms. The build suite is currently private. But I am not doing anything special, I just slightly modified the scripts you have provided.

probonopd, 2017-02-18:
Here are my logs of mine and yours:

[Slic3r-1.31.6.prusa3d.glibc2.14-x86_64.AppImage.log.txt](https://github.com/prusa3d/Slic3r/files/784972/Slic3r-1.31.6.prusa3d.glibc2.14-x86_64.AppImage.log.txt)
[Slic3r-1.33.5-prusa3d-linux64-full-201702171509.AppImage.log.txt](https://github.com/prusa3d/Slic3r/files/784973/Slic3r-1.33.5-prusa3d-linux64-full-201702171509.AppImage.log.txt)

probonopd, 2017-02-18:
Ah, `export LD_LIBRARY_PATH="$DIR/bin"` in usr/bin/slic3r throws away the `LD_LIBRARY_PATH` that AppRun has set before. I guess it should be `export LD_LIBRARY_PATH="$DIR/bin":$LD_LIBRARY_PATH`?

burbilog, 2017-02-18:
DEBUG=1  && ./Slic3r-1.33.5-prusa3d-linux64-full-201702171509.AppImage       

~~~
LC_PAPER=ru_RU.UTF-8
XDG_VTNR=7
MANPATH=/usr/share/man:/usr/openwin/share/man:/usr/local/man:/usr/local/lib/perl5/man:/usr/local/lib/perl5/5.00502/man:/usr/local/man:/usr/local/share/man:/usr/share/man:/usr/man:/var/qmail/man:/usr/local/nessus/man:/usr/local/lib/perl5/5.8.7/perl/man/:/usr/local/lib/perl5/5.8.7/man/:/usr/share/openssl/man/:/usr/local/lib/perl5/5.8.8/man/:/usr/local/share/man/
NNTPSERVER=news.rt.ru
KDE_MULTIHEAD=false
LC_ADDRESS=ru_RU.UTF-8
XDG_SESSION_ID=2
SSH_AGENT_PID=4961
PAM_KWALLET5_LOGIN=/tmp/kwallet5_rm.socket
LC_MONETARY=ru_RU.UTF-8
GPG_AGENT_INFO=/home/rm/.gnupg/S.gpg-agent:0:1
SHELL=/usr/bin/zsh
TERM=xterm
SAVEHIST=500
HISTSIZE=500
histsize=500
KONSOLE_DBUS_SERVICE=:1.46
GTK2_RC_FILES=/etc/gtk-2.0/gtkrc:/home/rm/.gtkrc-2.0:/home/rm/.config/gtkrc-2.0
DIRSTACKSIZE=20
dirstacksize=200
PERL5LIB=/home/rm/perl5/lib/perl5
perlbrew_command=perlbrew
KONSOLE_PROFILE_NAME=Profile #1
WATCH=notme
QT_LINUX_ACCESSIBILITY_ALWAYS_ON=1
CVSROOT=/usr/local/src/cvsroot
LD_PRELOAD=/tmp/.mount_93wzoj/libunionpreload.so
LC_NUMERIC=ru_RU.UTF-8
WINDOWID=85983237
PERL_MB_OPT=--install_base "/home/rm/perl5"
SHELL_SESSION_ID=6b3787e173924000a853f1691aceee49
LC_ALL=ru_RU.KOI8-R
XDG_SESSION_CLASS=user
KDE_FULL_SESSION=true
USER=rm
QT_ACCESSIBILITY=1
LC_TELEPHONE=ru_RU.UTF-8
LD_LIBRARY_PATH=/tmp/.mount_93wzoj/usr/lib/:/tmp/.mount_93wzoj/usr/lib/i386-linux-gnu/:/tmp/.mount_93wzoj/usr/lib/x86_64-linux-gnu/:/tmp/.mount_93wzoj/usr/lib32/:/tmp/.mount_93wzoj/usr/lib64/:/tmp/.mount_93wzoj/lib/:/tmp/.mount_93wzoj/lib/i386-linux-gnu/:/tmp/.mount_93wzoj/lib/x86_64-linux-gnu/:/tmp/.mount_93wzoj/lib32/:/tmp/.mount_93wzoj/lib64/:
MAILCHECK=604800
PERLBREW_BASHRC_VERSION=0.74
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session1
PGLIB=/usr/local/pgsql/lib
RPROMPT=[%~ %T %w] 
XDG_SEAT_PATH=/org/freedesktop/DisplayManager/Seat0
LOGCHECK=300
SSH_AUTH_SOCK=/tmp/ssh-6n21PUSBKlwu/agent.4893
SESSION_MANAGER=local/moloch:@/tmp/.ICE-unix/5048,unix/moloch:/tmp/.ICE-unix/5048
DEFAULTS_PATH=/usr/share/gconf//usr/share/xsessions/plasma.default.path
PERLBREW_ROOT=/home/rm/perl5/perlbrew
XDG_CONFIG_DIRS=/etc/xdg/xdg-/usr/share/xsessions/plasma:/etc/xdg:/usr/share/kubuntu-default-settings/kf5-settings
PATH=/tmp/.mount_93wzoj/usr/bin/:/tmp/.mount_93wzoj/usr/sbin/:/tmp/.mount_93wzoj/usr/games/:/tmp/.mount_93wzoj/bin/:/tmp/.mount_93wzoj/sbin/:/home/rm/perl5/bin:/home/rm/perl5/perlbrew/bin:/opt/android-sdk-linux/tools:/opt/android-sdk-linux/platform-tools:/usr/local/bin:/usr/local/sbin:/bin:/sbin:/usr/bin:/usr/sbin:/usr/local/etc/rc.d:/var/qmail/bin:/home/rm/bin:/opt/vpopmail/bin:/usr/local/pgsql/bin:/usr/local/nessus/bin:/usr/local/nessus/sbin:/usr/local/vpopmail/bin:/home/rm/android-sdk-linux_x86/platform-tools:/usr/brlcad/bin
DESKTOP_SESSION=/usr/share/xsessions/plasma
GSETTINGS_SCHEMA_DIR=/tmp/.mount_93wzoj/usr/share/glib-2.0/schemas/:
QT_IM_MODULE=compose
UNION_PRELOAD=/tmp/.mount_93wzoj
LC_IDENTIFICATION=ru_RU.UTF-8
BLOCKSIZE=1k
PWD=/home/rm/Downloads
XDG_SESSION_TYPE=x11
KONSOLE_DBUS_WINDOW=/Windows/1
KDE_SESSION_UID=1000
LANG=ru_RU.KOI8-R
MANDATORY_PATH=/usr/share/gconf//usr/share/xsessions/plasma.mandatory.path
LC_MEASUREMENT=ru_RU.UTF-8
PERLBREW_HOME=/home/rm/.perlbrew
ZLE_LINE_ABORTED=
PERLLIB=/tmp/.mount_93wzoj/usr/share/perl5/:/tmp/.mount_93wzoj/usr/lib/perl5/:
[... 128 lines cut ...]
+ RETURN=no
+ '[' no = no ']'
+ yesno 'Disable question?' 'Should this question be permanently disabled for Slic3r?\n\nTo re-enable this question you have to delete\n"/home/rm/.local/share/appimagekit/Slic3r_no_desktopintegration"'
+ TITLE='Disable question?'
+ TEXT='Should this question be permanently disabled for Slic3r?\n\nTo re-enable this question you have to delete\n"/home/rm/.local/share/appimagekit/Slic3r_no_desktopintegration"'
+ '[' -x /usr/bin/zenity ']'
+ LD_LIBRARY_PATH=
+ zenity --question '--title=Disable question?' '--text=Should this question be permanently disabled for Slic3r?\n\nTo re-enable this question you have to delete\n"/home/rm/.local/share/appimagekit/Slic3r_no_desktopintegration"'
+ RETURN=no
+ '[' no = yes ']'
+ exit 0
+ atexit
+ '[' -z '' ']'
+ '[' 1 -eq 0 ']'
+ exec /tmp/.mount_93wzoj/usr/bin/slic3r --gui
Can't locate Try/Tiny.pm in @INC (you may need to install the Try::Tiny module) (@INC contains: /tmp/.mount_93wzoj/usr/bin/bin/lib /home/rm/perl5/lib/perl5 /tmp/.mount_93wzoj/usr/bin/lib/site_perl/5.22.0/x86_64-linux-thread-multi /tmp/.mount_93wzoj/usr/bin/lib/site_perl/5.22.0 /tmp/.mount_93wzoj/usr/bin/lib/5.22.0/x86_64-linux-thread-multi /tmp/.mount_93wzoj/usr/bin/lib/5.22.0 .) at /home/rm/perl5/lib/perl5/LWP/Protocol.pm line 11.
BEGIN failed--compilation aborted at /home/rm/perl5/lib/perl5/LWP/Protocol.pm line 11.
Compilation failed in require at /home/rm/perl5/lib/perl5/LWP/UserAgent.pm line 13.
BEGIN failed--compilation aborted at /home/rm/perl5/lib/perl5/LWP/UserAgent.pm line 13.
Compilation failed in require at /tmp/.mount_93wzoj/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI/Plater.pm line 11.
BEGIN failed--compilation aborted at /tmp/.mount_93wzoj/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI/Plater.pm line 11.
Compilation failed in require at /tmp/.mount_93wzoj/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI.pm line 20.
BEGIN failed--compilation aborted at /tmp/.mount_93wzoj/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI.pm line 20.
Compilation failed in require at (eval 90) line 1.
~~~

probonopd, 2017-02-18:
I think the key lies in 

> @INC contains: /tmp/.mount_93wzoj/usr/bin/bin/lib /home/rm/perl5/lib/perl5 /tmp/.mount_93wzoj/usr/bin/lib/site_perl/5.22.0/x86_64-linux-thread-multi /tmp/.mount_93wzoj/usr/bin/lib/site_perl/5.22.0 /tmp/.mount_93wzoj/usr/bin/lib/5.22.0/x86_64-linux-thread-multi /tmp/.mount_93wzoj/usr/bin/lib/5.22.0

bubnikv, 2017-02-18:
@burbilog I think the problem seems to become evident. Greping perl from your log shows: 

PERL5LIB=/home/rm/perl5/lib/perl5
perlbrew_command=perlbrew
PERL_MB_OPT=--install_base "/home/rm/perl5"
PERLBREW_BASHRC_VERSION=0.74
PERLBREW_ROOT=/home/rm/perl5/perlbrew
PATH=/tmp/.mount_93wzoj/usr/bin/:/tmp/.mount_93wzoj/usr/sbin/:/tmp/.mount_93wzoj/usr/games/:/tmp/.mount_93wzoj/bin/:/tmp/.mount_93wzoj/sbin/:/home/rm/perl5/bin:/home/rm/perl5/perlbrew/bin:/opt/android-sdk-linux/tools:/opt/android-sdk-linux/platform-tools:/usr/local/bin:/usr/local/sbin:/bin:/sbin:/usr/bin:/usr/sbin:/usr/local/etc/rc.d:/var/qmail/bin:/home/rm/bin:/opt/vpopmail/bin:/usr/local/pgsql/bin:/usr/local/nessus/bin:/usr/local/nessus/sbin:/usr/local/vpopmail/bin:/home/rm/android-sdk-linux_x86/platform-tools:/usr/brlcad/bin
PERLBREW_HOME=/home/rm/.perlbrew
PERLLIB=/tmp/.mount_93wzoj/usr/share/perl5/:/tmp/.mount_93wzoj/usr/lib/perl5/:
PERL_LOCAL_LIB_ROOT=/home/rm/perl5
PERL_MM_OPT=INSTALL_BASE=/home/rm/perl5
Can't locate Try/Tiny.pm in @INC (you may need to install the Try::Tiny module) (@INC contains: /tmp/.mount_93wzoj/usr/bin/bin/lib /home/rm/perl5/lib/perl5 /tmp/.mount_93wzoj/usr/bin/lib/site_perl/5.22.0/x86_64-linux-thread-multi /tmp/.mount_93wzoj/usr/bin/lib/site_perl/5.22.0 /tmp/.mount_93wzoj/usr/bin/lib/5.22.0/x86_64-linux-thread-multi /tmp/.mount_93wzoj/usr/bin/lib/5.22.0 .) at /home/rm/perl5/lib/perl5/LWP/Protocol.pm line 11.

It looks like we (either me or probably @probonopd) need to filter out the PERL environment variables before running a perl script packed into an AppImage. Maybe AppImage should initialize the shell environment from scratch? I am far from being a Linux expert, so I don't know how to do that as of now. I will look into what PERL environment variables are used to put together the @INC paths and what are their priorities.

bubnikv, 2017-02-18:
@burbilog  would you please try to undefine PERL5LIB before running the AppImage?

bubnikv, 2017-02-18:
Reading 
http://stackoverflow.com/questions/2526804/how-is-perls-inc-constructed-aka-what-are-all-the-ways-of-affecting-where-pe

I need to unset PERL5LIB and PERLLIB in the slic3r shell wraper. I will try to update the Linux and OSX builds as both may potentially affected. Also the Windows slic3r.exe wrapper shall filter out the two variables.

burbilog, 2017-02-18:
@bubnikv yes, removing PERL5LIB from environment fixes the problem.

bubnikv, 2017-02-18:
@burbilog Great, I am running another build with
export PERL5LIB=
export PERLLIB=
at the start of the slic3r shell script. Hope this will fix it.

probonopd, 2017-02-18:
[This](https://github.com/probonopd/AppImageKit/blob/cc5513754e6cc9e6c4f2677a2afc211c7e08acc3/AppRun.c#L132-L134) is what I am currently doing in `AppRun.c`.

Its intention is to _add_ the inside of the AppImage to the Perl search path, similar to what I do with `$PATH`.

Should I change it?

bubnikv, 2017-02-18:
@probonopd I think it depends. I suppose Slic3r is the first AppImage packing its own Perl interpreter. If the AppImage uses the system-wide perl interpreter, then your implementation is all right.

I think it would be sufficient to keep it as it is, but to document it well, so if someone is packing a Perl interpreter into the AppImage, he would know he needs to unset PERL5LIB and PERLLIB in his startup script.

probonopd, 2017-02-18:
I think bundling the Perl interpreter is the right thing to do.

bubnikv, 2017-02-18:
@burbilog I have updated the Linux & OSX binary builds to unset PERLLIB & PERL5LIB. Please download the updated AppImage from 
https://github.com/prusa3d/Slic3r/releases/tag/version_1.33.5
and give it a try, so we can close this issue :-)

probonopd, 2017-02-18:
Don't know if this is serious, but I am still getting 

```
me@host:~/LibrePCB/build$ /home/me/Downloads/Slic3r-1.33.5-prusa3d-linux64-full-201702181304.AppImage 
Use of uninitialized value in subroutine entry at /tmp/.mount_BchhWk/usr/bin/lib/site_perl/5.22.0/Slic3r/GUI/Plater.pm line 170.
```

bubnikv, 2017-02-18:
@probonopd I don't know what is the Perl complaining about. The line initializes a menu, and with success.
        $self->{htoolbar}->AddTool(TB_LAYER_EDITING, 'Layer Editing', Wx::Bitmap->new($Slic3r::var->("variable_layer_height.png"), wxBITMAP_TYPE_PNG), wxNullBitmap, 1, undef, 'Layer Editing');

It must be something inside the wxWidgets perl binding. Not nice, but seems to be harmless. Something for the next release I suppose.

burbilog, 2017-02-19:
@bubnikv Yes, it works now.

bubnikv, 2017-02-19:
Thanks to all involved, happily closing :-)

closed 2017-02-19 by bubnikv