prusa3d/PrusaSlicer #141: 1.33.4 and 1.33.5 uses old perl supports

jinie, 2017-02-18:
### Version
1.33.5

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Mac OS 10.12.3

### Behavior
When generating supports, Slic3r uses the old Perl support functions instead of the new C++ ones added in 1.33.3

bubnikv, 2017-02-18:
I doubt it. I just tested it.

I think you have your Slic3r misconfigured. You likely want to disable the sheath and to switch to rectilinear support type. Best please use https://github.com/prusa3d/slic3r-settings

bubnikv, 2017-02-18:
I have also updated the OSX 1.33.5 build recently to reset the PERL5LIB and PERLLIB environment variables just in case you have some weird Perl installed on your system. Now the system perl should not intervene with Slic3r.

jinie, 2017-02-18:
I'll try disabling sheaths when I get home.
I enabled supports, and it immediately hangs while generating supports, and when I killed it, it said Perl was the cause of the hang.

jinie, 2017-02-18:
This is what i get when i kill it while stuck generating supports (Using Mk2 profile settings, dRectiliniar support, disabled sheaths) :
Date/Time:       2017-02-18 20:51:50 +0100
OS Version:      Mac OS X 10.12.3 (Build 16D32)
Architecture:    x86_64h
Report Version:  25

Data Source:     Stackshots

Command:         perl5.22.0
Path:            /Applications/Slic3r.app/Contents/MacOS/perl5.22.0
Version:         ??? (???)
Parent:          launchd [1]
PID:             29366

Event:           hang
Duration:        1.70s (process was unresponsive for 6 seconds before sampling)
Steps:           17 (100ms sampling interval)

Hardware model:  MacBookPro12,1
Active cpus:     4

Fan speed:       2930 rpm

--------------------------------------------------
Timeline format: stacks are sorted chronologically
Use -i and -heavy to re-report with count sorting
--------------------------------------------------

Heaviest stack for the main thread of the target process:
  17  <truncated backtrace>
  17  __semwait_signal + 10 (libsystem_kernel.dylib + 106458) [0x7fffa56dffda]
 *17  semaphore_wait_continue + 0 (kernel + 1140704) [0xffffff80003167e0]

Process:         perl5.22.0 [29366]
Path:            /Applications/Slic3r.app/Contents/MacOS/perl5.22.0
Architecture:    x86_64
Parent:          launchd [1]
UID:             501
Task size:       605 MB (-25)
CPU Time:        0.498
Note:            Unresponsive for 6 seconds before sampling
Note:            2 idle work queue threads omitted

  Thread 0x1cf89d           DispatchQueue 1           17 samples (1-17)         priority 47 (base 47)
  <process frontmost, thread QoS user interactive (requested user interactive), process unclamped, process received importance donation from WindowServer [159], IO tier 0>
  17  <truncated backtrace> 1-17
    17  __semwait_signal + 10 (libsystem_kernel.dylib + 106458) [0x7fffa56dffda] 1-17
     *17  semaphore_wait_continue + 0 (kernel + 1140704) [0xffffff80003167e0] 1-17

  Thread 0x1cf8d4           Thread name "com.apple.NSEventThread"               17 samples (1-17)         priority 47 (base 47)     cpu time 0.016
  <process frontmost, thread QoS user interactive (requested user interactive), process unclamped, process received importance donation from WindowServer [159], IO tier 0>
  17  thread_start + 13 (libsystem_pthread.dylib + 12797) [0x7fffa57c81fd] 1-17
    17  _pthread_start + 286 (libsystem_pthread.dylib + 14839) [0x7fffa57c89f7] 1-17
      17  _pthread_body + 180 (libsystem_pthread.dylib + 15019) [0x7fffa57c8aab] 1-17
        17  _NSEventThread + 205 (AppKit + 1657442) [0x7fff8dcd8a62] 1-17
          17  CFRunLoopRunSpecific + 420 (CoreFoundation + 555380) [0x7fff90059974] 1-17
            4   __CFRunLoopRun + 1361 (CoreFoundation + 557345) [0x7fff9005a121] 1-4
              4   __CFRunLoopServiceMachPort + 212 (CoreFoundation + 560276) [0x7fff9005ac94] 1-4
                4   mach_msg_trap + 10 (libsystem_kernel.dylib + 74634) [0x7fffa56d838a] 1-4
                 *4   ipc_mqueue_receive_continue + 0 (kernel + 886832) [0xffffff80002d8830] 1-4
            1   __CFRunLoopRun + 2389 (CoreFoundation + 558373) [0x7fff9005a525] 5
              1   __CFRunLoopDoSource1 + 465 (CoreFoundation + 591425) [0x7fff90062641] 5
                1   __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE1_PERFORM_FUNCTION__ + 41 (CoreFoundation + 591561) [0x7fff900626c9] 5
                  1   __CFMachPortPerform + 253 (CoreFoundation + 591837) [0x7fff900627dd] 5
                    1   MessageHandler(__CFMachPort*, void*, long, void*) + 51 (HIToolbox + 239353) [0x7fff8f5ef6f9] 5
                      1   PullEventsFromWindowServerOnConnection(unsigned int, unsigned char, __CFMachPortBoost*) + 102 (HIToolbox + 239501) [0x7fff8f5ef78d] 5
                        1   SLEventCreateNextEvent + 50 (SkyLight + 2029890) [0x7fffa1a90942] 5
                          1   SLSGetNextEventRecordInternal + 84 (SkyLight + 845990) [0x7fffa196f8a6] 5
                            1   CGSSnarfAndDispatchDatagrams + 180 (SkyLight + 177388) [0x7fffa18cc4ec] 5
                              1   mach_msg_trap + 10 (libsystem_kernel.dylib + 74634) [0x7fffa56d838a] 5
[... 154 lines cut ...]
                                 *2   ??? (kernel + 1507988) [0xffffff8000370294] 16-17
                                   *2   vm_fault_page + 2635 (kernel + 1488155) [0xffffff800036b51b] 16-17
                                     *2   vm_compressor_pager_get + 222 (kernel + 1453966) [0xffffff8000362f8e] 16-17
                                       *2   vm_compressor_get + 225 (kernel + 1446673) [0xffffff8000361311] 16-17
                                         *2   ??? (kernel + 1447515) [0xffffff800036165b] 16-17
                                           *2   c_seg_swapin + 161 (kernel + 1442113) [0xffffff8000360141] 16-17
                                             *2   vm_swap_get + 171 (kernel + 1464907) [0xffffff8000365a4b] 16-17
                                               *1   vm_swapfile_io + 118 (kernel + 6299734) [0xffffff8000802056] 16
                                                 *1   vm_map_create_upl + 99 (kernel + 1702099) [0xffffff800039f8d3] 16
                                                   *1   lck_rw_lock_shared_gen + 105 (kernel + 2089065) [0xffffff80003fe069] 16
                                                     *1   thread_block_reason + 222 (kernel + 1091230) [0xffffff800030a69e] 16
[... cut ...]
                                                               *1   lck_mtx_sleep + 132 (kernel + 1049444) [0xffffff8000300364] 17
                                                                 *1   thread_block_reason + 222 (kernel + 1091230) [0xffffff800030a69e] 17
                                                                   *1   ??? (kernel + 1095803) [0xffffff800030b87b] 17
                                                                     *1   machine_switch_context + 206 (kernel + 2102494) [0xffffff80004014de] 17

bubnikv, 2017-02-18:
> 17 1-17
> 17 ??? ( + 917770) [0x10e3ed10a] 1-17
> 17 ??? ( + 2339014) [0x10e5480c6] 1-17
> 17 ??? ( + 2374345) [0x10e550ac9] 1-17
> 17 ??? ( + 2408996) [0x10e559224] 1-17
> 17 ??? ( + 2480338) [0x10e56a8d2] 1-17

Would you please
1. Export and attach a config file (menu file, export config)
2.Atrach an stl

jinie, 2017-02-18:
[files.tar.gz](https://github.com/prusa3d/Slic3r/files/785294/files.tar.gz)

bubnikv, 2017-02-19:
I just tested on Windows 10 and OSX Maveric and both sliced your model with your settings without any issue.

jinie, 2017-02-19:
Something must be off here then. I let it run, and eventually Perl ate all my system memory while trying to slice my model. I'll try deleting the settings and try again.
I'm confused though, why would it try to generate supports using Perl if it has been removed ?

bubnikv, 2017-02-19:
Perl has not been fully removed. Perl calls a C++ (native) code, perl is the main process, so if the C++ code (a shared library) eats your memory, the system thinks it is the perl interpreter.

This may be a similar issue to yours and it is pretty critical. I am running a build now fixing this issue. Let's see, whether it helps you.
https://github.com/prusa3d/Slic3r/issues/142

bubnikv, 2017-02-19:
Would you please try with the latest release, so I can happily close this issue?
https://github.com/prusa3d/Slic3r/releases/tag/version_1.33.7

jinie, 2017-02-19:
latest release fixes my problem.

closed 2017-02-19 by jinie

bubnikv, 2017-02-19:
Great, thank you.