prusa3d/PrusaSlicer #142: Windows 10 machine locks up while generating supports

hmbusch, 2017-02-19:
### Version
Version 1.33.6-prusa3d-win64

### Operating system type + version
Windows 10 Home Edition, 64 bit, current patchlevel

### Behavior

1. I load the STL file from http://www.thingiverse.com/thing:1930525 (attached) and increase the number of copies to 3 total. I leave orientation and placement of the models as is.

2. I configure the support settings as shown in the screenshot.

3. As I have background processing disabled, I click on the "Slice now" button.

4. CPU usage rises to 100% (as is to be expected). A short while after the progress bar hits the "Generating support" phase, the whole computer locks up. The images freezes, the mouse cannot be moved, the Task Manager cannot be opened. I had to reset my machine.

5. The same settings with version 1.31.6-prusa3d-win64 result in a completed calculation and G-Code generation

#### STL/Config (.ZIP) where problem occurs
[image]

[Prusa_IGES_Drylin_adaptor.zip](https://github.com/prusa3d/Slic3r/files/786147/Prusa_IGES_Drylin_adaptor.zip)

[Slic3r_config_bundle.zip](https://github.com/prusa3d/Slic3r/files/786149/Slic3r_config_bundle.zip)

bubnikv, 2017-02-19:
I have never seen something like this. It allocated 30GB of RAM on my computer, then the screen started to flicker :-)

bubnikv, 2017-02-19:
Thanks, it is an infinite loop due to numerical rounding effects.
Fixed in https://github.com/prusa3d/Slic3r/issues/142

closed 2017-02-19 by bubnikv

hmbusch, 2017-02-19:
Well that was fast. Thank you!

bubnikv, 2017-02-19:
Yeah, it is a new code, there are dragons. I am trying to stabilize Slic3r for the final release to our customers. So I am thankful to you for the bug report.