prusa3d/PrusaSlicer #143: Infill patterns triangles/stars seem to be mixed-up

simonkuehling, 2017-02-20:
### Version
1.33.7-prusa3d-linux64

### Operating system type + version
Ubuntu Linux 16.04 LTS

### Behavior
With the config attached below, when switching infill patterns between "stars" and "triangles", i seem to get switched-around patterns ("stars" generates triangles and "triangles" generates stars)...

[image]

Can someone confirm this?

#### STL/Config (.ZIP) where problem occurs
[1-33-7_config.ini.zip](https://github.com/prusa3d/Slic3r/files/786937/1-33-7_config.ini.zip)

bubnikv, 2017-02-20:
You seem to be right.

bubnikv, 2017-02-20:
Fixed. https://github.com/prusa3d/Slic3r/commit/006173342cb339b3782f9cbb603dea82abb60331

closed 2017-02-20 by bubnikv