prusa3d/PrusaSlicer #144: Weight: %.1fg, Cost: %.1f

NiklasOlofsson, 2017-02-21:
### Version
1.33.8 prusa3d

### Operating system type + version
Windows 1o Pro

### Behavior
I am using Slic3r with Repetier-Host.
The output looks like this:
11:45:20.603 : Slic3r command:C:\Slic3r\Slic3r-Prusa\slic3r.exe --load "slic3r_settings.ini" --print-center 100.00,100.00 -o "composition.gcode" "composition.amf"
11:45:21.208 : <Slic3r> => Generating perimeters
11:45:21.210 : <Slic3r> => Processing triangulated mesh
11:45:21.426 : <Slic3r> => Infilling layers
11:45:21.427 : <Slic3r> => Preparing infill
**11:45:21.562 : <Slic3r> => Weight: %.1fg, Cost: %.1f**
11:45:21.563 : <Slic3r> => Generating skirt
11:45:21.614 : <Slic3r> => Exporting G-code to composition.gcode
11:45:22.280 : <Slic3r> Done. Process took 0 minutes and 1.074 seconds
11:45:22.280 : <Slic3r> Filament required: 492.2mm (1.2cm3)

I have filled in density and cost. Is it a Slic3r or Repetier-Host problem when it shows "Weight: %.1fg, Cost: %.1f" ?

bubnikv, 2017-02-21:
It's slic3r's glitch.

bubnikv, 2017-02-22:
Fixed in https://github.com/prusa3d/Slic3r/commit/9f660738b36a5e83933385d2463b24b4df50b81f

NiklasOlofsson, 2017-02-22:
I added the sprintf and now it outputs "<Slic3r> => Weight: 0.0g, Cost: 0.0".

bubnikv, 2017-03-23:
You have to fill in the filament weight and cost into filament settings.
Does it work for you then?

NiklasOlofsson, 2017-03-23:
This is what I have
[image]

Sometimes the output is a very high number but only with the first, really small object. Most of the time it reports 0.0.

NiklasOlofsson, 2017-03-25:
Like this I got just now:
<Slic3r> => Weight: 0.0g, Cost: 1222848817564466700000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0

bubnikv, 2017-03-30:
Would you please provide a config.ini and an stl to reproduce this behavior?

NiklasOlofsson, 2017-03-30:
stl does not matter, it is a bit random. Most of the time, it reports 0.0
Just now, I got Weight: 562950093375614.5g, Cost: 0.0. Slicing again and it's back at 0.0 again.
[config.zip](https://github.com/prusa3d/Slic3r/files/882761/config.zip)

bubnikv, 2017-03-30:
Thanks, my college has pointed the error today and he connects it to an
STL, for which Slic3r reports errors and holes.
He will hopefully provide me with some data as well.

NiklasOlofsson, 2017-03-30:
Slicing this stl several times
[plate.zip](https://github.com/prusa3d/Slic3r/files/882925/plate.zip)

0.0 most of the time, but suddenly:

> <Slic3r> => Weight: 0.0g, Cost: 48909926794989617000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0

NiklasOlofsson, 2017-04-04:
```
17:11:05.384 : <Slic3r> => Processing triangulated mesh
17:11:05.401 : <Slic3r> => Generating perimeters
17:11:05.506 : <Slic3r> => Infilling layers
17:11:05.507 : <Slic3r> => Preparing infill
17:11:05.612 : <Slic3r> => Weight: 54927406468307401000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000.0g, Cost: 0.0
17:11:05.614 : <Slic3r> => Generating skirt
17:11:05.660 : <Slic3r> => Exporting G-code to composition.gcode
17:11:06.159 : <Slic3r> Done. Process took 0 minutes and 0.775 seconds
17:11:06.159 : <Slic3r> Filament required: 795.4mm (1.9cm3)
17:11:06.238 : <Slic3r> Slic3r installed in a loclized path. Using an 8.3 path: "C:\Users\XXXXX\3D\Slic3r\SLIC3R~2\"

```

[bbClampRight.zip](https://github.com/prusa3d/Slic3r/files/893610/bbClampRight.zip)

bubnikv, 2018-02-12:
May I ask you to re-test with the current Slic3r PE 1.39.1-alpha? I have fixed an issue, which sounds similar to yours https://github.com/prusa3d/Slic3r/commit/ea9920e5d679e2deb19e553ee504cafea7b89714.
Thank you.

bubnikv, 2018-02-12:
I just quickly tested myself and I am getting reasonable numbers wiht the Slic3r PE 1.39.1-alpha.

bubnikv, 2018-03-14:
This issue is stalled, most likely fixed in 1.39.1. Closing.

closed 2018-03-14 by bubnikv