prusa3d/PrusaSlicer #145: [Request] Add Used Filament (length) and Estimated Print Time to the Sliced Info

Anubis9, 2017-02-21:
### Version
1.33.7-prusa3d-win64

### Operating system
Windows 10 Pro

### Feature Request
1. Slic3r shows only the weight and the volume of used filament. Knowing that might still not be enough to estimate if there is enough filament left on the spool. Also, as far as I am aware the generated g-code already has the length of the filament used. Would be nice to have the filament length added to the Sliced info panel, together with the volume and weight.

2. Deciding on the layer height is sometimes done based on the time needed to print the object. For the moment, I use http://gcode.ws/ to estimate the print time based on the g-code, but it requires constant uploading and processing. It would be useful to test the estimated time for different presets of layer heights inside Slic3r in the Sliced Info panel.

NikoDelarich, 2017-02-21:
Accurate estimation is IMO a very important feature that nobody seems to get right.
I currently use Simplify3D and/or OctoPrint to estimate print-time, both of which can be very far off the actual print time.

It would be great if Slic3r could use the Marlin planning code to get an accurate estimate.
This was already proposed once in an OctoPrint issue: https://github.com/foosel/OctoPrint/issues/270#issuecomment-25325951
A working version can be found here: https://github.com/davr/Marlin/tree/estimate

Dougwar, 2017-02-21:
It has to be like in kisslicer, where you can put your acceleration value, it will be much more accuracy.

bubnikv, 2017-03-23:
Cura does that as well. If only day had more than 24 hours.

m-Th, 2017-06-29:
Backport the code from PrusaControl for the estimated print time...

bubnikv, 2018-02-12:
We have integrated an accurate print time estimator, please see the relase notes.
https://github.com/prusa3d/Slic3r/releases/tag/version_1.39.1-alpha
Closing.

closed 2018-02-12 by bubnikv

eyal0, 2018-06-24:
I am working on integrating the new, accurate estimate into https://github.com/OctoPrint/OctoPrint-Slic3r .

If this feature would be useful to you, @NikoDelarich , let me know.

I currently only support a single material but multi material support is available.  If someone wants to provide a sample .gcode file with multimaterial comments in it, I would appreciate it.  Just send me the lines of the gcode file around "estimated printing time" and "filament used".  Thanks.