prusa3d/PrusaSlicer #147: [Request] Place surface in bed

Dougwar, 2017-02-21:
Version
1.33.7-prusa3d-win64
Operating system
Windows 10 Pro
Feature Request
Place surface in bed, this allow you to select a surface to align with the bed, now I have to use another software for this feature

Thanks

bubnikv, 2017-02-22:
I suppose the request is to pick a surface on an object and then to rotate the object, so the selected surface of the object aligns with the bed, right?

Dougwar, 2017-02-22:
yes

Eternyt, 2017-12-12:
Will be great!

mariusstrom, 2018-05-14:
FYI this is something that Simplify3D can do (CTRL+L in S3D). I know Slic3r probably isn't trying to compete for feature parity with S3D, but this is exceptionally helpful with complex models.

qknight, 2018-05-15:
https://github.com/slic3r/Slic3r/issues/4397

bubnikv, 2018-07-24:
@lukasmatena will take care of it. Thanks, Lukas!

bubnikv, 2018-08-23:
@lukasmatena is already polishing the feature.

bubnikv, 2018-09-01:
Implemented in 1.41.0 final release.

closed 2018-09-01 by bubnikv

ctgreybeard, 2018-09-01:
I was looking forward to this feature but I don't see how to use it in 1.41.0+ on my Mac ...

geeksville, 2018-09-01:
@bubnikv Do you know where I would find this feature in the 1.41.0 final GUI?  On my linux-64 bit machine I've tried right/left clicking on faces and looking in the menus without luck.  Thanks!

bubnikv, 2018-09-02:
It is the third icon in the scene at the left side of the window.

BernhardBaumrock, 2018-10-28:
great, thx, needed this today :) 
[image]

wavexx, 2018-11-06:
I'm having an hard time to use this feature. I downloaded the linux64 build to be certain, but as soon as I try to select any object on the build plate, they're immediately deselected. If I select them though the part list (on the right pane), they stay selected, but pressing the buttons on the left side doesn't really do much (the part the deselected again).

Anybody else experiencing this issue?

lukasmatena, 2018-11-06:
@wavexx Thanks for the feedback. If you feel there is a bug, it might be cleaner to open a new issue. It would help if you provided exact version of Slic3r and ideally your Linux distribution and version.

The function is out there for quite some time already and there is just one selection-related annoyance regarding the place-to-bed function that I know of: if you select an object, activate the function and then try to select another object by clicking it in the scene, and happen to be unlucky enough to click it on a spot where an active plane is showing up, the function is deactivated and all objects deselected. But this is not a show-stopper. The bug that you report seems to be more serious, since if I understood correctly, you are unable to use the function at all, is that correct? Can you confirm there are no other selection-related issues beside one with the place-to-bed function?

wavexx, 2018-11-06:
On Tue, Nov 06 2018, lukasmatena wrote:
> @wavexx Thanks for the feedback. If you feel there is a bug, it might
> be cleaner to open a new issue. It would help if you provided exact
> version of Slic3r and ideally your Linux distribution and version.

This is on debian unstable (generally updated daily). I'm using the 64
bit binary build as I didn't want to rebuild by source. Since it's
probably relevant, this is using the intel KMS drivers on linux 4.18, in
plain X11 (no compositor/wayland).

> objects deselected. But this is not a show-stopper. The bug that you
> report seems to be more serious, since if I understood correctly, you
> are unable to use the function at all, is that correct? Can you
> confirm there are no other selection-related issues beside one with
> the place-to-bed function?

It seems to me that selection itself behaves badly. Clicking on an
object placed on the bed works a few times, then misbehaves. The object
is selected and then deselected by itself for some reason.

I noticed the binary build provides wx, but not gtk. As such, this is
using gtk 2.24.32 from debian.

I can open a new issue if this sounds new to you.

lukasmatena, 2018-11-07:
@wavexx If it's really a different problem that I described above, it is new for me. However, your description is not clear enough for me to judge. At first I thought that you cannot use the function at all, now I'm hearing about "an object placed on bed", which would suggest otherwise. All I can say is that If you see the problem with only one object present, it is something new. The same if you're having selection issues without ever clicking the place-on-bed button.

Now if you're really experiencing different issue than I described above... Using the unstable debian should not cause issues. Your gtk version should also be fine. If you know how to do it, you can try compiling Slic3r (master) yourself and check if it helps (unlikely). In theory you could also try compiling branch dev_native and try it there. dev_native is the current development branch which differs greatly from master and suffers from many bugs, but this bug may be already solved there. There is nothing else that I can think of at this point.