prusa3d/PrusaSlicer #148: Test failed

burbilog, 2017-02-22:
### Version
8174c05df4601a1b8b8225baffc0a0873e2f6557

### Operating system type + version
14.04.5 LTS, Trusty Tahr

### Behavior
Compilation fails:

~~~
t/cooling.t .................... 1/11 Use of uninitialized value $_ in numeric gt (>) at t/cooling.t line 154.
Can't call method "first" without a package or object reference at t/cooling.t line 154.
# Looks like your test exited with 255 just after 11.
t/cooling.t .................... Dubious, test returned 255 (wstat 65280, 0xff00)
All 11 subtests passed 
~~~

#### STL/Config (.ZIP) where problem occurs
No config

bubnikv, 2017-02-22:
I am no it, it hurts me more than you :-(

bubnikv, 2017-02-22:
Fixed in the current master. But please expect some quirks in the non-release state of the code.

closed 2017-02-22 by bubnikv