prusa3d/PrusaSlicer #149: Slice bug

sh4jonny, 2017-02-22:
last version - Slic3r-1.33.8-prusa3d-win64-full-201702210907
windows 8.1 Pro x64 with last updates

slice a popular model hair lion from http://www.thingiverse.com/

http://www.thingiverse.com/thing:2007221

and have this troubles

[image]
[image]

in slic3r 1.2.9 didn't see it.

bubnikv, 2017-02-22:
Disable "detect thin walls" feature, it will print the perimeter with two lines.

bubnikv, 2017-02-22:
This is not really a bug of Slic3r, it is a result of your printer (likely a 0.5mm nozzle?) and the way Slic3r tries to squeeze one or two lines into the ring. You are on the threshold of placing 1 or two lines. Normally you would modify the model, but I suppose you have no access to it. You may also increase or decrease the external perimeter extrusion width to move away from the threshold a bit.

I agree the current behavior is not ideal and it is on my list to do something about it.

sh4jonny, 2017-02-22:
thank you!
but i have 2 slicers with common profiles. yes nozzle is 0.4. external perimeter is 0.4 and internal 0.5 detect thin walls is enabled in both slicers. but in slic3r 1.2.9 i didn't see any troubles on wall

bubnikv, 2017-02-22:
Would you please provide a screenshot of the layers generated by 1.2.9?

I don't think 1.2.9 is better in this regard, it is just different.

sh4jonny, 2017-02-22:
[image]
[image]

sh4jonny, 2017-02-22:
and ugly preview
[image]

bubnikv, 2017-02-22:
It is the same issue as discussed here:
http://shop.prusa3d.com/forum/print-tips-slic3r-settings-kisslicer-model-repair--f12/give-up-and-bite-the-bullet-pay--t3005.html

[image]

[image]

Here is my quite long description of what's going on:
http://shop.prusa3d.com/forum/print-tips-slic3r-settings-kisslicer-model-repair--f12/give-up-and-bite-the-bullet-pay--t3005-s30.html#p24021

I am going to leave your issue here as a topic for improvement, and rename the title. Thanks for the report.

sh4jonny, 2017-02-22:
@bubnikv thank you for support!

bubnikv, 2017-06-02:
Another example of the issue is attached to https://github.com/prusa3d/Slic3r/issues/308

ahalekelly, 2017-06-08:
Reposting this from #340. Seems to be that this tapered wall is thin enough to qualify as a thin wall, but for some reason only at certain Z heights.
[image]

closed 2026-09-29 by prusaslicer-bot[bot]