prusa3d/PrusaSlicer #150: [Request]Smart Prime Pillar

sarf2k4, 2017-02-22:
Would like to ask this feature, what I meant by "smart" is the prime pillar's layer varies depending on what toolhead are used.

For example T1 will be used starting at 5mm, from 0-5mm, the prime pillar layer height are reduced in resolution probably somewhere at 0.3mm, when other toolhead are being used, it will be the same as the printing layer height or lower for even more toolhead used.

It sounds the same with simplify3d, but I liked slic3r more, but lacked this feature. 

Suggesting to add several variables for this:
Raw Filament Extrude - how far extruder will push the filament to flush the previous colour
Variable Pillar Height - boolean, to speed up printing and not affected by cooldown feature in "filament" section
Layer Step - optimal z-layer to avoid ribbing after done with the pillar

Sorry if this sounds too long or confusing, trying my best to be very specific

DrLex0, 2017-02-23:
I would like to take this a step further and revamp the entire dual extrusion strategy because the current one is so poor that I doubt whether any of the major Slic3r developers have ever tried to use a printer with two extruders. I have been thinking about this during the last week and attempted to write a proof-of-concept script that transforms Slic3r output into more sensible dual extrusion code. It is not fully functional yet, but I do have a big pile of text that I wrote down to figure out how to make it work, so get ready for some prose… Yes, this is extremely long but it is also very, very specific. Enjoy.

**How Slic3r currently treats dual extrusion (starting on layer 1):**
1. Print skirt with extruder 0.
2. Tool change.
3. Print skirt with extruder 1.
4. Tool change.
5. Print object part with extruder 0.
6. Tool change.
7. Print object part with extruder 1.
8. Move to second layer.
9. Tool change.
10. Print with extruder 0.
11. Tool change.
12. Print with extruder 1.
Steps 8 to 12 repeat.

**What is wrong with this:**
* 3 tool changes in the first layer for no good reason.
* 2 tool changes for every other layer.
* There is no real provision for priming the nozzles. A tall skirt can be used to prime only *one* of the extruders: the skirt is printed with a dedicated nozzle. The other nozzle is not primed and will fail to extrude material immediately after a tool change.
* Slic3r only provides a tall skirt as a means to keep some of the ooze from the inactive nozzle from being sprinkled across the model, but because the inactive nozzle of a dual-nozzle carriage will pass over the model while printing the skirt, this is quite ineffective.

Tool changes are to be considered *extremely expensive,* so we want to reduce them to the very minimum. We also want the deactivated nozzle to get enough time to let its filament ooze out, and wipe it before resuming printing with the other nozzle. The active nozzle must be primed because it will have considerable lag due to having oozed from the last tool change.
The ooze prevention feature that drops nozzle temperature is useless because it is riddled with bugs, and conceptually it is of little help because the nozzle still has a lot of time to ooze while it cools down, or when it heats up again.

**A proposal for a better strategy  (starting on layer 1):**
1. Print skirt with extruder 0.
2. Print object part with extruder 0.
3. Advanced tool change as described below.
4. Print object part with extruder 1.
5. Move to second layer.
6. Print object part with extruder 1.
7. Advanced tool change.
8. Print object part with extruder 0.
9. Move to third layer.
10. Print object part with extruder 0.
11. Advanced tool change.
12. Print object part with extruder 1.
and so on. Steps 5 to 12 repeat.

The advanced tool change goes as follows:
a. Move the carriage to a position outside the print, far enough that even after the tool change, the inactive nozzle will not be hovering above the print while it oozes. Ideally this position would minimise travel distance across the print, but as a start, use the average X coordinate and an Y coordinate far enough above the largest Y in the print.
b. Do the actual tool change.
c. Optional: wait to let the deactivated nozzle ooze.
d. Unretract the active nozzle and print a square filled with perimeters only, starting at the center and going outwards. Do this slowly enough that the deactivated nozzle gets enough time to stop oozing.
e. Move the inactive nozzle to the center of the tower. This causes it to be wiped on the freshly printed square.

If a layer is printed that only uses one material, then there will be no tool change. However, the structure of the priming tower must be maintained, therefore right before moving to the next layer:
a. Move to the same position as described above.
b. Print only a few of the outer perimeters of the square with the current extruder. There is no need to print an entire solid layer, we only need to provide sufficient structure.

Of course, building the priming tower is only useful up to the last layer where a tool change is needed. The heater of the inactive nozzle could be disabled after this as well, although some may prefer to keep it heated to let it plough through small bumps.

This strategy only needs one tool change per layer, it primes both nozzles, and it wipes ooze on the tower before moving back to the actual object. By printing the priming tower as an outwards spiral, the nozzle will be primed by the time it prints its outer perimeters, which are important for the tower's structure.
Of course oozing cannot be fully eliminated unless we wait a frustratingly long time between the tool swap and priming the other nozzle, but it will be a whole lot better than the current situation. Wait time and tower size should be configurable.

Here's a chunk of a semi-manually processed file of just two simple cubes in two materials ([dualTest.amf.zip](https://github.com/prusa3d/Slic3r/files/795174/dualTest.amf.zip)) that illustrates the strategy. View it in [gcode.ws](http://gcode.ws) or something to see what it does: [DUAL-TEST-HACKED.gcode.txt](https://github.com/prusa3d/Slic3r/files/795202/DUAL-TEST-HACKED.gcode.txt)
I manipulated the file a bit to drop one of the materials from one of the layers so you can see what happens in that case.

Of course this single-tower strategy cannot be used when printing two incompatible materials that cannot be printed on top of each other. In that case two priming towers are required, and both towers must be maintained at every layer, even if the layer only has one material. This is slow and wasteful, but it is the only way to get a decent print within the limitations of a dual extruder carriage design.

This strategy could also become complicated when using support material that is printed with a different layer height than the model. But I see a new option “synchronize with object layers” has recently been added…

sarf2k4, 2017-02-23:
I am using a single nozzle from two different extruder, I also believe that the way slic3r are built is to have dual nozzle and dual extruder setup (probably I'm wrong). 

I wonder if a precedence type of gcode generation can be introduced in terms of which toolhead are used previously, In essence, complete a single toolhead first before toolchange occurs, I'm guessingmore like a toolhead ID?

I'm not strong in programming but I know basic concept of it

bubnikv, 2017-03-03:
We are working on multi-material printing.
http://www.prusaprinters.org/original-prusa-i3-mk2-multi-material-upgrade-release/
http://www.prusaprinters.org/update-multimaterial-upgrade/

We currently have an external post-processor generating a wipe tower, but soon I will have to start adding stuff to Slic3r to improve our multi-material prints.

DrLex0, 2017-03-03:
For what it's worth, here's the semi-finished version of my own post-processing script, tweaked after quite a few test prints. The comments are probably more important than the code itself.
[dualstrusion-postproc.pl.txt](https://github.com/prusa3d/Slic3r/files/816830/dualstrusion-postproc.pl.txt)

It works pretty well, but if there are two materials across most of the object, the print takes ages. This is inevitable with two nozzles on the same carriage, because the only way to stop the ooze, is to wait and preferably also cool down the unused nozzle. For typical coasters and the like, where there are only a few transitions between the first and second material, this works great with only a small time penalty, and I finally don't have to pause prints and swap filaments anymore to get a perfect result.

Enabling a tall skirt is still highly recommended because the wipe is not always perfect, and at times the nozzles tend to pick up previously wiped strands. Moreover, there is also some ooze on the active nozzle when moving from the tower back to the print.

bubnikv, 2017-05-05:
> I am using a single nozzle from two different extruder, I also believe that the way slic3r are built is to have dual nozzle and dual extruder setup (probably I'm wrong).

You are right.

> I wonder if a precedence type of gcode generation can be introduced in terms of which toolhead are used previously, In essence, complete a single toolhead first before toolchange occurs, I'm guessingmore like a toolhead ID?

Slic3r currently orders the extruders by their index with the exception of the last extruder used at a layer being the first extruder used at the start of the next layer.

bubnikv, 2017-05-20:
The current master has the extruder ordering completely reordered to minimize tool changes including the supports. For non-soluble supports a support extruder shall be set to 0 to indicate that the non-soluble support shall be extruded with whatever extruder is active. For soluble supports, the "synchronize support layers" option shall be enabled to minimize tool changes.

Also the initial tool could be set in the start G-code by the [initial_tool] placeholder.

The current master implements a smart wipe tower, but the current implementation is very specific to the prusa single extruder multi material setup, it includes sequences for ramming hot material out of the nozzle, cooling and shaping the filament end during the tool change sequence, cleaning the newly loaded filament etc. The current smart wipe tower implementation will be likely extended to be parametrized in the future, but we need to support our customers first.

bubnikv, 2017-05-20:
I just released a beta to support the Prusa Multi Material upgrade. It is not the master, but it contains the extruder ordering fixes.
https://github.com/prusa3d/Slic3r/releases/tag/version_1.34.1.24

bubnikv, 2017-06-02:
The Slic3r PE 1.35.1 implements many of the features asked here.
https://github.com/prusa3d/Slic3r/releases

@DrLex0 @sarf2k4 Would you please review the multi material changes done to the latest Slic3r Prusa Editon and summarize the remaining requirements?

The latest Slic3r Prusa Edition supports both the "multiple heater/nozzle multiple extruder" and "single nozzle  multiple feeders" scenarios and it works hard to minimize tool changes, including the support, brim and skirt tool ordering.

What is certainly missing is a general wipe tower and ooze shield. The Prusa Smart Wipe Tower has been ported from our post processor and it is highly specific to our i3 MK2 Multi Material upgrade kit.

sarf2k4, 2017-06-02:
Okay, i just checked the new version briefly, while i was downloading and
it got me thinking that different material would require different
temperature set. I would like to propose another set of feature that is
"temperature switching"

What i had in mind was that, if i were to print in abs, switching to pla
would requirenme to stay in abs temp range to flush the abs at 240c. If
this option turned on, slic3r would flush at least an amount of filament
length, once it has done with the desired amount, then the head would
somewhat prime it again by a certain amount at the edge of the prime tower
while a command issued to adjust the temp before priming the new material.
That is if from high temp to lower temp material. Next is low to high temp
switching.
In this scenario, switching to higher temp would require the hotend to be
heated up to the abs/nylon printing temp while the head moves to the prime
tower position, then flushes the low temp material the pla slowly as the
hotend catches up to the abs/nylon printing temp, then another extra
priming line at the edge of the prime tower there.

For now, i am certain that this slic3r release is a huge leap towards a
better slicer software. Several thing that i want to point out is that:
1. printing the skirt, slic3r would use extruder 2 to print a bit of it as
well after all e1 tool has finished on that layer
2. I dont understand whats the usage of clip multi part object yet as it is
so vague
3. Skirt colouring can be confusing although this is nothing critical, my
test currently on slic3r parameter, e1 as blue, e2 as green. On preview it
shows as green portraying that it will be printed by using e2, if changing
e2 to different colour, the skirt will still be shown as green in the
preview
4. Having a fixed flush length is a good idea but actually can be bad, when
switching from black to yellow, it would require to flush even more to get
good yellow colour but not yellow to black. In here, i would waste more
blacks since yellow to black might require 15mm, while black to yellow is
30mm. If this were to be implemented, i suggest to put under per filament
setting
5. On the filament section i noticed filament type, might be too early to
ask, but what is this exactly?
6. I want to know about the T99 gcode issued there, care to explain that?

I guess thats all listed above for now. Right now, i tested the latest
slic3r version produced gcode in simplify 3d gcode animation, so far
nothing out of the ordinary. I cannot test it in real environment yet as i
am still waiting my multi material upgrade to arrive, to be precise, i
ordered mk2s upgrade kit and mm kit. Should be arriving soon but i dont
know if mm kit is included or not yet with my mk2s upgrade kit. The only
thing im able to test is on a custom built machine with single nozzle multi
feeder setup based on prusa archetype, but more tests to be done on it as
well.

Hope that the "temperature switching" can solve problems in mixing material
requiring different printing temp.

Hope all the info above i gave are useful.

Just a personal note, simplify3d are somewhat doesnt have any updates
anymore and to be frank, it is nowhere simple.

bubnikv, 2017-06-03:
> I would like to propose another set of feature that is "temperature switching"

Currently the Slic3r just switches extruder temperature together witht he filament for the Prusa MultiMaterial setup and it seems to be working for the PLA/PVA combination. We need to experiment with other material combinations in the future, ABS/PVA combination may need a better logic similar to what you have described.

Please note hat not many material combinations will stick together, so this may be a theoretical discussion as of now.

> What i had in mind was that, if i were to print in abs, switching to pla
would requirenme to stay in abs temp range to flush the abs at 240c. If
this option turned on, slic3r would flush at least an amount of filament
length, once it has done with the desired amount, then the head would
somewhat prime it again by a certain amount at the edge of the prime tower
while a command issued to adjust the temp before priming the new material.
That is if from high temp to lower temp material.

This is how the wipe tower currently works.

> Next is low to high temp switching.
In this scenario, switching to higher temp would require the hotend to be
heated up to the abs/nylon printing temp while the head moves to the prime
tower position, then flushes the low temp material the pla slowly as the
hotend catches up to the abs/nylon printing temp, then another extra
priming line at the edge of the prime tower there.

Currently when switching to a higher temperature filament, the Slic3r just primes the new filament while heating up. This may not be ideal for some materials.

> 1. printing the skirt, slic3r would use extruder 2 to print a bit of it as
well after all e1 tool has finished on that layer

I understand what you mean. You want to check all the filaments print before printing the 1st layer. We have such a function in the post processor, it primes all the filaments into the wipe tower, but this function has not been ported into Slic3r yet. To the priming on the skirt at the 1st layer, one has to weight the cost of extruder / filament switching and whether there will be space at the wipe tower at the 1st layer for two exchanges per filament.

> 2. I dont understand whats the usage of clip multi part object yet as it is
so vague

If you have modeled a sheep, then you modeled the eyes without substracting them from the sheep head, you may use this "clip" function in Slic3r to the boolean subtraction. Then the order dictates, that you subtract the eyes from the head and not vice versa.

> 3. Skirt colouring can be confusing

It has not been implemented yet.

> 4. Having a fixed flush length is a good idea but actually can be bad, when
switching from black to yellow, it would require to flush even more to get
good yellow colour but not yellow to black.

We are thinking of providing a filament table dictating the necessary purge per filament combination. Having the table, one can even optimize the order of filament exchanges to minimize material consumption (makes sense for 3 and more filaments). It is very time consuming though to collect the filament data.

> 5. On the filament section i noticed filament type, might be too early to
ask, but what is this exactly?

That configures the filament switching sequence for the Prusa Multi Material. The filament switching sequence is quite special to each filament type and it is required to form a nice tip at the end of the pulled out filament.

> 6. I want to know about the T99 gcode issued there, care to explain that?

That is old, it should be T[initial_tool] now. The T99 used to be a tag for our post processor.

sarf2k4, 2017-06-03:
It comes to my mind that the current slic3r has its own toolchange retract value. I don't know the specific value is but is there any particular specs for the built-in toolchange retract on the current slic3r?

I noticed that I didn't put anything in tool-change g-code and retraction when tool is disable are set to 0, the printer would retract about 10mm to initiate the exchange. 

I briefly tested it on my custom machine with multi-colour setup

sarf2k4, 2017-06-03:
I also wish that there is another field or change the existing "retraction when tool is disabled" behavior regarding the toolchange section

Following were extracted from the g-code produced in slic3r 1.35.1
```
G1 E-4.00000 F2100.00000
G1 E-2.00000 F2100.00000
G1 X127.479 Y134.527 F7200.000
G1 E6.00000 F2100.00000
;--------------------
; CP TOOLCHANGE START
; toolchange #0
; material : DEFAULT (PLA)
;--------------------
M220 S100
M907 E750
; CP TOOLCHANGE UNLOAD
G1 X128.229 Y134.627 F4000
G1 X186.229 E2.7608
G1 Y135.227 E0.0145 F4600
G1 X127.729 E2.8471
G1 X128.229 Y135.827 E0.0145 F5200
G1 X185.729 E3.0024
G1 E-15.0000 F5000
G1 E-50.0000 F5400
G1 E-15.0000 F3000
G1 E-12.0000 F2000
```

At the upper portion of the g-code snippet, these are the values from "retraction when tool is disabled" section, which is this:
```
G1 E-4.00000 F2100.00000
G1 E-2.00000 F2100.00000
G1 X127.479 Y134.527 F7200.000
G1 E6.00000 F2100.00000
```

The problem is that the retraction was done before the toolchange script really happens. The length originally 6mm, but split apart to -4 and -2, if it can be changed how it works like the behavior below.

The next part of the of the snippet from the above which is this section:
```
;--------------------
; CP TOOLCHANGE START
; toolchange #0
; material : DEFAULT (PLA)
;--------------------
M220 S100
M907 E750
; CP TOOLCHANGE UNLOAD
G1 X128.229 Y134.627 F4000
G1 X186.229 E2.7608
G1 Y135.227 E0.0145 F4600
G1 X127.729 E2.8471
G1 X128.229 Y135.827 E0.0145 F5200
G1 X185.729 E3.0024
G1 E-15.0000 F5000
G1 E-50.0000 F5400
G1 E-15.0000 F3000
G1 E-12.0000 F2000
```
The way slic3r handles, the toolchange retraction were done based on what was hardcoded. I'm pretty sure the dev team are going to improve further on this.

bubnikv, 2017-07-07:
> It comes to my mind that the current slic3r has its own toolchange retract value. I don't know the specific value is but is there any particular specs for the built-in toolchange retract on the current slic3r? (does it shape the filament inside the hot end for smooth transition or something else?)

The filament switching sequence is hardcoded and fine tuned for the MK2 Multi-Material setup. The switching sequence is there to shape the tip  of the pulled out filament.

> I noticed that I didn't put anything in tool-change g-code and retraction when tool is disable are set to 0, the printer would retract about 10mm to initiate the exchange.

Yes, there are some retractions hard coded into the wipe tower tool change sequence.

> The problem is that the retraction was done before the toolchange script really happens. The length originally 6mm, but split apart to -4 and -2, if it can be changed how it works like the behavior below.

The current wipe tower toolchange sequence is hardly compatible with any user provided tool change G-code.

sarf2k4, 2017-07-08:
What about adding a checkmark to have "prusa filament shaping retraction"? then this option is a macro to shape the filament to retract out of the hotend, then that "retract upon toolchange" value will be inserted after the filament shaping.

It's understandable to use hardcoded settings because to rectify certain issues immediately XD

bubnikv, 2017-07-10:
We will likely make the filament exchange procedure configurable through
some table, but right now there are some more pressing issues to be
implemented.

boelle, 2017-08-27:
i know this might be pushing it... 

but since the prusa version is the only one that is active devloped and with frequent releases i wonder if there could be a prime tower that supports more than 2 extruders pr layer?

its for us crazy people that like to experiment with single nozzle color mixing hotends where we define virtual extruders in the firmware and just ask for a toolhead to change the color.

repetier supports 16 virtual extruders and duet up to 400... marlin i dont know since i have abandoned it... to many faults etc etc...

and in theory we could have 3-4 colours pr layer

could it be done with a dynamic sized primetower or maybe several?

sarf2k4, 2017-08-27:
@boelle , I tried my MM upgrade on the latest prusa slicer release with 4 extruder and the software able to handle it better than s3d. The only limitation is that it is currently only for Prusa mk2 MM units. It can be used on other printer but need post-processing. Unsure of more than 4 extruders

boelle, 2017-08-27:
hmm... i will give it a go

for hotends like the diamond hotend, you use 5 filaments at the same time and in firmware you just tell it how much of each to push in to the nozzle so its quite different than the MK2

boelle, 2017-08-27:
i could have used post procession if i was a programmer :-/

sarf2k4, 2017-08-27:
Yeah, diamond hotends kind of simpler, but a filament needed to somewhat "plug" the open end so it wont backflow (correct me if i'm wrong, i never tried diamond type).

The ordeal with single hotend style with multi extruder is that the filament will expand during retraction, furthermore, stringing will occur. Current prusa slic3r has the retraction for tool change hardcoded, other printer unable to use it without post processing.

yes, i would too. I'm not a great programmer myself =\

boelle, 2017-08-27:
with a diamond you dont do retract

you just change ratio on the 5 filaments to change the color, but you need
to purge/prime until the color has changed

-- 
                 |||||
               @(~Ô^Ô~)@
-------------oOo---U---oOo-------------
|                                     |
|  Bo Herrmannsen                     |
|                                     |
|                                     |
| "blessed are the "cracked",         |
|  for its they who let in the light" |
|                         Ooo         |
|_________________ooO____(   )________|
                 (   )    ) /
                  \ (    (_/
                   \_)

sarf2k4, 2018-09-02:
Hi, 

It's been one year since the last post. Currently using 1.41.0 final release. I am not sure if I should add more into this topic/issues about my suggestion or open another issues about the following suggestion.

Wipe into infill is ok but the prime pillar still get generated anyway unless I turned the prime pillar off manually. The thing is, between "wipe into infill" on or off, both the prime pillar's size is just the same. If I'm not mistaken, "wipe into infill" is to reduce the required material to be purged as infill. But with wipe tower remains the same size, it is just the same as not having "wipe into infill". I strongly think that there should be another optional feature of "adaptive wipe tower size".

For example, from T1 to T2 requires 50mm to be purged in order to change colour properly. The current wipe tower (correct me if I'm wrong here) will scale based on that 50mm purge amount needed. With "adaptive wipe tower size" that must be paired with the "wipe into infill", for example the infill only has about 30mm of filament purged but requires an extra 20mm purging in order to have a good colour transition. Here is where the wipe tower comes in, it will be the extra 20mm purged filament there and the wipe tower will be smaller.

I'm not sure if the current version of slic3rpe already have that adaptive wipe tower paired with "wipe into infill" or not. It is hard to compare to be precise.

sarf2k4, 2018-09-02:
I'm sorry, it was my bad that I didn't export and compare the previews directly. I guess I should close this since this feature has been achieved

closed 2018-09-02 by sarf2k4