prusa3d/PrusaSlicer #151: Infill extrusion width affected by variable layer heights

patrikrosen, 2017-02-22:
### Version
1.33.8-prusa3d-win64
Print settings for the gMax printer from gCreate (can be found here: http://forum.gcreate.com/downloads.php?cat=9&sid=284f3732af05e99efe747a2b430e1b3d)

This does NOT happen when I switch to the regular prusa MK2 profiles.

### Operating system type + version
Windows 10

### Behavior
When using the variable layer feature the infill extrusion width seems to increase when the layer height decreases. 

[image]
[image]
[image]

bubnikv, 2017-02-24:
Patrik, 

would you please attach your slic3r profile? Go to menu file->export config.

Most likely the effect you are seeing is caused by what the original author calls "sane defaults" for the extrusion width. If you left any of the extrusion width at zero, this will kick in. If you want to create a profile for your other machine, I recommend to start with the MK2 profiles and to modify them, instead o starting from the "default" profile.

bubnikv, 2017-03-23:
@patrikrosen Any news on this issue?

bubnikv, 2017-06-06:
Probably the "sane defaults" effect. I did not receive any follow up information, closing.

closed 2017-06-06 by bubnikv