prusa3d/PrusaSlicer #152: [Request] Go Direct to the preview screen after slicing

Bubolz, 2017-02-23:
### Version
1.33.8

### Operating system type + version
Win7 64

### Behavior
It would be more practical  If the preview screen of the sliced object was displayed immediately after clicking the "Slice Now" button.

cerdeira, 2017-02-27:
Following this request maybe also open 3D view automaticaly when loading a new model and when changing print, filament and printer settings.

bubnikv, 2017-10-03:
A similar request: https://github.com/prusa3d/Slic3r/issues/436

bubnikv, 2018-12-11:
Implemented with 09c539a24233a0ff41b54720870fec3725f130d7

Instead of the 3D / Preview tabs, there are now two overlay buttons, the "Preview" button now triggers reslice before switching to the Preview.

It will be part of the upcoming 1.42 alpha. Closing.

closed 2018-12-11 by bubnikv