prusa3d/PrusaSlicer #153: [Request] Advanced seam hiding feature

Bubolz, 2017-02-23:
### Version
1.33.8

### Operating system type + version
Win7 64

### Behavior

Would be great if it were implemented features to hide the seam within the model as in the Kisslicer, putting the beginning and end of each line within the object. 

I believe that that would have a great impact on the final print quality with regard to the finishing of seam and blobs reduction.

[image]

bubnikv, 2017-02-23:
Slic3r already does it.
https://github.com/prusa3d/Slic3r/blob/master/xs/src/libslic3r/GCode.cpp
lines 767-799

Bubolz, 2017-02-23:
Wow! Slicer is evolving very fast! 👍 :-)

bubnikv, 2017-02-23:
As far as I am known, this has already been there and Slic3r is known for hiding the seam very well.

closed 2017-02-23 by bubnikv

Bubolz, 2017-02-23:
This feature already in version 1.33.8? Is there a configuration parameter for it, or is it a native resource?

bubnikv, 2017-02-23:
As far as I am known, this has always been in Slic3r.

The way it works is a bit different from kisslicer. Slic3r does the following:

It always clips the perimeter by 15% of the nozzle diameter. This means there will be a 15% gap between the start and the end of the perimeter.

For the external loops, Slic3r moves inside the perimeter by the length of a nozzle diameter before traveling, thus smearing the seam inside.

In my opinion this works very well, at least for the Prusa Research printers.