prusa3d/PrusaSlicer #154: [Request] Advanced Fan Control by Path Type

Bubolz, 2017-02-23:
### Version
1.33.8

### Operating system type + version
Win7 64

### Behavior
If the cooling fan control of the part is associated with the type of path (as in Kisslicer), There would be better control of the quality and strength of printed parts.

For example: During the printing of an ABS piece the fan could be configured to be turned on 30% only at the external perimeter and would turn off during infill printing.

bubnikv, 2017-07-07:
I checked with Kisslicer, there is really only a differentiation between the "all outer" and "all inner". From the popup help it is not clear to me, which features fall into which category. Are just two categories what you want?

Bubolz, 2017-07-08:
I think two categories would be enough. An independent control for Loops (internal and external) and another independent control for Infill (internal and solid).

I've attached a sample image of how the setting could be:
[image]

bubnikv, 2017-07-10:
I wonder why would you want to decrease the fan speed for infills?

Bubolz, 2017-07-10:
To improve the quality of ABS printing!

Printing ABS with the fan On allows very smooth prints, but unfortunately the fan action also causes poor adhesion between layers, causing cracking and delamination.

If during printing infill (and better thinking) the inner perimeter is printed with the fan off and on only while printing the outer perimeter we would have the best of two world, a very soft and strong impression of ABS (It's more or less this which Kisslicer can do with independent controls).

Below another image with a more simple suggestion of how it could be.

[image]

rennne, 2019-02-25:
"Keep fan always on only for external perimeter" would help for ABS but not for PLA, where I would also benefit from a bit slower fan for the infill and inner perimeters regarding layer adhesion

jschuh, 2019-09-15:
Sorry, I only just now saw this but I added issue #2776 to support an independent fan speed for external perimeters and have a PR for my change here: https://github.com/prusa3d/PrusaSlicer/pull/2781

closed 2026-09-29 by prusaslicer-bot[bot]