prusa3d/PrusaSlicer #155: Mac build 1.33.8 seems to have broken support pillars...

macdarren, 2017-02-24:
I no longer get the traditional pillar for support when pillars is choose as the support method.

Instead I get something that looks almost identical to the Hex support.
Other support types seem to work fine.

Also the loops / no loops for the support interface seems to have no affect..now it never makes loops.

bubnikv, 2017-02-24:
The pillars were not implemented with the new supports. I should at least
make a different default I suppose. The interface loops shall work if you
have at least a single interface layer.

The pillars never quite worked, they did not ensure their minimal cross
section, so more often than not some of the pillars were not build able.

bubnikv, 2017-02-24:
I just tested the interface loops on the following model:

[Support test 8.zip](https://github.com/prusa3d/Slic3r/files/798650/Support.test.8.zip)

There are indeed interface loops generated. They are generated below the overhangs only.
[image]

When enabling supports below the overhangs, the interface loops will mostly not be generated.
[image]

When I ported the interface loops from Perl to C++, I wondered about their purpose. I think the Alessandro's intent was to make the support interface mostly perpendicular to overhangs. So the interface loops are really placed around the support edges only, and only if the support edges are really supporting an overhang.

Do you have an example, where this does not work, or where you think the old algorithm worked better?

macdarren, 2017-02-24:
here is an object link....

https://www.dropbox.com/s/9tz738dni70znz8/LPsideglassholder9.stl?dl=0

Another odd thing was if I select to build support only on the bed I still get support on the model but less of it...and a strange gap in the interface layer where the support should not be but is.

Strange now the loops seem to work, I will see if I have the slightly older revision of the model that seemed to have the problem...My assumption was always that the loops which form along the edges were there to make support removal easier allowing you to pry it out.

Got to say however no recent mainline version of slicer does supports as well as the Prusa3d versions, so thanks for making this happen.

darknz, 2017-03-15:
I have noticed the same problem in the PC version, 'Pillars' generates honeycomb.
I agree that pillars had some problems, however the current options produce supports that i have found extremely hard to remove from small intricate models (think 20mm role playing miniatures). I would like to see something similar to pillars continued to be implemented, my current options are only to go back to using an earlier version of Slic3r.

bubnikv, 2017-03-15:
> I agree that pillars had some problems, however the current options produce supports that i have found extremely hard to remove from small intricate models (think 20mm role playing miniatures).

Would you please provide such models? We will print them here. Please provide not only the STLs, but also your config file. Are you printing on the Prusa i3 MK2?

bubnikv, 2018-03-14:
I have removed the unsupported "pillars" support pattern from Slic3r PE with https://github.com/prusa3d/Slic3r/commit/61ae78432ce92adb47ea6a7eb3aca65bb353459e
Closing.

closed 2018-03-14 by bubnikv