prusa3d/PrusaSlicer #156: Define wall thickness in mm

CReimer, 2017-02-24:
Please make it possible to define wall thicknesses and thickness of top and bottom layers in millimeter. That's one of the features from Cura I miss most in Slic3r.

Sometimes I want to print something with 0.15 or even less instead of 0.2 and I always have to recalculate the solid layer count to be back at around 1.2mm thickness.

bubnikv, 2017-02-24:
It is even more important for the variable layer thickness feature.

JohnnyDeer, 2019-10-11:
Hi, thank you for idea. We decided to not implemented it in PrusaSlicer.  I will close this issue in framework of improving the clarity of the issues list. Thank you for understanding.

Kachidoki2807, 2019-10-14:
Huh? Any argument to not implement it?
I feel like I'm dealing with an IT service paid on the number of tickets closed...

This Feature Request must remain open at least as a backlog or wishlist, unless you have a valid reason. If it is, please be so kind to share it with us.

bubnikv, 2020-02-05:
> I feel like I'm dealing with an IT service paid on the number of tickets closed...

I feel like I deserve a chocolate bar for every issue closed.

Anyway, implemented with 495a71ed00a295847edaec879d8cc7006a810636
It will be part of the upcoming PrusaSlicer 2.2.0-alpha4 release.

The two new config keys top_solid_min_thickness / bottom_solid_min_thickness define a minimum vertical shell thickness. The top shell thickness is calculated as a maximum of (sum over top_solid_layers * their respective layer heights) and top_solid_min_thickness, the bottom shell thickness is calculated similarly.
    
The results of the formula above are shown at the Print parameter page below the two new values to hint the user about the interaction of the old versus new config values.
    
top_solid_min_thickness has no meaning if top_solid_layers is zero,
bottom_solid_min_thickness has no meaning if bottom_solid_layers is zero.

Closing.

closed 2020-02-05 by bubnikv

Kachidoki2807, 2020-02-06:
> I feel like I deserve a chocolate bar for every issue closed.
Which is your favorite? I'll be glad to send you some. Really.
Nothing angry on my side, but there was a period during which several legitimate and still valid issues were closed without further explanation other than: it will not be implemented.

I'll give a try to this new improvement, especially in combination with the new auto layer height that I use more and more. Thanks for the implementation effort.

aminerhayem123, 2023-04-06:
can you guys help me with calculating the lowest thin wall 
thickness