prusa3d/PrusaSlicer #157: Handle single extruder prints on dual extruder machines correctly

freiser77, 2017-02-25:
### Version
1.33.8-prusa3d

### Operating system type + version
macOS Sierra

### Behavior
When printing with one filament only on an dual extruder printer like the FFCP 2016, you have to print PLA on the left (second) extruder, because this is the one with the fan.
However the temperature setting for the heated bed is being taken from the right (first) extruder without filament.

The following would be nice to have:
* A "None" filament causing Slic3r to retrieve all values which are currently programmed fix to get from the first extruder, to be retrieved from the second one (if the first has set the None-filament) and to switch off the heater of the extruder set to it.
* The possibility to name the extruders (instead of Extruder 1 and Extruder 2). This names should be displayed next to the filament selection, so it is more clear for which extruder you select the filament.

Yes, this is a new feature request

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2017-03-03:
> A "None" filament causing Slic3r to retrieve all values which are currently programmed fix to get from the first extruder, to be retrieved from the second one (if the first has set the None-filament) and to switch off the heater of the extruder set to it.

This could be implemented, but for now (and maybe even for the future) it may be more useful from the user perspective to have three Printer settings for your machine: Dual extruder printer settings, 1st extruder only printer settings and 2nd extruder only printer settings. The 2nd extruder only would differ from the 1st extruder only settings by the initial and final G-code sequence. At least this is how I would handle the situation and it would make it easier for me to switch between the extruders than tinkering with the extruder assignment every time I decide to go from one color to the other.

DrLex0, 2017-03-03:
If the extruder for the object is manually set to 2 (= the left extruder), Slic3r **will** use the correct bed temperature. However, you must do it exactly this way: click the object, click ‘Settings…’, select the STL itself in the list (the yellow brick icon), then select extruder 2. You _will_ forget to do this often. I have added a beep command with a message to my start G-code for the left extruder so I can at least abort the print and grudgingly fix it and start over.

The approach proposed by @bubnikv cannot avoid the need to perform this little ritual. If you don't perform it, Slic3r _always_ uses the bed temperature of the ‘default extruder’, which seems to be hard-coded to ‘1’, to fill in [first_layer_bed_temperature] in G-code templates. That is why I created the now mothballed [issue 3398](https://github.com/alexrj/Slic3r/issues/3398) in the master branch. Fix this (i.e. allow to modify the default extruder in printer profiles), and this approach will work nicely.

@freiser77: you may want to take a look at [this](http://www.thingiverse.com/thing:1675208) if you want to be freed from the obligation to print PLA with the left extruder. You may also want to take a look [here](https://www.dr-lex.be/info-stuff/print3d-ffcp.html) if you want G-code to populate the right, left, and dual printer profiles as @bubnikv explained.

freiser77, 2017-03-04:
@DrLex0: Thanks for your help, that's exactly what I've looked for. However I second your #3398.

freiser77, 2017-03-04:
@DrLex0: The trick with setting the [first_layer_bed_temperature] correctly by setting the default extruder on the part does not function. The variable is set to 0 as this is the value of the None filament in the first extruder. The PLA config in the second extruder holds 55 degrees as value.

DrLex0, 2017-03-04:
Have you set the extruder for the .stl file inside the object? Setting it for the whole object has no effect. You need to do it exactly as I described above.
[image]

freiser77, 2017-03-04:
Yes, I've set it exactly that way.

DrLex0, 2017-03-04:
OK, then the reason why it has always worked for me, is that I always configure the filaments for both extruders to the same one I will be printing with when doing a single-extrusion print. I do this anyway because it is unclear which of the two drop-down menus applies to what extruder, so setting them both to the same filament avoids any mistakes. It is only for a dualstrusion that I need to remember that the top one in the interface is the right extruder, and the bottom one is the left.

bubnikv, 2017-03-04:
I proposed to create a single extruder profile and to switch the tool in
the start g-code sequence. I believe at least for Marlin this should be
possible. Or am I wrong? Just put a line T2 at the very start and all the
commands will relate to extruder nr. 2.

DrLex0, 2017-03-04:
@bubnikv: OK, that seems a more sensible approach and it avoids all these issues. Still, when doing a dual extrusion, it is unclear what temperature will be used for the bed, so we probably still need [#3398](https://github.com/alexrj/Slic3r/issues/3398).

freiser77, 2017-03-04:
@bubnikv: When I configure the printer as you suggested, the switch from first_layer_temperature to normal layer temperature is done with
```
M104 S200 T0 ; set temperature
```

This is obviously wrong, as it heats the first extruder. But the second one should be heated.
Is it possible to configure Slic3r to drop the T-parameter? The G-code flavor is Sailfish (Makerbot).

DrLex0, 2017-03-04:
Aha, that is a problem I almost forgot. For this, setting the extruder to 2 for the object is an effective but annoying workaround. You will very often forget to do this, and will have to abort many prints after noticing the wrong extruder starts heating and dripping ooze after the first layer.
Dropping the T argument if there is only one extruder configured, is indeed the simplest solution.

bubnikv, 2017-03-04:
You can set a post processing script at the print settings. It would be
cool to have a post processing script field at the printer settings for the
feature discussed.

DrLex0, 2017-03-04:
Yes, I have just added this to the post-processing script that I need anyway to make an X3G file out of the G-code. But the point is that this should not be necessary.
Adding the T0 argument to the M104 command is plain wrong if there is no reason to assume that the active extruder is T0. Without a T argument, the M104 command applies to the active extruder (according to the [RepRapWiki](http://reprap.org/wiki/G-code#M104:_Set_Extruder_Temperature), and I verified that at least for my printer, it indeed works like that).
As @freiser77 said, if only one extruder is specified in the printer profile, Slic3r should insert `M104 S[temperature]` when starting the second layer, without an extra T argument.

bubnikv, 2017-06-02:
> Is it possible to configure Slic3r to drop the T-parameter? The G-code flavor is Sailfish (Makerbot).

This is what Slic3r already does for all G-code flavors but MakerWare and Sailfish. Do you have any reference pointing to the handling of the M104/M109 codes by Sailfish? I don't want to make a change if I don't know the reasons for the current implementation.

bubnikv, 2017-06-02:
Anyway, it is true that Slic3r uses always the bed temperature settings of the first filament. This problem applies to the following parameters as well:

bed_temperature
bridge_fan_speed
cooling
disable_fan_first_layers
fan_always_on
fan_below_layer_time
max_fan_speed
min_fan_speed
min_print_speed
slowdown_below_layer_time

bubnikv, 2017-06-21:
I have modified Slic3r https://github.com/prusa3d/Slic3r/commit/c1146e298b8ff96bd029bad494627c5a6b547f56 to use the bed_temperature and first_layer_bed_temperature of the active extruder.

It remains to somehow handle the cooling parameters 

bridge_fan_speed
cooling
disable_fan_first_layers
fan_always_on
fan_below_layer_time
max_fan_speed
min_fan_speed
min_print_speed
slowdown_below_layer_time

in case of multi-material print. Some new rules are needed to blend the parameters of multiple materials into something meaningful. At Prusa Research, we currently have issues printing PLA together with PVA as PVA does not like the fan.

bubnikv, 2017-07-07:
Support of extruder specific cooling has been implemented with commits
https://github.com/prusa3d/Slic3r/commit/1158ce41df8f9219605b4efb0ce08cf4384d08e3
to
https://github.com/prusa3d/Slic3r/commit/0ad4e9d51f3171d68efea5249f1831b749ea97c8

The bed temp is taken from the 1st used extruder.

Is there anything missing to be implemented to fulfill this request? I am tempted to close this issue and if there is anything left to do, to ask you to open another issue to make it tractable.

closed 2017-10-03 by bubnikv