prusa3d/PrusaSlicer #158: custom G-Code miss handled in comments... ver 1.33.8

macdarren, 2017-02-27:
While doing some calibration runs and tweaking my start codes, I got few odd surprises when running some g-code, where the printer seemed to start a new print immediately after finishing a print job crashing into the original print.

What I found is that where Slic3r puts in comments at the end of the file with all the parameters (which is really nice) it also puts in the start and stop G-code...now this would not be a problem but the slicer sometimes does something odd...somehow the stop and start g-codes are not commented out so they start to run....

What seems to be the case is that on the line above where the G-code would go there is no carriage return...thus when my g-code which contains comments and carriage returns is inserted it is somehow not marked as a comment but treated as runable code...you can see this if you load the downloaded file into Repetier Host which is what I typically use.

This is not a huge big problem and the start and stop codes are placed in the g-code correctly just not in the comment section.... seems like maybe it is something that should get fixed at some point.  I actually think this is a problem has existed in earlier versions of Slic3r.  maybe it is a line ending issue between platforms or something not sure when exactly the "\n" get into the picture but I have tried to get them out with no success.

here is a code excerpt from the comments at the end of a slicer generated g-code file:

; retract_restart_extra_toolchange = 0
; retract_speed = 50
; skirt_distance = 10
; skirt_height = 1
; skirts = 1
; slowdown_below_layer_time = 40
; spiral_vase = 0
; standby_temperature_delta = -5
; start_gcode = ; *** CUSTOM START CODE ***
\n
\nG21 ; set units to millimeters
\nG90 ; use absolute coordinates
\nM82 ; use absolute distances for extrusion
\n
\nM106 S0 ;set fan speed to zero, to stop quick fan on
\nM107  ;fan off in case somehow it was left on
\n
\n
\nM104 S150 T0 ;set extruder temp for pre heat Ext 0 no wait
\nM104 S150 T1 ;set extruder temp for pre heat Ext 1 no wait
\n
\nM201 X500 Y500; adjust acceleration to slow
\nG28 ;home all axes gets everything in a known position
\n
\nM190 S[first_layer_bed_temperature] ;sets bed temp and waits
\n
\nG1 X-60 Y-120 Z100 F8000.0	;quick move to heating position
\n
\nM109 S[first_layer_temperature_0] T0 ;sets extruder temp and waits
\nM109 S[first_layer_temperature_1] T1 ;sets extruder temp and waits 
\n
\n;G1 X-60 Y-120 Z.4 F8000.0	;quick move down to priming position
\n;G92 E0.0	;set extruder to zero
\n;G1 E12.0 F600	;extrude a blob for inital prime
\n;G1 X60 E20.0 F600.0 ;extrude a line to get nice flow started
\n;G92 E2.0	;reset extruder to 2mm
\n;G1 F3000 E0 ;retract to reduce stringing and set E=0 on inital start
\n;G1 X0 F800 ;slow wipe back 
\n;G1 Z.1 F800  ;slow down stamp to break any strings
\n;G1 Z10 F8000 ;Quick jerk up before print starts
\n;T0
\n
\n
\nG1 X-60 Y-125 Z.4 F2000.0	;move down to priming position
\nG92 E0.0	;set extruder to zero
\nG1 E12.0 F600	;extrude a blob for inital prime
\nG1 X60 Y-125 E20.0 F600.0 ;extrude a line to get nice flow started
\n
\n
\n
\nG92 E0 ; reset extrusion distance
\nG1 E-5.00000 F3000.00000 ; retract
\n
\nG1 X0 F800 ;slow wipe back 
\nG1 Z.1 F800  ;slow down stamp to break any strings
\nG1 Z10 F12000 ;Quick jerk up before print starts
\nG1 E4.00000 F3000.00000 ; unretract
\n
\nM201 X2500 Y2500; adjust acceleration to Normal
\n
\n
\n; *** END CUSTOM START CODE ***
\n;
; temperature = 200
; threads = 5
; toolchange_gcode = 
; travel_speed = 200
; use_firmware_retraction = 0
; use_relative_e_distances = 0
; use_volumetric_e = 0
; vibration_limit = 0
; wipe = 0
; z_offset = 0
; dont_support_bridges = 1
; extrusion_width = 0.5
; first_layer_height = 0.25

bubnikv, 2017-02-28:
Would you please provide your config.ini? Which platform are you running
on? Are you running the official build, or did you compile yourself?
Thank you.

bubnikv, 2017-03-03:
> Update 2: I believe now this does NOT happen with 1.33.8 so I guess this whole post can be removed...not sure how to do that...
> Update: This maybe not always happen and I suspect I actually was using an older version of Slic3r...

Thanks for response. I am closing this issue. Please feel free to reopen if you find it happening again.

closed 2017-03-03 by bubnikv