prusa3d/PrusaSlicer #159: A lot of unneccessary retract movements

CReimer, 2017-02-28:
Version 1.33.8.r49.g9ea570ea

The GCode analyzer (gcode.ws) show a lot of retract movements. I think that shouldn't be like that.
But I cannot tell, if this is some kind of bug in gcode.ws or in Slic3r.

[bearing6.gcode.zip](https://github.com/prusa3d/Slic3r/files/805384/bearing6.gcode.zip)

bubnikv, 2017-02-28:
Please provide an stl and config, as the retracts are very much dependent on the retract settings.

Slic3r will retract on every travel longer than the value "retract_before_travel" configured at the printer->extruder page, with the exception of moving inside support islands or when moving inside infill islands if only_retract_when_crossing_perimeters is set.

I know Slic3r incorrectly (at least in my point of view) applies wipe movements on infills, supports and such, which does not make sense as long as the next movement does not cross a perimeter, but that is another issue.

CReimer, 2017-02-28:
[Slic3r-config.zip](https://github.com/prusa3d/Slic3r/files/806544/Slic3r-config.zip)
http://www.thingiverse.com/thing:53451 (bearing6.stl)

It is actually the "Wipe before retract" feature. This could be a misinterpretation of the GCode Analyzer.

bubnikv, 2017-02-28:
> It is actually the "Wipe before retract" feature.

Shall I rename the title to make it clear it is a "wipe before retract" issue?

Are you using the Prusa i3 MK2, or are you using our slic3r profiles, or did you enable the wipe yourself?

CReimer, 2017-02-28:
Yes, feel free to rename the title.

I'm not using a Prusa i3 MK2, it's actually a P3Steel printer.
The print settings are more or less the Prusa3D Slic3r profile for 0,20mm. I usually disable "Ensure vertical shell thickness" as it causes a lot of small movements, which are quite loud on at least my printer.
And I increase the "Minimum extrusion length" as I don't have such a fancy start gcode to clean the nozzle.
And in this case I set the "Fill density" to 0%

When or why I enabled "Wipe while retracting" I can't recollect. But I imagine that I enabled it because the description sounds intriguing.

bubnikv, 2020-12-11:
"Avoid crossing perimeters" has been reworked with 5f456ebbe78f3b994ea7ecc7de23c119e4705b61 (see the branch lh_avoid_crossing_perimeters).
The new "Avoid crossing perimeters" feature disables wipes when the straight line from the travel start point to the travel end point is fully inside an island.

The new algorithm will be available in PrusaSlicer-2.3.0-beta3. Please check the PrusaSlicer-2.3.0-beta3 change log for details.

closed 2026-09-29 by prusaslicer-bot[bot]