prusa3d/PrusaSlicer #161: G-Code contains negative feed rate

kbranch, 2017-03-01:
### Version
1.33.8-prusa3d-linux64

### Operating system type + version
64 bit Ubuntu 16.10

### Behavior
The generated g-code for the attached STL contains a feed rate of -681190 when sliced with the attached configuration (line 9603, "G1 F-681190").  My printer aborts the print and needs to be reset when it encounters this line.  It works fine if I replace it with the appropriate feed rate for that section (900).

#### STL/Config (.ZIP) where problem occurs
[negative-feed-rate.zip](https://github.com/prusa3d/Slic3r/files/809679/negative-feed-rate.zip)

bubnikv, 2017-03-03:
There were cases before, when Slic3r generated negative extrusions, but these were usually connected with the pressure advance feature, which I have verified you have disabled in your profile. 

Anyway, I see you have configured Slic3r in a way we usually don't.

At the Print Settings -> Advanced -> Extrusion width you left all the fields at zero, meaning "sane defaults", which are not always as sane as one would wish.

At the Print Settings -> Infill -> Advanced you have Only retract when crossing perimeters enabled.

Would you please try to pinpoint the combination of features, which cause the negative extrusions?

kbranch, 2017-03-04:
Aha, thank you for the insight.

"Only retract when crossing perimeters" did not seem to have an effect, although I didn't test exhaustively.

The negative feed rate does go away if I put a value in any one of "Default extrusion width", "Perimeters" or "External perimeters".  The trouble spot was an external perimeter, so that would seem to make sense.

I've always assumed that Slic3r knew what extrusion widths to pick better than I did, but I've never really looked too closely.  Am I losing anything by setting fixed values here?  For example, does Slic3r do something like adjust extrusion width to ensure dimensional accuracy for narrow sections if left to its own devices?

kbranch, 2017-03-16:
Need anything else from me?

This isn't a big issue for me since I can avoid it with settings, but I am curious if I'm missing out on anything by manually setting the extrusion width.

bubnikv, 2017-03-23:
> I've always assumed that Slic3r knew what extrusion widths to pick better than I did, but I've never really looked too closely. Am I losing anything by setting fixed values here? For example, does [...]

AFAIK you are mostly not losing anything. Only place where Slic3r varies the extrusion width on its own are the infill lines and this is questionable anyway.

kbranch, 2017-03-23:
Great, thanks!

bubnikv, 2017-07-19:
I have changed the "sane" default extrusion widths to something close to the Prusa MK2 values, dependent on the nozzle diameter only, not on the layer height. Closing this issue.
https://github.com/prusa3d/Slic3r/commit/ac672d9dc592dc94ea1efdfeecdb75efc0a19233

closed 2017-07-19 by bubnikv