prusa3d/PrusaSlicer #162: Replace perls Wx dependency with C++ code

freiser77, 2017-03-01:
As the Wx library from perl is creating constant headache (for example when trying to compile from source): Please replace perls Wx dependency with C++ frontend logic.

bubnikv, 2017-03-01:
That's a great idea. Please provide a pull request.

Now, jokes aside. It is quite a lot of work and there is no immediate benefit to the users of the official builds. The Perl is stable and the Perl wxWidgets are stable as well. There are currently more pressing issues, which have effect on the print quality, print time, usability etc.

freiser77, 2017-03-01:
As I am a professional developer, I know that it is a lot of work.
And I would like to participate in Slic3r development. But with Perl driving me crazy and failing the builds, I am not able. So I think it is not a matter of providing more features to the users of Slic3r, but of lowering the entry barrier for developers wanting to assist in Slic3r development.

bubnikv, 2017-03-01:
@alexrj just proposed couple of toppics for Google Summer of Code.
https://github.com/alexrj/Slic3r/issues/3730

One of these topics is the C++ port of the wxWidgets UI. I know @alexrj and @lordofhyphens did some initial experiments confirming, that you can merry the Perl and C++ wxWidgets together, allowing to port the UI piece by piece.

I think you develop on OSX. I will be more than happy to help you getting your development environment up and running. Me personally, I ended up compiling the complete Perl and all dependencies using Visual Studio, so I am able to debug comfortably. It took me a working week or so to do that and it takes now about 4 hours to repeat the process. You should be able to do the same on OSX in about the same time frame (4 hours with my assistance).

I am actually slowly reducing the number of Perl dependencies. For example, the recent commits removed the Perl Expat XML parser dependencies, last fall I removed the Perl Math::PlanarCurve or whatever it was (I happily forgot the name since it is gone).

I think to lower the entry barrier, it would be sufficient to document the setup process well for now. I agree it would be better to have everything in C++, but it is not my priority.

freiser77, 2017-03-01:
@bubnikv : If you would help me getting my development environment (mainly Perl) up and running, I would be happy to help with C++ migration. I will try another build this afternoon and report about the (eventual) failures occurring.

freiser77, 2017-03-01:
@bubnikv: When I try to compile Slic3r on my iMac, it includes this file: /System/Library/Perl/5.18/darwin-thread-multi-2level/CORE/perl.h. The perl version used to compile is 5.24 (retrieved with perlbrew). Can you tell me, how you got it working on your build machine for the dmg images?

lordofhyphens, 2017-03-01:
https://github.com/alexrj/Slic3r/issues/3717

I have been working on one of the small, but technically interesting bits.
I feel the rest of the UI is a straightforward port.

My primary dev environment is Linux.

bubnikv, 2017-03-01:
My best guess is that you are not running perl from the perlbrew. What does
it tell you if you execute
perl -V
?

freiser77, 2017-03-01:
Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=darwin, osvers=16.1.0, archname=darwin-thread-multi-2level
    uname='darwin sierravm.local 16.1.0 darwin kernel version 16.1.0: thu oct 13 21:26:57 pdt 2016; root:xnu-3789.21.3~60release_x86_64 x86_64 '
    config_args='-des -Dprefix=/usr/local/Cellar/perl/5.24.0_1 -Dprivlib=/usr/local/Cellar/perl/5.24.0_1/lib/perl5/5.24.0 -Dsitelib=/usr/local/Cellar/perl/5.24.0_1/lib/perl5/site_perl/5.24.0 -Dotherlibdirs=/usr/local/lib/perl5/site_perl/5.24.0 -Dperlpath=/usr/local/opt/perl/bin/perl -Dstartperl=#!/usr/local/opt/perl/bin/perl -Dman1dir=/usr/local/Cellar/perl/5.24.0_1/share/man/man1 -Dman3dir=/usr/local/Cellar/perl/5.24.0_1/share/man/man3 -Duseshrplib -Duselargefiles -Dusethreads'
    hint=recommended, useposix=true, d_sigaction=define
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=define, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cc', ccflags ='-fno-common -DPERL_DARWIN -mmacosx-version-min=10.12 -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include -DPERL_USE_SAFE_PUTENV',
    optimize='-O3',
    cppflags='-fno-common -DPERL_DARWIN -mmacosx-version-min=10.12 -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include'
    ccversion='', gccversion='4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.42.1)', gccosandvers=''
    intsize=4, longsize=8, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=define, longlongsize=8, d_longdbl=define, longdblsize=16, longdblkind=3
    ivtype='long', ivsize=8, nvtype='double', nvsize=8, Off_t='off_t', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='cc', ldflags =' -mmacosx-version-min=10.12 -fstack-protector-strong -L/usr/local/lib'
    libpth=/usr/local/lib /usr/include/libxml2 /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../lib/clang/8.0.0/lib /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/lib /usr/lib
    libs=-lpthread -ldbm -ldl -lm -lutil -lc
    perllibs=-lpthread -ldl -lm -lutil -lc
    libc=, so=dylib, useshrplib=true, libperl=libperl.dylib
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_dlopen.xs, dlext=bundle, d_dlsymun=undef, ccdlflags=' '
    cccdlflags=' ', lddlflags=' -mmacosx-version-min=10.12 -bundle -undefined dynamic_lookup -L/usr/local/lib -fstack-protector-strong'

Characteristics of this binary (from libperl): 
  Compile-time options: HAS_TIMES MULTIPLICITY PERLIO_LAYERS
                        PERL_COPY_ON_WRITE PERL_DONT_CREATE_GVSV
                        PERL_HASH_FUNC_ONE_AT_A_TIME_HARD
                        PERL_IMPLICIT_CONTEXT PERL_MALLOC_WRAP
                        PERL_PRESERVE_IVUV PERL_USE_SAFE_PUTENV
                        USE_64_BIT_ALL USE_64_BIT_INT USE_ITHREADS
                        USE_LARGE_FILES USE_LOCALE USE_LOCALE_COLLATE
                        USE_LOCALE_CTYPE USE_LOCALE_NUMERIC USE_LOCALE_TIME
                        USE_PERLIO USE_PERL_ATOF USE_REENTRANT_API
  Built under darwin
  Compiled at Nov 12 2016 16:10:25
  %ENV:
    PERL5LIB="/Volumes/LaCie/florian/perl5/lib/perl5"
    PERL_LOCAL_LIB_ROOT="/Volumes/LaCie/florian/perl5"
    PERL_MB_OPT="--install_base "/Volumes/LaCie/florian/perl5""
    PERL_MM_OPT="INSTALL_BASE=/Volumes/LaCie/florian/perl5"
  @INC:
    /Volumes/LaCie/florian/perl5/lib/perl5/5.24.0/darwin-thread-multi-2level
    /Volumes/LaCie/florian/perl5/lib/perl5/5.24.0
    /Volumes/LaCie/florian/perl5/lib/perl5/darwin-thread-multi-2level
    /Volumes/LaCie/florian/perl5/lib/perl5
    /usr/local/Cellar/perl/5.24.0_1/lib/perl5/site_perl/5.24.0/darwin-thread-multi-2level
    /usr/local/Cellar/perl/5.24.0_1/lib/perl5/site_perl/5.24.0
    /usr/local/Cellar/perl/5.24.0_1/lib/perl5/5.24.0/darwin-thread-multi-2level
    /usr/local/Cellar/perl/5.24.0_1/lib/perl5/5.24.0
    /usr/local/lib/perl5/site_perl/5.24.0
    .

bubnikv, 2017-03-01:
You seem to have the PERL5LIB set correctly, but you are executing the
system perl. You need to update your PATH to point to the perlbrew's perl
first.

bubnikv, 2017-03-01:
You don't have the PERLBREW_XXX environment variables shown in your printout and your @INC differs significantly. If you check mine, they all point to the perlbrew directory.

bubnikv, 2017-03-01:
Did you add 
source ~/perlbrew/etc/bashrc

at the end of yours 
~/.bash_profile
?

freiser77, 2017-03-01:
I've reinstall perl with perlbrew and reactivated perlbrew. Now perl -V shows the following

```
Summary of my perl5 (revision 5 version 25 subversion 10) configuration:
   
  Platform:
    osname=darwin
    osvers=16.4.0
    archname=darwin-2level
    uname='darwin florians-imac.local 16.4.0 darwin kernel version 16.4.0: thu dec 22 22:53:21 pst 2016; root:xnu-3789.41.3~3release_x86_64 x86_64 '
    config_args='-de -Dprefix=/Volumes/LaCie/florian/perl5/perlbrew/perls/perl-5.25.10 -Dusedevel -Aeval:scriptdir=/Volumes/LaCie/florian/perl5/perlbrew/perls/perl-5.25.10/bin'
    hint=recommended
    useposix=true
    d_sigaction=define
    useithreads=undef
    usemultiplicity=undef
    use64bitint=define
    use64bitall=define
    uselongdouble=undef
    usemymalloc=n
    bincompat5005=undef
  Compiler:
    cc='cc'
    ccflags ='-fno-common -DPERL_DARWIN -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include'
    optimize='-O3'
    cppflags='-fno-common -DPERL_DARWIN -fno-strict-aliasing -pipe -fstack-protector-strong -I/usr/local/include'
    ccversion=''
    gccversion='4.2.1 Compatible Apple LLVM 8.0.0 (clang-800.0.42.1)'
    gccosandvers=''
    intsize=4
    longsize=8
    ptrsize=8
    doublesize=8
    byteorder=12345678
    doublekind=3
    d_longlong=define
    longlongsize=8
    d_longdbl=define
    longdblsize=16
    longdblkind=3
    ivtype='long'
    ivsize=8
    nvtype='double'
    nvsize=8
    Off_t='off_t'
    lseeksize=8
    alignbytes=8
    prototype=define
  Linker and Libraries:
    ld='env MACOSX_DEPLOYMENT_TARGET=10.3 cc'
    ldflags =' -fstack-protector-strong -L/usr/local/lib'
    libpth=/usr/local/lib /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../lib/clang/8.0.0/lib /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib /usr/lib
    libs=-lpthread -ldbm -ldl -lm -lutil -lc
    perllibs=-lpthread -ldl -lm -lutil -lc
    libc=
    so=dylib
    useshrplib=false
    libperl=libperl.a
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_dlopen.xs
    dlext=bundle
    d_dlsymun=undef
    ccdlflags=' '
    cccdlflags=' '
    lddlflags=' -bundle -undefined dynamic_lookup -L/usr/local/lib -fstack-protector-strong'

Characteristics of this binary (from libperl): 
  Compile-time options:
    HAS_TIMES
[... 69 lines cut ...]
                       ^
                        
/System/Library/Perl/5.18/darwin-thread-multi-2level/CORE/pad.h:349:27: error: invalid suffix on literal; C++11 requires a space
      between literal and identifier [-Wreserved-user-defined-literal]
              "Pad 0x%"UVxf"[0x%"UVxf"] restore_local\n",       \
                                 ^
                                  
In file included from src/libslic3r/utils.cpp:66:
In file included from src/xsinit.h:39:
/System/Library/Perl/5.18/darwin-thread-multi-2level/CORE/perl.h:4257:37: error: invalid suffix on literal; C++11 requires a space
      between literal and identifier [-Wreserved-user-defined-literal]
  INIT("Use of uninitialized value%"SVf"%s%s");
                                    ^
                                     
/System/Library/Perl/5.18/darwin-thread-multi-2level/CORE/perl.h:4277:75: error: invalid suffix on literal; C++11 requires a space
      between literal and identifier [-Wreserved-user-defined-literal]
  INIT("Modification of non-creatable hash value attempted, subscript \"%"SVf"\"");
                                                                          ^
                                                                           
8 errors generated.
error building src/libslic3r/utils.o from 'src/libslic3r/utils.cpp' at /System/Library/Perl/5.18/ExtUtils/CBuilder/Base.pm line 177.
FAIL
! Installing ./xs failed. See /Volumes/LaCie/florian/.cpanm/work/1488382660.75657/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting
```

bubnikv, 2017-03-01:
For some reason your Build.PL is pulling wrong includes.

If you change xs/Build.PL and add at line 17
my @cflags = qw(....... -M)

This will instruct clang or gcc to display, which file includes which.

freiser77, 2017-03-01:
The last lines with the -M option set:

```
Building and testing Slic3r-XS-0.01 ... Building Slic3r-XS
Processing XS typemap files...
Generating main XS file...
cc -I/System/Library/Perl/5.18/darwin-thread-multi-2level/CORE -arch x86_64 -arch i386 -g -pipe -fno-common -DPERL_DARWIN -fno-strict-aliasing -fstack-protector -xc++ -Isrc/libslic3r -Isrc/glew/include -I/usr/local/include -I/usr/include -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -M -std=c++11 -DBOOST_LOG_DYN_LINK -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -arch x86_64 -arch i386 -g -pipe -fno-common -DPERL_DARWIN -fno-strict-aliasing -fstack-protector -Os -o src/poly2tri/sweep/advancing_front.o src/poly2tri/sweep/advancing_front.cc
clang: error: cannot use 'dependencies' output with multiple -arch options
error building src/poly2tri/sweep/advancing_front.o from 'src/poly2tri/sweep/advancing_front.cc' at /System/Library/Perl/5.18/ExtUtils/CBuilder/Base.pm line 177.
FAIL
! Installing ./xs failed. See /Volumes/LaCie/florian/.cpanm/work/1488395923.78684/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

```

bubnikv, 2017-03-01:
Frankly I don't fully understand the Perl build process, but it goes like
this. The Perl interpreter running knows it's @INC containing the search
paths of the Perl modules. One of the Perl modules is Config_heavy.pm,
which contains all the C compiler and linker switches to build Perl itself
and it's extension modules. The Config_heavy.pm is created during the Perl
compilation and installation process.

Now your slic3r Build.pm seems to be pulling a wrong Config_heavy.pm,
likely from the system path. I am not sure, but maybe something gets cached
somewhere re in the slicer directory from the times you ran build.pl with
the wrong perl. Would you please purge slicer directory and try to fetch
and compile slicer from scratch? Before running build.pm please verify that
you have the correct Perl in your search path and that the @INC returned by
Perl -V points to the perlbrew folder only.

alexrj, 2017-03-01:
@freiser77, that GUI porting work was already started in the "xsgui" branch of the main repository. We set-up the infrastructure for the double linking (Perl/C++) which is good for the transition because both bindings can coexist, and you can see how we ported some GUI parts. @lordofhyphens is actively working on porting the most difficult bit which is the OptionsGroup code which uses the dynamic features of Perl and doesn't translate straight to C++.

I strongly advice you contribute to the "xsgui" branch of the main repository so that we join efforts and work on the same codebase, otherwise we'll be duplicating the work. Forks like prusa3d can pull the changes when we finish our work. Thank you :)

freiser77, 2017-03-02:
@alexrj: thanks, I will do that.
@bubnikv: I did a fresh clone of Slic3r (alexrj's version because of the xsgui branch). But the error stays the same.

bubnikv, 2017-03-03:
This is a snippet from our build script to get compile parameters for the windows slic3r.exe
If you run it, you should get the C compilation / linking parameters, the same set the Perl is using to compile the Slic3r XS module. What do you get?

function perlcfg
{
  PATH="$PERL_PATH_OVERRIDE" "$PERL_BIN" -e "use Config; print \$Config{$1}"
}
function perlembedcfg
{
  PATH="$PERL_PATH_OVERRIDE" "$PERL_BIN" -MExtUtils::Embed -e $1 | sed 's/"//g'
}
cp "$BUILD_UTILS_DIR/slic3r.c" slic3r.c
PERL_CCFLAGS=$(perlcfg ccflags)
# or PER_CCFLAGS=$(perlembedcfg ccopts)
PERL_INCLUDES=$(perlembedcfg perl_inc)
PERL_LDFLAGS=$(perlembedcfg ldopts)
PERL_CC=$(perlcfg cc)
PERL_LD=$(perlcfg ld)

freiser77, 2017-03-03:
This is the output of the script (stored it in test.sh in the Slic3r directory)

```
cp: /slic3r.c: No such file or directory
./test.sh: line 3: : No such file or directory
./test.sh: line 7: : No such file or directory
./test.sh: line 7: : No such file or directory
./test.sh: line 3: : No such file or directory
./test.sh: line 3: : No such file or directory
```

freiser77, 2017-03-06:
I found out, that cpanm was running with the perl delivered by OSX. After I've installed it again with cpan running under perlbrews perl, the build of the command-line version runs okay.
However when I try to build it with ```perl Build.PL --gui``` it again causes an error. I will post it this evening.

bubnikv, 2017-03-06:
Great, good to hear it.

I would be thankful, if you could please document your findings, so we
could sum it up in a wiki page.

lordofhyphens, 2017-03-06:
I am about to write up my work in getting the OSX 10.9 server going over on
the main wiki.

The package script will be available too (currently in branch osx-package).

freiser77, 2017-03-07:
@lordofhyphens: If you need it: The way I found out, that under which version cpanm is running is:

```
Florians-iMac:~ florian$ cpanm -v
cpanm (App::cpanminus) 1.7042 on perl 5.025010 built for darwin-2level
Work directory is /Volumes/LaCie/florian/.cpanm/work/1488890450.914
You have make /usr/bin/make
You have /usr/bin/curl
You have /usr/bin/tar: bsdtar 2.8.3 - libarchive 2.8.3
You have /usr/bin/unzip
Usage: cpanm [options] Module [...]

Try `cpanm --help` or `man cpanm` for more options.
```

If there is any other perl than reasonable listed, then do ```cpan -v```and check if it is running under the correct perl version. After that do a ```cpan[1]> i App::cpanminus```

bubnikv, 2017-03-23:
As @alexrj is seemingly working on a C++ port of the GUI, I am closing this issue for now. Let's wait on what he will come up with. I am eager to accept these changes.

closed 2017-03-23 by bubnikv