prusa3d/PrusaSlicer #164: Idea: separate acceleration setting for external perimeters and top solid infill

burbilog, 2017-03-01:
Tuning down acceleration a lot (down to 150 mm/s^2) improves outer surface quality on my Anet A8 printer, but significantly increases print time. I know, it's more an issue of lousy, non-rigid printer, but many of us buy cheap lousy printers because we have no money for quality stuff. Here is an idea -- add two options to advanced acceleration control: external perimeters acceleration and top solid infill acceleration, because extreme slowing down makes sense only while printing outer shell and makes no sense while printing internal layers and perimeters. I think it should reduce print time significantly (can't prove it though).

bubnikv, 2017-03-03:
Your request makes sense, though it is always a good idea to try to keep the number of settable parameters manageable, both from the point of software development and maintenance, and from the point of usability.

burbilog, 2017-03-03:
It's an advanced box, after all :)

I don't like the idea of keeping number of parameters down. From advanced user's standpoint it's much better to have many settable parameters, because sometimes it's the only way to fix your own problem. Of couse they must be organized into groups (and hidden from basic user) and have sensible defaults. From developer's point of view -- yes, it's easier to develop with less GUI and less options, but usually this approach quickly leads into iphoneland...

bubnikv, 2017-03-03:
> From advanced user's standpoint it's much better to have many settable
parameters, because sometimes it's the only way to fix your own problem.

The problem is, with too many options only the author of the proposal will
understand his option. If you have too many options, only a handful will
understand them, therefore the usefulness of the settings will diminish and
it will only put burden on the developers and it will only introduce new
bugs. This is a general statement, it does not necessary regards this
feature request.

I am actually on the same page with Alessandro, that the best way is to
make things automagic.

lonelymyp, 2017-03-16:
In  simplify3d you can make an acceleration correction in the postscript.  For example for the outer perimeter:
> {REPLACE "; outer perimeter\n" "; outer perimeter\nM204 P150\n"}

Can it be done in a sliс3r Post-processing?

bubnikv, 2017-03-16:
Our Slic3r builds come with the perl executable, so you may write the
post-processing scripts in Perl and invoke the perl with the "post
processing scripts" settings.

http://manual.slic3r.org/advanced/post-processing
http://hackaday.com/2016/07/20/3d-printering-g-code-post-processing-with-perl/

lonelymyp, 2017-03-16:
Looks difficult for a non-programmer :(

HyGy, 2017-03-16:
I thing this is a good idea. In advanced mode it is good to enable such features. Set a good default value and basic users forget it.

What will be good, to that the outer permiters can be print with eg. 0.1mm the infill with 0.3mm. This make the outer surface nice, but speeds up the print process.

PaparazziN, 2023-10-22:
Hi @burbilog!
With some release of PrusaSlicer this option seem to have been introduced, so I think it would be appropriate to close this issue.
[image]

lukasmatena, 2023-10-23:
@PaparazziN You are correct, this is already implemented in 2.6.0. Closing.

closed 2023-10-23 by lukasmatena