prusa3d/PrusaSlicer #165: Mac build 1.33.8 seems to have broken variable layer height editing

mhackney, 2017-03-01:
### Version
1.33.8 Mac OSX

### Operating system type + version
Mac OSX 10.12.3

### Behavior
Previous beta version worked fine for editing variable heights. This version does not show the layer height editing sliders when I click the Layer Editing button on the main toobar.

#### STL/Config (.ZIP) where problem occurs
All STLs

mhackney, 2017-03-02:
I've reinstalled version 1.33.4 and confirmed that the Layer Editing works correctly. I am in 3D view and when I click the Layer Editing button in 1.33.8, nothing changes and the editing slider does not appear.

bubnikv, 2017-03-03:
I think you have the variable layer thickness editing disabled in the printer settings. I believe this checkbox was not in 1.33.4 yet, it has been implemented later.

[image]

mhackney, 2017-03-03:
I'm sorry for not being clear in my original post, the enable variable layer height feature IS enabled. And it was in 1.33.4. So the issue still remains that variable height editing is not working. Can anyone else confirm this please.

[image]

bubnikv, 2017-03-03:
I just tested on OSX Sierra and it works. I don't know how to help you.

mhackney, 2017-03-03:
Ok, what version of Sierra are you running? I'm on 10.12.3. There was a minor upgrade from apple last week that broke my 3DConnexion Space Navigator 3D mouse. They are investigating that issue. I doubt this is related but...

Is there any diagnostics I can do to get more info? It is very odd that the previous version works just fine but this one doesn't. And I have both installed simultaneously. I've also reinstalled 1.33.8.

bubnikv, 2017-03-03:
It would help me, if you could try the other versions (1.33.5, 1.33.6,
1.33.7) to find out, which is the first one broken on your system, please.
Thanks

mhackney, 2017-03-03:
Yes, I can do that. I'll start with 33.7 and go backwards!

macdarren, 2017-03-03:
I had the same issue....went back to 1.33.7 and it worked fine...then went back to 1.33.8 and what do you know that works too....maybe it was just quitting and restarting 1.33.8....

MacOS 10.12.4 Beta (16E175b)

mhackney, 2017-03-03:
Ok, same here! I had quit, uninstalled and installed and run 1.33.4 and that worked. But 1.33.8 did not. But, I installed .7 - it worked and when I simply ran .8 again (with .7 running) it now magically works too.

bubnikv, 2017-03-03:
Thanks guys for heads up. This is magic.

I have changed the OpenGL shaders between 1.33.7 and 1.33.8. The 1.33.8 shows the layers only if your camera is close enough to the object for the layers to be distinctive. If one moves far from the object, the layers are blended away now, in 1.33.7 and older they were rendered causing an ugly moire effect.

It is quite possible, that the graphics driver caches the precompiled OpenGL shaders somewhere and that the driver failed to detect a change of the shader in 1.33.8. I don't know. Anyway, please keep me updated. I will leave this issue open as I expect more people having trouble with that.

sgano, 2017-03-14:
I had the same issue on my mac.  I deleted the app and then just re installed the same v1.33.8 (as I didn't have any older versions on hand) and it worked after that. Can't explain it; but thought it might help others.

bubnikv, 2017-03-30:
The variable layer height visualization has been rewritten to OpenGL 2.0 in the alpha 1.34.0. I hope this issue goes away as Slic3r is now using a standard OpenGL 2.0 API instead of a legacy OpenGL 1.1 API with extensions.

Closing. Please feel free to reopen if the issue reappears.

closed 2017-03-30 by bubnikv