prusa3d/PrusaSlicer #166: Need to include '-lboost_filesystem' in build

ceryen, 2017-03-02:
The commits 6a90fceaaf38e30b4ac4e00cb503625a7c5ca7e9 and 6f119692571dca109ed4534549472154b792ec25 added some calls to boost::filesystem.  As a result, I believe '-lboost_filesystem' needs to be added to the build.  I believe this can be done by modifying 'xs/Build.PL' to include 'filesystem' in the 'boost_libraries' array:

    my @boost_libraries = qw(system thread log filesystem);  # we need these

Without this change, my build fails with errors that look like:

g++ -o "blib\arch\auto\Slic3r\XS\XS.dll" -Wl,--base-file,"buildtmp\XS.base" -Wl,--image-base,0x343f0000 -mdll -s -L"C:\Dev\strawberry\perl\lib\CORE" -L"C:\Dev\strawberry\c\lib" "buildtmp\XS.lds" -lstdc++ -Lsrc/libslic3r -LC:\dev\boost_1_59_0\stage\lib -lboost_system-mgw47-mt-s-1_59 -lboost_thread-mgw47-mt-s-1_59 -lboost_log-mgw47-mt-s-1_59 "buildtmp\XS.exp"
src\libslic3r\Print.o:Print.cpp:(.text+0xe4): undefined reference to `boost::filesystem::path::codecvt()'
src\libslic3r\Print.o:Print.cpp:(.text+0x101): undefined reference to `boost::filesystem::path_traits::convert(char const*, char const*, std::basic_string<wchar_t, std::char_traits<wchar_t>, std::allocator<wchar_t> >&, std::codecvt<wchar_t, char, int> const&)'
src\libslic3r\Print.o:Print.cpp:(.text+0x14c2): undefined reference to `boost::filesystem::path::parent_path() const'
...

ceryen, 2017-03-02:
I'm building for Windows using the instructions found here: https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-Windows

I'm using the first set of instructions and not the Visual Studio instructions

bubnikv, 2017-03-02:
Fixed by https://github.com/prusa3d/Slic3r/commit/349cf841d6d600594e4d45e68603dbf4f60aee83

closed 2017-03-02 by bubnikv