prusa3d/PrusaSlicer #168: Mixing extruder and wipe tower issue

Megane008, 2017-03-03:
### Version
Slic3r 1.33.8

### Operating system type + version
Win 10x64
### Behavior
Im using a mixing extruder (diamond hotend) Im triing to print support from PVA and two more PLA colors. 
Im using manual wipe tower placed behind the object which results in every layer starting there and purging the hotend. But its not like that with support, everytime slic3r starts new layer with support closest to the latest position and doesnt start at wipe tower. 
Is there any way to change that behaviour? Way around it would be to load support as separate STL but I dont know about any way how to export support as generated STL from Slic3r. 

Thanks for help

bubnikv, 2017-05-05:
How did you create the manual wipe tower? There is no support for a wipe tower in Slic3r as of now. Did you post-process your g-code to add a wipe tower?

Megane008, 2017-05-05:
[image]
Im using manually placed object behind the printed object and it results in starting point of every layer with new tool selected there, resulting in purging chamber before printing main object. But as I wrote above there is some predefined behaviour inside Slic3r which results in printing generated support starting point not the furthest from XYZ0 but from closest last position. 

btw Im also from Czech as I hear you are bubnikv, we could probably discuss this personaly, I would apreciate some direction to solution of this behaviour.

bubnikv, 2017-05-09:
I am currently working on a purge tower for slic3r for our multi-material setup. I am also reworking the ordering of extruders, so the soluble support will not trigger an unnecessary tool change (https://github.com/prusa3d/Slic3r/issues/212).

You are most welcome to visit us at our new site 
https://www.google.cz/maps/@50.1089566,14.4372466,3a,75y,262.97h,89.14t/data=!3m6!1e1!3m4!1snzItCqHtFYwyp_NCvmh2VA!2e0!7i13312!8i6656?hl=en

bubnikv, 2017-05-20:
The latest beta has some multi material improvements implemented:
https://github.com/prusa3d/Slic3r/releases/tag/version_1.34.1.24

Megane008, 2017-05-22:
Thanks it works nicely now Im just testing and the result seems OK. One thing I found out is that you need to load wipe tower object first to have right efect of purging before printing support. Maybe add option support before object/after object into slic3r?

bubnikv, 2017-05-22:
How exactly are you defining the wipe tower? As a separate object? As a
separate volume?

> One thing I found out is that you need to load wipe tower object first to
have right efect of purging before printing support. Maybe add option
support before object/after object into slic3r?

Would you please be very detailed? I don't quite follow you.

Megane008, 2017-05-22:
I do following
1. load separate object ( can be any shape I use custom shape to minimalize material consuption but do succesfull purge ) - this is wipe tower - it can be 1, 2 or 16 part object
2. I load main object 
3. I place wipe tower behind main object ( X,Y,Z coordinates are larger than main object ) 
4. I select color ( extruder ) for main object and for support ( multimaterial tab slic3r ) 
5. I select color ( extruder ) for wipe tower part 1 correspoding with color ( extruder ) of main object, for wipe tower part 2 corresponding with color ( extruder ) of support 
6. slice and done

I hope its understandable now
About the option support before after main object it would be only more understandable to users I found only by experiment that it works now only if you load wipe object as first

lukasmatena, 2019-12-17:
This issue is from old times when there was even no wipe tower in Slic3rPE. I think it should now be closed as outdated (although maybe not completely solved).

In my opinion, issue #3044 is the place to discuss remaining diamond hotend issues.

closed 2019-12-17 by lukasmatena