prusa3d/PrusaSlicer #170: Failing All Tests During Build

JakeAllison, 2017-03-04:
OS: Ubuntu 16.04.1 (Fresh install)

Slic3r Prusa Edition Version: version_1.33.8-85-ga9a2000
- The build is failing. The output has been included in "Slic3r_Prusa_Build.txt"
- This issue has only existed in the past few days.
- The following text was given at the end:

> Files=32, Tests=0,  1 wallclock secs ( 0.07 usr  0.02 sys +  1.14 cusr  0.09 csys =  1.32 CPU)
> Result: FAIL
> Some tests failed. Please report the failure to the author!

Slic3r Version: 1.2.9-647-g4e586b8
- As a comparison, the regular version of Slic3r works fine.

[Slic3r_Build.txt](https://github.com/prusa3d/Slic3r/files/818863/Slic3r_Build.txt)
[Slic3r_Prusa_Build.txt](https://github.com/prusa3d/Slic3r/files/818864/Slic3r_Prusa_Build.txt)

bubnikv, 2017-03-05:
This is weird.I just tested on our buildbot machine and it is all right:

Linux BuildBot 3.16.0-4-amd64 #1 SMP Debian 3.16.7-ckt11-1+deb8u5 (2015-10-09) x86_64 GNU/Linux

bubnikv, 2017-03-05:
There is something wrong with your system. First the Slic3r XS.so is installed into

/home/jake/perl5/lib/perl5/x86_64-linux-gnu-thread-multi/auto/Slic3r/XS/XS.so

Then when running the integration tests, Perl complains, that it cannot find Slic3r/XS/XS.so in (@INC contains: /home/jake/Documents/3D_Printing/Prusa3D/Slic3r/t/../lib /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.22.1 /usr/local/share/perl/5.22.1 /usr/lib/x86_64-linux-gnu/perl5/5.22 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.22 /usr/share/perl/5.22 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base .)

Indeed, there is no /home/jake/perl5/.... in your @INC. Frankly I don't know why is it that. You may try to add /home/jake/perl5/lib/perl5/x86_64-linux-gnu-thread-multi/ to PERL5LIB environment variable and the test cases will likely run then.

JakeAllison, 2017-03-05:
Running with the --sudo option did the trick. The question i have is what is the difference between the regular Slic3r and Prusa Edition that requires this one to run --sudo in my case. All steps are otherwise identical. I ran the instructions from [the Slic3r website](https://github.com/alexrj/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux) line for line, using the latest code from the master branch of each.

Running --sudo most noticeably made the below message go away as it should, and it did not create the ~/perl5 folder.
> !
> ! Can't write to /usr/local/share/perl/5.22.1 and /usr/local/bin: Installing modules to /home/jake/perl5
> ! To turn off this warning, you have to do one of the following:
> !   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.22.1 and /usr/local/bin)
> !   - Configure local::lib in your existing shell to set PERL_MM_OPT etc.
> !   - Install local::lib by running the following commands
> !
> !         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
> !

bubnikv, 2017-03-17:
@lordofhyphens integrated a "local" Perl installation into their Slic3r and I did not integrate these changes yet as they are not that important as we are providing binary builds for most platforms and I have too much to do.

I have to update build instructions for our fork though.

bubnikv, 2017-03-17:
Closing this issue.

closed 2017-03-17 by bubnikv

lordofhyphens, 2017-03-17:
@JakeAllison local::lib can be used with the Prusa build (and I still recommend it on Linux to avoid trashing your distro's perk module setup).

Google local::lib for the basics.

It was @alexrj who integrated it, I just was an advocate of doing so ;-)