prusa3d/PrusaSlicer #172: Failed to build 06f82d1db5e9286a6ea6af16816f117aa8a2bb4f

burbilog, 2017-03-06:
### Version
06f82d1db5e9286a6ea6af16816f117aa8a2bb4f
also d20a9d73d4f07b981dc72809ba637c1d33b53a1f

### Operating system type + version
Ubuntu 14.04.5 LTS

### Behavior
~~~
cc -I/usr/lib/perl/5.18/CORE -fPIC -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -std=c++11 -DBOOST_LOG_DYN_LINK -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/clipper.o src/clipper.cpp
In file included from /usr/include/c++/4.8/algorithm:62:0,
                 from src/clipper.cpp:44:
/usr/include/c++/4.8/bits/stl_algo.h: In instantiation of '_RandomAccessIterator std::__unguarded_partition(_RandomAccessIterator, _RandomAccessIterator, const _Tp&, _Compare) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<ClipperLib::IntersectNode*, std::vector<ClipperLib::IntersectNode> >; _Tp = ClipperLib::IntersectNode; _Compare = ClipperLib::Clipper::FixupIntersectionOrder()::__lambda1]':
/usr/include/c++/4.8/bits/stl_algo.h:2296:78:   required from '_RandomAccessIterator std::__unguarded_partition_pivot(_RandomAccessIterator, _RandomAccessIterator, _Compare) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<ClipperLib::IntersectNode*, std::vector<ClipperLib::IntersectNode> >; _Compare = ClipperLib::Clipper::FixupIntersectionOrder()::__lambda1]'
/usr/include/c++/4.8/bits/stl_algo.h:2337:62:   required from 'void std::__introsort_loop(_RandomAccessIterator, _RandomAccessIterator, _Size, _Compare) [with _RandomAccessIterator = __gnu_cxx::__normal_iterator<ClipperLib::IntersectNode*, std::vector<ClipperLib::IntersectNode> >; _Size = long int; _Compare = ClipperLib::Clipper::FixupIntersectionOrder()::__lambda1]'
/usr/include/c++/4.8/bits/stl_algo.h:5499:44:   required from 'void std::sort(_RAIter, _RAIter, _Compare) [with _RAIter = __gnu_cxx::__normal_iterator<ClipperLib::IntersectNode*, std::vector<ClipperLib::IntersectNode> >; _Compare = ClipperLib::Clipper::FixupIntersectionOrder()::__lambda1]'
src/clipper.cpp:2660:143:   required from here
/usr/include/c++/4.8/bits/stl_algo.h:2263:35: error: no match for call to '(ClipperLib::Clipper::FixupIntersectionOrder()::__lambda1) (ClipperLib::IntersectNode&, const ClipperLib::IntersectNode&)'
    while (__comp(*__first, __pivot))
                                   ^
src/clipper.cpp:2660:62: note: candidates are:
   std::sort(m_IntersectList.begin(), m_IntersectList.end(), [](IntersectNode &node1, IntersectNode &node2) { return node2.Pt.Y < node1.Pt.Y; });
                                                              ^
In file included from /usr/include/c++/4.8/algorithm:62:0,
                 from src/clipper.cpp:44:
/usr/include/c++/4.8/bits/stl_algo.h:2263:35: note: bool (*)(ClipperLib::IntersectNode&, ClipperLib::IntersectNode&) <conversion>
    while (__comp(*__first, __pivot))
                                   ^
/usr/include/c++/4.8/bits/stl_algo.h:2263:35: note:   candidate expects 3 arguments, 3 provided
src/clipper.cpp:2660:106: note: ClipperLib::Clipper::FixupIntersectionOrder()::__lambda1
   std::sort(m_IntersectList.begin(), m_IntersectList.end(), [](IntersectNode &node1, IntersectNode &node2) { return node2.Pt.Y < node1.Pt.Y; });
                                                                                                          ^
src/clipper.cpp:2660:106: note:   no known conversion for argument 2 from 'const ClipperLib::IntersectNode' to 'ClipperLib::IntersectNode&'
In file included from /usr/include/c++/4.8/algorithm:62:0,
                 from src/clipper.cpp:44:
/usr/include/c++/4.8/bits/stl_algo.h:2266:34: error: no match for call to '(ClipperLib::Clipper::FixupIntersectionOrder()::__lambda1) (const ClipperLib::IntersectNode&, ClipperLib::IntersectNode&)'
    while (__comp(__pivot, *__last))
                                  ^
src/clipper.cpp:2660:62: note: candidates are:
   std::sort(m_IntersectList.begin(), m_IntersectList.end(), [](IntersectNode &node1, IntersectNode &node2) { return node2.Pt.Y < node1.Pt.Y; });
                                                              ^
In file included from /usr/include/c++/4.8/algorithm:62:0,
                 from src/clipper.cpp:44:
/usr/include/c++/4.8/bits/stl_algo.h:2266:34: note: bool (*)(ClipperLib::IntersectNode&, ClipperLib::IntersectNode&) <conversion>
    while (__comp(__pivot, *__last))
                                  ^
/usr/include/c++/4.8/bits/stl_algo.h:2266:34: note:   candidate expects 3 arguments, 3 provided
src/clipper.cpp:2660:106: note: ClipperLib::Clipper::FixupIntersectionOrder()::__lambda1
   std::sort(m_IntersectList.begin(), m_IntersectList.end(), [](IntersectNode &node1, IntersectNode &node2) { return node2.Pt.Y < node1.Pt.Y; });
                                                                                                          ^
src/clipper.cpp:2660:106: note:   no known conversion for argument 1 from 'const ClipperLib::IntersectNode' to 'ClipperLib::IntersectNode&'
error building src/clipper.o from 'src/clipper.cpp' at /usr/share/perl/5.18/ExtUtils/CBuilder/Base.pm line 177.
FAIL
! Installing ./xs failed. See /home/rm/.cpanm/work/1488804089.29163/build.log for details. Retry with --force to force install it.
~~~

#### STL/Config (.ZIP) where problem occurs
No config

bubnikv, 2017-03-06:
Does a following commit fix it for you?
https://github.com/prusa3d/Slic3r/commit/c23c0ee7d2369913f2f2f97008e003b7f0307a0a

burbilog, 2017-03-06:
Yep

closed 2017-03-06 by burbilog