prusa3d/PrusaSlicer #175: No E distances generated when support is selected. bug?

DjDemonD, 2017-03-06:
Using windows version slic3r 1.33.8-prusa3d-win64

I have uploaded 2 gcode files. 
The first has support selected.
All the first layer E distances are 2.8mm so nothing is extruded on layer 1.

The second has support off and it works normally, the absolute e-distances increase throughout the file.

I might be doing something stupid but I've never seen this in slic3r before.

[GCODE.zip](https://github.com/prusa3d/Slic3r/files/821561/GCODE.zip)

bubnikv, 2017-03-06:
Please provide an STL and a config.ini. Thank you.

DjDemonD, 2017-03-06:
The stl, and my config bundle. If this is not the right config file please let me know where to locate the one you need.

Also I do have two other versions of slic3r on my machine, and I know they share config, might this be relevant?

[config_and_stl.zip](https://github.com/prusa3d/Slic3r/files/821779/config_and_stl.zip)

nebbian, 2017-03-18:
I've found the same thing.  When support is used, the outer skirt has no extrusion.  I've had to manually push the filament through while the head is moving around the skirt to prime the nozzle.  
This is on the Mac OS X version, and only started happening on the latest release.

bubnikv, 2017-03-24:
@DjDemonD the zip provided contains a link to the config bundle, but not the actual config file. Would you  please provide a config? Also please export a config, not a config bundle. Thank you.

bubnikv, 2017-03-27:
This is duplicate of https://github.com/prusa3d/Slic3r/issues/184
Hopefully fixed by https://github.com/prusa3d/Slic3r/commit/640698d28bc66a248199be95f70e2eed26bff6a8

closed 2017-03-27 by DjDemonD

bubnikv, 2017-03-27:
@DjDemonD That was fast :-)
Thanks for the report and for the testing. Looking for another valid high quality bug report :-)

DjDemonD, 2017-03-27:
You're welcome.