prusa3d/PrusaSlicer #176: Can't start application with codeset problem, unless use Strawberry Perl

hiseob, 2017-03-07:
### Version
1.33.8-prusa3d-win64

### Operating system type + version
Windows 10 Pro, Korean

### Behavior

1. cannot launch this application with slic3r.exe right after extract from zip.

2. cannot launch with perl5.24.0.exe either.

```
Slic3rPE> perl5.24.0.exe slic3r.pl
The locale codeset (cp949) isn't one that perl can decode, stopped at D:/Personal/Desktop/Slic3r-1.33.8-prusa3d-win64-full-201702210907/lib/Encode/Locale.pm line 120.
Compilation failed in require at D:/Personal/Desktop/Slic3r-1.33.8-prusa3d-win64-full-201702210907/lib/Slic3r.pm line 89.
BEGIN failed--compilation aborted at D:/Personal/Desktop/Slic3r-1.33.8-prusa3d-win64-full-201702210907/lib/Slic3r.pm line 89.
Compilation failed in require at slic3r.pl line 15.
BEGIN failed--compilation aborted at slic3r.pl line 15.
```
this is freaking old problem, pains me every time when updating slic3rpe... so started digging.

2. I've added some printing code into perl libraries, inside lib/

```
diff -urN "c/Program Files/Prusa3D/Slic3rPE_orig/lib/Encode/KR.pm" "c/Program Files/Prusa3D/Slic3rPE/lib/Encode/KR.pm"
--- "c/Program Files/Prusa3D/Slic3rPE_orig/lib/Encode/KR.pm"    2017-02-21 18:30:22.918000000 +0900
+++ "c/Program Files/Prusa3D/Slic3rPE/lib/Encode/KR.pm" 2017-03-07 20:59:50.801519100 +0900
@@ -9,8 +9,9 @@
 use Encode;
 our $VERSION = do { my @r = ( q$Revision: 2.3 $ =~ /\d+/g ); sprintf "%d." . "%02d" x $#r, @r };
 use XSLoader;
+print "Encode/KR.pm before XSLoader::load\n";
 XSLoader::load( __PACKAGE__, $VERSION );
-
+print "Encode/KR.pm after XSLoader::load\n";
 use Encode::KR::2022_KR;

 1;
diff -urN "c/Program Files/Prusa3D/Slic3rPE_orig/lib/Encode/Locale.pm" "c/Program Files/Prusa3D/Slic3rPE/lib/Encode/Locale.pm"
--- "c/Program Files/Prusa3D/Slic3rPE_orig/lib/Encode/Locale.pm"        2017-02-21 18:30:23.183625000 +0900
+++ "c/Program Files/Prusa3D/Slic3rPE/lib/Encode/Locale.pm"     2017-03-07 21:00:22.436645200 +0900
@@ -102,10 +102,16 @@

     # final fallback
     $ENCODING_LOCALE ||= $^O eq "MSWin32" ? "cp1252" : "UTF-8";
+    #$ENCODING_LOCALE = "cp1252"; # !! this will make application runnable, but not using now
     $ENCODING_LOCALE_FS ||= $ENCODING_LOCALE;
     $ENCODING_CONSOLE_IN ||= $ENCODING_LOCALE;
     $ENCODING_CONSOLE_OUT ||= $ENCODING_CONSOLE_IN;

+    print "Encode/Locale.pm ENCODING_LOCALE $ENCODING_LOCALE\n";
+    print "Encode/Locale.pm ENCODING_LOCALE_FS $ENCODING_LOCALE_FS\n";
+    print "Encode/Locale.pm ENCODING_CONSOLE_IN $ENCODING_CONSOLE_IN\n";
+    print "Encode/Locale.pm ENCODING_CONSOLE_OUT $ENCODING_CONSOLE_OUT\n";
+
     unless (Encode::find_encoding($ENCODING_LOCALE)) {
        my $foundit;
        if (lc($ENCODING_LOCALE) eq "gb18030") {
diff -urN "c/Program Files/Prusa3D/Slic3rPE_orig/lib/Encode.pm" "c/Program Files/Prusa3D/Slic3rPE/lib/Encode.pm"
--- "c/Program Files/Prusa3D/Slic3rPE_orig/lib/Encode.pm"       2017-02-21 18:30:21.980500000 +0900
+++ "c/Program Files/Prusa3D/Slic3rPE/lib/Encode.pm"    2017-03-07 21:23:30.656031200 +0900
@@ -120,10 +120,13 @@
         if ( my $mod = $ExtModule{$name} || $ExtModule{$lc} ) {
             $mod =~ s,::,/,g;
             $mod .= '.pm';
+            print "Encode.pm getEncoding mod $mod\n";
             eval { require $mod; };
+            warn() if $@;
             exists $Encoding{$name} and return $Encoding{$name};
         }
     }
+    print "Encode.pm getEncoding FAILURE\n";
     return;
 }

```

3. and now..... 

```
C:\Program Files\Prusa3D\Slic3rPE>perl5.24.0.exe slic3r.pl
Encode/Locale.pm ENCODING_LOCALE cp949
Encode/Locale.pm ENCODING_LOCALE_FS cp949
Encode/Locale.pm ENCODING_CONSOLE_IN cp949
Encode/Locale.pm ENCODING_CONSOLE_OUT cp949
Encode.pm getEncoding mod Encode/KR.pm
Encode/KR.pm before XSLoader::load
Can't locate loadable object for module Encode::KR in @INC (@INC contains: C:/Program Files/Prusa3D/Slic3rPE/lib .) at C:/Program Files/Prusa3D/Slic3rPE/lib/Encode.pm line 124.
Compilation failed in require at C:/Program Files/Prusa3D/Slic3rPE/lib/Encode.pm line 124.
        ...caught at C:/Program Files/Prusa3D/Slic3rPE/lib/Encode.pm line 125.
Encode.pm getEncoding FAILURE
Encode.pm getEncoding mod Encode/KR.pm
Attempt to reload Encode/KR.pm aborted.
Compilation failed in require at C:/Program Files/Prusa3D/Slic3rPE/lib/Encode.pm line 124.
        ...caught at C:/Program Files/Prusa3D/Slic3rPE/lib/Encode.pm line 125.
Encode.pm getEncoding FAILURE
The locale codeset (cp949) isn't one that perl can decode, stopped at C:/Program Files/Prusa3D/Slic3rPE/lib/Encode/Locale.pm line 126.
Compilation failed in require at C:/Program Files/Prusa3D/Slic3rPE/lib/Slic3r.pm line 89.
BEGIN failed--compilation aborted at C:/Program Files/Prusa3D/Slic3rPE/lib/Slic3r.pm line 89.
Compilation failed in require at slic3r.pl line 15.
BEGIN failed--compilation aborted at slic3r.pl line 15.
```
cannot run with built-in perl BUT

```
C:\Program Files\Prusa3D\Slic3rPE>c:\Strawberry\perl\bin\perl.exe  slic3r.pl
Encode/Locale.pm ENCODING_LOCALE cp949
Encode/Locale.pm ENCODING_LOCALE_FS cp949
Encode/Locale.pm ENCODING_CONSOLE_IN cp949
Encode/Locale.pm ENCODING_CONSOLE_OUT cp949
Encode.pm getEncoding mod Encode/KR.pm
Encode/KR.pm before XSLoader::load
Encode/KR.pm after XSLoader::load
```
with Strawberry Perl distribution, working well.

I've download and installed from http://strawberryperl.com/ 5.24.1.1 64-bit.

Please solve this problem...

[image]

bubnikv, 2017-03-07:
Dear @hiseob

The problem bugs me as well. I would be very thankful, if you could help me debug the issue as I don't have an access to an asian Windows machine.

If you are willing to help, I will think up some tests, which will hopefully help me to find the missing modules.

By the way, I added Korean locales to the 1.33.8 build, so I hoped this is fixed already.

bubnikv, 2017-03-07:
There seems to be missing DLLs in lib\auto\Encode.
I will add them at our build server.

hiseob, 2017-03-07:
I've digged more,

```
diff -urN "c/Program Files/Prusa3D/Slic3rPE_orig/lib/XSLoader.pm" "c/Program Files/Prusa3D/Slic3rPE/lib/XSLoader.pm"
--- "c/Program Files/Prusa3D/Slic3rPE_orig/lib/XSLoader.pm"     2017-02-21 18:30:29.121125000 +0900
+++ "c/Program Files/Prusa3D/Slic3rPE/lib/XSLoader.pm"  2017-03-08 01:57:41.395350100 +0900
@@ -40,7 +40,7 @@
     my $c = () = split(/::/,$caller,-1);
     $modlibname =~ s,[\\/][^\\/]+$,, while $c--;    # Q&D basename
     my $file = "$modlibname/auto/$modpname/$modfname.xs\.dll";
-
+    print "XSLoader.pm file $file\n";
 #   print STDERR "XSLoader::load for $module ($file)\n" if $dl_debug;

     my $bs = $file;
@@ -91,6 +91,7 @@

 sub bootstrap_inherit {
     require DynaLoader;
+    print "XSLoader.pm DynaLoader::bootstrap_inherit\n";
     goto \&DynaLoader::bootstrap_inherit;
 }
```

run result

```
c:\Program Files\Prusa3D\Slic3rPE>set PERL_DL_DEBUG=1

c:\Program Files\Prusa3D\Slic3rPE>perl5.24.0.exe slic3r.pl
XSLoader.pm file c:/Program Files/Prusa3D/Slic3rPE/lib/auto/Cwd/Cwd.xs.dll
DynaLoader.pm loaded (c:/Program Files/Prusa3D/Slic3rPE/lib ., C:\Strawberry64\c\lib C:\Strawberry64\c\x86_64-w64-mingw32\lib C:\Strawberry64\c\lib\gcc\x86_64-w64-mingw32\4.9.2)
DynaLoader::bootstrap for Win32 (auto/Win32/Win32.xs.dll)
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/auto/List/Util/Util.xs.dll
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/auto/Fcntl/Fcntl.xs.dll
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/auto/POSIX/POSIX.xs.dll
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/auto/mro/mro.xs.dll
XSLoader.pm DynaLoader::bootstrap_inherit
DynaLoader::bootstrap for mro (auto/mro/mro.xs.dll)
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/auto/threads/threads.xs.dll
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/auto/threads/shared/shared.xs.dll
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/auto/attributes/attributes.xs.dll
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/Wx/auto/Wx/Wx.xs.dll
XSLoader.pm DynaLoader::bootstrap_inherit
DynaLoader::bootstrap for Wx (auto/Wx/Wx.xs.dll)
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/Wx/auto/Wx/Print/Print.xs.dll
XSLoader.pm DynaLoader::bootstrap_inherit
DynaLoader::bootstrap for Wx::Print (auto/Wx/Print/Print.xs.dll)
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/auto/Slic3r/XS/XS.xs.dll
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/auto/B/B.xs.dll
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/auto/Class/XSAccessor/XSAccessor.xs.dll
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/auto/Encode/Encode.xs.dll
Encode/Locale.pm ENCODING_LOCALE cp949
Encode/Locale.pm ENCODING_LOCALE_FS cp949
Encode/Locale.pm ENCODING_CONSOLE_IN cp949
Encode/Locale.pm ENCODING_CONSOLE_OUT cp949
Encode.pm getEncoding mod Encode/KR.pm
Encode/KR.pm before XSLoader::load
XSLoader.pm file C:/Program Files/Prusa3D/Slic3rPE/lib/auto/Encode/KR/KR.xs.dll
XSLoader.pm DynaLoader::bootstrap_inherit
DynaLoader::bootstrap for Encode::KR (auto/Encode/KR/KR.xs.dll)
dl_findfile(-LC:/Program Files/Prusa3D/Slic3rPE/lib -Lc:/Program Files/Prusa3D/Slic3rPE/lib -L. KR)
 checking in C:/Program Files/Prusa3D/Slic3rPE/lib for KR.xs.dll
 checking in C:/Program Files/Prusa3D/Slic3rPE/lib for KR.dll
 checking in C:/Program Files/Prusa3D/Slic3rPE/lib for libKR.dll
 checking in C:/Program Files/Prusa3D/Slic3rPE/lib for KR
 checking in c:/Program Files/Prusa3D/Slic3rPE/lib for KR.xs.dll
 checking in c:/Program Files/Prusa3D/Slic3rPE/lib for KR.dll
 checking in c:/Program Files/Prusa3D/Slic3rPE/lib for libKR.dll
 checking in c:/Program Files/Prusa3D/Slic3rPE/lib for KR
 checking in . for KR.xs.dll
 checking in . for KR.dll
 checking in . for libKR.dll
 checking in . for KR
dl_findfile found:
Can't locate loadable object for module Encode::KR in @INC (@INC contains: C:/Program Files/Prusa3D/Slic3rPE/lib c:/Program Files/Prusa3D/Slic3rPE/lib .) at C:/Program Files/Prusa3D/Slic3rPE/lib/Encode.pm line 124.
Compilation failed in require at C:/Program Files/Prusa3D/Slic3rPE/lib/Encode.pm line 124.
        ...caught at C:/Program Files/Prusa3D/Slic3rPE/lib/Encode.pm line 125.
Encode.pm getEncoding FAILURE
Encode.pm getEncoding mod Encode/KR.pm
Attempt to reload Encode/KR.pm aborted.
Compilation failed in require at C:/Program Files/Prusa3D/Slic3rPE/lib/Encode.pm line 124.
        ...caught at C:/Program Files/Prusa3D/Slic3rPE/lib/Encode.pm line 125.
Encode.pm getEncoding FAILURE
The locale codeset (cp949) isn't one that perl can decode, stopped at C:/Program Files/Prusa3D/Slic3rPE/lib/Encode/Locale.pm line 126.
Compilation failed in require at C:/Program Files/Prusa3D/Slic3rPE/lib/Slic3r.pm line 89.
BEGIN failed--compilation aborted at C:/Program Files/Prusa3D/Slic3rPE/lib/Slic3r.pm line 89.
Compilation failed in require at slic3r.pl line 15.
BEGIN failed--compilation aborted at slic3r.pl line 15.
```
You're right.  Unlike strawberry's lib/auto there is missing dlls in lib/auto/ 

Thanks!

bubnikv, 2017-03-07:
These library sets should fix it.

I shall update the official builds.

bubnikv, 2017-03-07:
Until then, these dll packs shall fix it for you.

[Slic3r-1.33.8-win32-encode.zip](https://github.com/prusa3d/Slic3r/files/825187/Slic3r-1.33.8-win32-encode.zip)
[Slic3r-1.33.8-win64-encode.zip](https://github.com/prusa3d/Slic3r/files/825188/Slic3r-1.33.8-win64-encode.zip)

bubnikv, 2017-03-07:
@hiseob I have updated the Windows packages on the Release page:
https://github.com/prusa3d/Slic3r/releases/tag/version_1.33.8

Please confirm, that it solves your issue. Thanks.

hiseob, 2017-03-08:
problem solved :D

thanks for updating.

[image]

bubnikv, 2017-03-08:
Great! Thank you for your help. Otherwise I was slowly getting prepared to buy an Asian Windows 7 for testing.

closed 2017-03-08 by bubnikv