prusa3d/PrusaSlicer #177: Failed to build 32b042830359bdc4ae4cbe13c172e2e85f8b2a66

burbilog, 2017-03-07:
### Version
32b042830359bdc4ae4cbe13c172e2e85f8b2a66

### Operating system type + version
Ubuntu 14.04.5 LTS

### Behavior
Since Slic3r now requires intel tbb library, I downloaded  it  from official site and untarred it, then set up environment variable:

TBB_DIR=/home/rm/tbb2017_20170118oss

Yet perl Build.PL fails:

~~~
Running Build.PL
Configuring /home/rm/slic3r.prusa/Slic3r/xs ... Slic3r requires the Intel Thread Building Blocks libraries. Please make sure the library is installed.

If the Intel Thread Building Blocks library is installed, this script should be able to locate them in several
standard locations. If this is not the case, you might want to supply a path to the library 
through the TBB_DIR environment variable:

    TBB_DIR=/path/to/TBB perl Build.PL

Or you may specify TBB_INCLUDEPATH and TBB_LIBRARYPATH separatly, which
is handy, if you have built the Thread Building Blocks libraries with mutliple settings.

N/A
! Configure failed for ./xs. See /home/rm/.cpanm/work/1488897219.21239/build.log for details.
The XS/C++ code failed to compile, aborting
~~~

Setting individual TBB_INCLUDEPATH and/or TBB_LIBRARYPATH does not help.

#### STL/Config (.ZIP) where problem occurs
No config

bubnikv, 2017-03-07:
I am trying to integrate the Intel Thread Building Blocks as they offer a much finer control of multi threading then just the boost.

This code is yet steaming hot. I set up our build servers to link TBB statically. There is no easy way to compile TBB statically from the original Intel sources, but a known scientist in the computer graphics community Wenzel Jakob  maintains a fork to be compiled with CMake, which generates these static libraries.

https://github.com/wjakob/tbb

I will appreciate though, if you could try to make it compile with the distro delivered TBB dynamically. You may need to fix xs/Build.PL

burbilog, 2017-03-07:
There is no tbb in ubuntu distro, I downloaded it from Intel's site and untarred it. 

Also I just tried to clone https://github.com/wjakob/tbb and while it compiles just fine (using make command), Slic3r build fails with the same error.

bubnikv, 2017-03-07:
I am building Slic3r with environment variable SLIC3R_STATIC=1
and with static boost.

I also had to add

set(CMAKE_POSITION_INDEPENDENT_CODE ON)

after the 2nd line of CMakeLists.txt, which is
project (tbb CXX)

to make the static library rellocatable, so it could be linked into a
static library.

But I think it should be easier to link TBB dynamically from the distro.
Are you sure there is no TBB for Ubuntu? There is one for Debian and there
is even one for OSX through brew.

burbilog, 2017-03-07:
Yeah, I see. I installed libtbb-dev and that fixed the problem.

You have to document build differences somehow, because now I see at least two different things: original Slic3r does NOT require eval anymore and now Prusa version requires  libtbb-dev. Some kind of  build faq is needed. This knowledge is kinda hard to obtain from scratch without lengthy reading issues...

bubnikv, 2017-03-07:
I agree I have to document it, but the day has only 24 hours.

> original Slic3r does NOT require eval anymore

What is it? Maybe I can get rid of that as well?

burbilog, 2017-03-07:
> What is it? Maybe I can get rid of that as well?

To build Prusa Slic3r I have to run

~~~
eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)     
~~~

Without that I get a lot of messages like this:

~~~
! Can't write to /usr/local/share/perl/5.22.1 and /usr/local/bin: Installing modules to /home/rm/perl5
! To turn off this warning, you have to do one of the following:
!   - run me as a root or with --sudo option (to install to /usr/local/share/perl/5.22.1 and /usr/local/bin)
!   - Configure local::lib in your existing shell to set PERL_MM_OPT etc.
!   - Install local::lib by running the following commands
!
!         cpanm --local-lib=~/perl5 local::lib && eval $(perl -I ~/perl5/lib/perl5/ -Mlocal::lib)
!
~~~

Original Slic3r does not require that anymore. Same system.

bubnikv, 2017-03-08:
Thanks. That is the "local" installation of the Perl libraries required by Slic3r. You can achieve a similar effect with perlbrew.

I may adapt this "local" approach one day, but it has not a priority to me as we have a reliable build system producing binary packages. But I understand that the approach is important for those wanting to co-develop or test early.

For those who want to test early, I may also put the development builds somewhere into the wild.

From a poor despair, I started to put together a windows binary distribution of Perl compiled with Visual Studio 2013, with which I found Slic3r runs some 10% faster, and it is certainly much easier to debug. The download links are at the top of 
https://github.com/prusa3d/Slic3r/wiki/How-to-compile-on-MS-Windows

But the page is far from being finished.

HaoTang1, 2017-03-09:
I'm having the same issue on Windows 8.1 64bit with the latest build 50976e1. 
I downloaded TBB from [tbb-2017_U5.7z ](https://bintray.com/bubnikv/wxPerl-for-Slic3r-with-VisualStudio/download_file?file_path=tbb-2017_U5.7z ). 
I tried to set the following environment variables: 
```
SET TBB-DIR=C:\local\tbb 
SET TBB_INCLUDEPATH=C:\local\tbb\include
SET TBB_LIBRARYPATH=C:\local\tbb\lib64-msvc-12.0\Release
```
and I'm still getting the same error. Am I doing something wrong here?

I also tried the TBB release from https://github.com/01org/tbb/releases but it didn't help. 

I saw installing libtbb-dev fixed the issue on Ubuntu but I'm not sure what shall I do on Windows. Any help?

Thanks.

bubnikv, 2017-03-09:
@HaoTang1 
Are you compiling with straberry perl or with visual studio?

For Strawberry Perl, you have to compile TBB yourself:
Pull https://github.com/wjakob/tbb 
Install cmake https://cmake.org/
Install some gnu make, for example you may install https://www.cygwin.com/
cd into the tbb directory, make a new directory, for example "outbuild"
cd outbuild
cmake -G "MinGW Makefiles" ..
then run make
This will generate the static and dynamic TBB libraries for the mingw compiler packed with the strawberry perl.

> I'm having the same issue on Windows 8.1 64bit with the latest build 50976e1.
> I downloaded TBB from tbb-2017_U5.7z .
This is supposed to be used with the Visual Studio 2013 community, best with my precompiled wperl packages.

HaoTang1, 2017-03-09:
@bubnikv 
I'm using Visual Studio Professional 2013 Version 12.0.21005.1 REL, with the 64bit release mode precompiled wperl package wperl64.

bubnikv, 2017-03-09:
@HaoTang1 
Please 
set SLIC3R_STATIC=1
this instruct Slic3r to link statically.

You will also have to set the environment for the static boost:
set BOOST_DIR=C:\local\boost_1_63_0
set BOOST_INCLUDEDIR=%BOOST_DIR%\
set BOOST_LIBRARYDIR=%BOOST_DIR%\lib64-msvc-12.0

supposed you installed
https://sourceforge.net/projects/boost/files/boost-binaries/1.63.0/boost_1_63_0-msvc-12.0-64.exe

HaoTang1, 2017-03-09:
@bubnikv 
Sorry that I should made it more clear. 

### Version
50976e1

### Operating system type + version
Windows 8.1 Pro 64 bit, Visual Studio Pro 2013 12.0.40629.00 Update 5 , release mode wperl64

Also tried on another server, Windows 10 Pro 64 bit, Visual Studio Pro 2013 12.0.21005.1 REL ,debug mode wperl64d

### Behavior
I saw your comment on [alexry #3744](https://github.com/alexrj/Slic3r/issues/3744) then what I did was basically following your instructions. 

1. Download your pre-compiled 64 bit release package wperl64 and extract it as c:\wperl64
 (Download your pre-compiled 64 bit debug package wperl64d and extract it as c:\wperl64d on the other server)
2. Download [boost](https://sourceforge.net/projects/boost/files/boost-binaries/1.63.0/b oost_1_63_0-msvc-12.0-64.exe) and extract the binary to c:\local
3. Download [tbb](https://bintray.com/bubnikv/wxPerl-for-Slic3r-with-VisualStudio/download_file?file_path=tbb-2017_U5.7z) and extract the binary to c:\local
4. Make sure everything is correct in the command line c:\wperl64\cmdline.lnk, then run it as  administrator. Check the environment variables are correctly set by echo each of them.
5. Pull Slic3r with `git clone https://github.com/prusa3d/Slic3r.git ` 
6. Build Slic3r with `perl Build.pl --sudo` , updtaing Moo failed. Run `cpanm Moo` , successfully update Moo to 2.003001. 
7. Rerun `perl Build.pl --sudo` , got the TBB error. 
8. Run `SET TBB_DIR=C:\local\tbb `, build again, same error.
9. Run `SET TBB_INCLUDEPATH=C:\local\tbb\include` and `SET TBB_LIBRARYPATH=C:\local\tbb\lib64-msvc-12.0\Release` (`SET TBB_LIBRARYPATH=C:\local\tbb\lib64-msvc-12.0\Debug` when building the debug version on the other server), build again, still same error. 

Step 4 to 9 are inside the same cmdline window so all the environment variables are correct.

I just removed everything and did all those steps again, still got the error.

I have the cmdline output for all steps in case you need it.
[slic3r.txt](https://github.com/prusa3d/Slic3r/files/832171/slic3r.txt)

Thanks.

HaoTang1, 2017-03-17:
Figured out that TBB_INCLUDEPATH and  TBB_LIBRARYPATH should be TBB_INCLUDEDIR and TBB_LIBRARYDIR.

In xs/build.pl, the error message `Or you may specify TBB_INCLUDEPATH and TBB_LIBRARYPATH separatly` doesn't match the code `if (defined $ENV{TBB_INCLUDEDIR}) `   `if (defined $ENV{TBB_LIBRARYDIR}) ` .

bubnikv, 2017-03-17:
@HaoTang1 I updated the error messages. Thanks for the hint.

I am leaving this issue open as a reminder, that I should update build instructions.

closed 2018-03-16 by vojtechkral