prusa3d/PrusaSlicer #178: The arrange button disables the zoom. Clicking on the blank path canvas re-enables it.

WhiteDragonD, 2017-03-10:
### Version
1.33.8

### Operating system type + version
Windows 7

### Behavior
This is the same issue as found in:
https://github.com/alexrj/Slic3r/issues/2424

I dragged/dropped a STL file into the window, clicked Arrange, then I lost the ability to scroll zoom with my mouse wheel.  When I clicked on the Print Settings tab then back to the Plater tab, this also lost the ability to zoom.  

You can refer to the other Slic3r Issue for more detailed information.

bubnikv, 2017-03-30:
The scrolling works only when the mouse cursor hovers over the 3d view. Does that work for you, or is there another issue I am not aware of?

WhiteDragonD, 2017-03-30:
https://github.com/alexrj/Slic3r/issues/2424

Look at that thread, it references the problem w/ the main Slic3r program and has little animations showing you how to reproduce it.

FidelCapo, 2018-12-12:
This should be fix in Slic3r PE 1.42 alpha, which is to be released soon. Closing.

closed 2018-12-12 by FidelCapo