prusa3d/PrusaSlicer #179: Variable Layer Editing - OpenGL Error

wywywywy, 2017-03-10:
### Version
1.33.8-prusa3d-win64

### Operating system type + version
Windows 7 x64

### Behavior
When I click on the "Layer editing" button, an error message shows up -

```
Cannot compile an OpenGL Shader, therefore the Variable Layer Editing will be disabled.

Fragment shader compilation failed:
No errors.
```

Then the button becomes disabled.

I already have the latest driver from Nvidia.

Any ideas?

Thanks.

faeton13, 2017-03-14:
same thing here. win10 x64 opengl 2.1

bubnikv, 2017-03-17:
The current code is patched using OpenGL 1.1 and outdated OpenGL extensions as this is the only OpenGL shader API supported by Perl. I am currently re-writing this piece of code into C++, so it will use OpenGL 2.0. This should fix most of these issues.

bubnikv, 2017-03-21:
Would you please try the following alpha build, whether this solves your issue with the variable layer height editing?

Just please understand that it is an alpha. There has been a lot of work done on it, which was not properly tested yet, for example paralellization of many slicing steps and a revamp of the OpenGL visualization code.

[Slic3r-1.33.8.124-prusa3d-win64-full-gd8f0c0b-201703201456.zip](https://github.com/prusa3d/Slic3r/files/857534/Slic3r-1.33.8.124-prusa3d-win64-full-gd8f0c0b-201703201456.zip)

wywywywy, 2017-03-21:
Hi,

Just tried it and the layer editing doesn't error out any more.  Hooray.

But the preview tab seems to have become very slow - is it related?

Thanks.

bubnikv, 2017-03-21:
> But the preview tab seems to have become very slow - is it related?

Which preview?

bubnikv, 2017-03-21:
I have rewritten the 3D visualization to use Vertex Buffer Objects and OpenGL 2.0 shaders. What graphics card are you using? Would you please post a log generated by slic3r help menu -> system info?

wywywywy, 2017-03-21:
See below.  It's using the Intel GPU rather than switching to the Nvidia.  Maybe that's the problem.

Slic3r Prusa Edition
Version: 1.33.8.125-prusa3d-win64
Build:   Slic3r-1.33.8.125-prusa3d-win64-full-g6b99cbd-201703210808
Operating System:    MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: 
Microsoft Windows [Version 6.1.7601]

Summary of my perl5 (revision 5 version 24 subversion 0) configuration:
   
  Platform:
    osname=MSWin32, osvers=10.0, archname=MSWin32-x64-multi-thread
    uname=''
    config_args='undef'
    hint=recommended, useposix=true, d_sigaction=undef
    useithreads=define, usemultiplicity=define
    use64bitint=define, use64bitall=undef, uselongdouble=undef
    usemymalloc=n, bincompat5005=undef
  Compiler:
    cc='cl', ccflags ='-nologo -GF -W3 -O1 -MD -Zi -DNDEBUG -GL -fp:precise -DWIN32 -D_CONSOLE -DNO_STRICT -DWIN64 -DCONSERVATIVE -D_CRT_SECURE_NO_DEPRECATE -D_CRT_NONSTDC_NO_DEPRECATE  -DPERL_TEXTMODE_SCRIPTS -DPERL_IMPLICIT_CONTEXT -DPERL_IMPLICIT_SYS -DWIN32_NO_REGISTRY',
    optimize='-O1 -MD -Zi -DNDEBUG -GL -fp:precise',
    cppflags='-DWIN32'
    ccversion='18.00.40629', gccversion='', gccosandvers=''
    intsize=4, longsize=4, ptrsize=8, doublesize=8, byteorder=12345678, doublekind=3
    d_longlong=undef, longlongsize=8, d_longdbl=define, longdblsize=8, longdblkind=0
    ivtype='__int64', ivsize=8, nvtype='double', nvsize=8, Off_t='__int64', lseeksize=8
    alignbytes=8, prototype=define
  Linker and Libraries:
    ld='link', ldflags ='-nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'
    libpth=\lib
    libs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    perllibs=oldnames.lib kernel32.lib user32.lib gdi32.lib winspool.lib  comdlg32.lib advapi32.lib shell32.lib ole32.lib oleaut32.lib  netapi32.lib uuid.lib ws2_32.lib mpr.lib winmm.lib  version.lib odbc32.lib odbccp32.lib comctl32.lib msvcrt.lib
    libc=msvcrt.lib, so=dll, useshrplib=true, libperl=perl524.lib
    gnulibc_version=''
  Dynamic Linking:
    dlsrc=dl_win32.xs, dlext=dll, d_dlsymun=undef, ccdlflags=' '
    cccdlflags=' ', lddlflags='-dll -nologo -nodefaultlib -debug -opt:ref,icf -ltcg  -libpath:"c:\wperl64\lib\CORE"  -machine:AMD64 -subsystem:console,"5.02"'

  @INC:
    C:/Users/wl/Desktop/Slic3r-1.33.8-prusa3d-win64-full-201703072158/lib
    .
OpenGL installation
  Using POGL v0.6704
  GL version:   4.0.0 - Build 9.18.10.3204
  vendor:       Intel
  renderer:     Intel(R) HD Graphics 4600
  GLSL version: 4.00 - Build 9.18.10.3204
Installed extensions (* implemented in the module):
  GL_3DFX_texture_compression_FXT1
  GL_ARB_ES2_compatibility
  GL_ARB_base_instance
  GL_ARB_blend_func_extended
* GL_ARB_color_buffer_float
  GL_ARB_compatibility
  GL_ARB_conservative_depth
  GL_ARB_copy_buffer
  GL_ARB_depth_buffer_float
  GL_ARB_depth_clamp
* GL_ARB_depth_texture
* GL_ARB_draw_buffers
  GL_ARB_draw_buffers_blend
  GL_ARB_draw_elements_base_vertex
  GL_ARB_draw_indirect
  GL_ARB_draw_instanced
  GL_ARB_explicit_attrib_location
  GL_ARB_fragment_coord_conventions
* GL_ARB_fragment_program
* GL_ARB_fragment_program_shadow
[... 93 lines cut ...]
  GL_EXT_texture_edge_clamp
  GL_EXT_texture_env_add
* GL_EXT_texture_env_combine
* GL_EXT_texture_filter_anisotropic
  GL_EXT_texture_integer
* GL_EXT_texture_lod_bias
  GL_EXT_texture_rectangle
  GL_EXT_texture_sRGB
  GL_EXT_texture_shared_exponent
  GL_EXT_texture_snorm
  GL_EXT_texture_storage
  GL_EXT_texture_swizzle
  GL_EXT_transform_feedback
  GL_IBM_texture_mirrored_repeat
  GL_INTEL_map_texture
  GL_INTEL_performance_queries
* GL_NV_blend_square
  GL_NV_conditional_render
  GL_NV_primitive_restart
* GL_NV_texgen_reflection
* GL_SGIS_generate_mipmap
  GL_SGIS_texture_edge_clamp
* GL_SGIS_texture_lod
  GL_SUN_multi_draw_arrays
  GL_WIN_swap_hint

wywywywy, 2017-03-21:
Ok I just updated the Intel driver and it's a lot better although not perfect.
Also the initialisation takes about 30s.

bubnikv, 2017-03-21:
If you have an NVIDIA card installed, then switching to it in the NVidia
application panel makes a huge difference. Just tried it today at one
co-worker's computer.

> Ok I just updated the Intel driver and it's a lot better although not
perfect.

The Intel card is not perfect. Are you saying that the alpha is worse than
the official 1.33.8 build?

> Also the initialisation takes about 30s.

Which initialization? At which step?

bubnikv, 2017-03-23:
@wywywywy Please provide additional information on what initialisation takes 30 seconds and whether it is worse than in 1.33.8. Otherwise I will close this issue.

I am going to release an official alpha soon, so we will collect more data on the behavior with the OpenGL vertex buffer objects.

bubnikv, 2017-03-30:
The variable layer height visualization has been rewritten to OpenGL 2.0. Please use the current alpha 1.34.0.

Closing. Please feel free to reopen this issue, if you still have problems.

closed 2017-03-30 by bubnikv