prusa3d/PrusaSlicer #180: Supports are not generated symmetrically

tcm0116, 2017-03-10:
### Version
1.33.8-prusa3d

### Operating system type + version
OS X 10.11.5

### Behavior
As can be seen in the images below, the supports are not symmetrical. The handle on the part is definitely symmetrical as it was modeled such that the left side is a mirror of the right. I'm using rectilinear grid for this particular part, but the issue remains regardless of the infill pattern selected (on an unrelated note, the pillars pattern results in a honeycomb patterned support).

[image]

[image]

#### STL/Config (.ZIP) where problem occurs
[Archive.zip](https://github.com/prusa3d/Slic3r/files/834147/Archive.zip)

landodragon141, 2017-03-11:
I'm sure someone else will chime in on this but I know from previous posts that the pillars pattern is not currently available with the new supports.

bubnikv, 2017-03-13:
The current support algorithm is not symmetrical as it discretizes into a regular grid and the model is most likely not symmetrical to the grid. I don't see it to be a problem.

tcm0116, 2017-03-13:
The model was centered on the build plate. Everything about the sliced output was symmetrical, including the infill, so I don't see why the supports were not symmetrical.

bubnikv, 2017-03-13:
The supports are generated relative to some grid, over which placement you have no control.

bubnikv, 2017-03-23:
I don't think this is an issue. Closing.

closed 2017-03-23 by bubnikv

tcm0116, 2017-03-23:
While it may not be a problem, I would still think that a symmetrical part centered on the bed should result in symmetrical support material in the same way that the infill is symmetrical. Perhaps this could be converted into a request?