prusa3d/PrusaSlicer #181: Unnecessary supports generated

tcm0116, 2017-03-10:
### Version
1.33.8-prusa3d

### Operating system type + version
OS X 10.11.5

### Behavior
Support material is generated where no support is required, as can be seen in the image below. In the attached configuration, I have "Support on build plate only" enabled, so definitely shouldn't be generating support material on top of the part, but it also generates this unnecessary material even if that option is not enabled.

[image]

#### STL/Config (.ZIP) where problem occurs
[Archive.zip](https://github.com/prusa3d/Slic3r/files/834175/Archive.zip)

werewolf1479, 2017-03-10:
I haven't seen this exact problem yet, but I have noticed it is a little bit liberal with its use of supports.  The ability to manually delete individual supports would go a long way to fix this if the algorithm is hard to change.

bubnikv, 2017-03-13:
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/66
Closing.

tcm0116, 2017-03-13:
I don't think this is a duplicate of #66 as the issue occurs regardless of weather "Support on build plate only" is enabled or not.

bubnikv, 2017-03-13:
The title of #66 is misleading as the content mixes two issues now. But
your issue is identified by my last comment at #66, namely:

It is not perfect, the supports leak through the walls to the other side.
This is due to the way how this algorithm inflates the support islands into
a rectangular grid. It is something I have on my list to improve, but I
felt to better provide an imperfect build to our customers now as IMHO it
is still much better than the previous one in most cases.

bubnikv, 2017-04-05:
Fixed with https://github.com/prusa3d/Slic3r/commit/6f5700a3a6241a56df9c7dcca88bbd05c224de98

closed 2017-04-05 by bubnikv