prusa3d/PrusaSlicer #182: Errors when starting Slic3r Appimage on Linux Mint 17.3

andershem, 2017-03-10:
### Version
Slic3r-1.33.8-prusa3d-linux64-full-201702210906.AppImage

### Operating system type + version
Linux Mint 17.3 Rosa

### Behavior
* _Describe the problem_
App image wont start, throws errors

* _Steps needed to reproduce the problem_
Downloaded appimage
Make it executable (chmod +x Slic3r-1.33.8-prusa3d-linux64-full-201702210906.AppImage)
Run the appimage (./Slic3r-1.33.8-prusa3d-linux64-full-201702210906.AppImage)
* _Expected Results_
I expected Slic3r to start

* _Actual Results_
Appimage default questions, then it throws errors in terminal

$ ./Slic3r-1.33.8-prusa3d-linux64-full-201702210906.AppImage 
Can't load '/tmp/.mount_dVcGS8/usr/bin/lib/site_perl/5.22.0/auto/Slic3r/XS/XS.so' for module Slic3r::XS: /usr/lib/x86_64-linux-gnu/libstdc++.so.6: version `CXXABI_1.3.8' not found (required by /tmp/.mount_dVcGS8/usr/bin/lib/site_perl/5.22.0/auto/Slic3r/XS/XS.so) at /tmp/.mount_dVcGS8/usr/bin/lib/site_perl/5.22.0/XSLoader.pm line 70.
 at /tmp/.mount_dVcGS8/usr/bin/lib/site_perl/5.22.0/Slic3r/XS.pm line 22.
Compilation failed in require at /tmp/.mount_dVcGS8/usr/bin/lib/site_perl/5.22.0/Slic3r.pm line 58.
BEGIN failed--compilation aborted at /tmp/.mount_dVcGS8/usr/bin/lib/site_perl/5.22.0/Slic3r.pm line 58.
Compilation failed in require at /tmp/.mount_dVcGS8/usr/bin/bin/slic3r.pl line 15.
BEGIN failed--compilation aborted at /tmp/.mount_dVcGS8/usr/bin/bin/slic3r.pl line 15.

probonopd, 2017-03-10:
This means that this AppImage requires a newer base system than you are running. If the Slic3r developers would use an older build system for producing the ingredients that go into the AppImage, then it might work with distributions like yours.

More on this on https://github.com/probonopd/AppImageKit/wiki/Creating-AppImages#creating-portable-appimages.

bubnikv, 2017-03-23:
@andershem why don't you upgrade your Linux Mint? I am hesitant to do through the loops of downgrading our build system and yours is the first request to do that.

probonopd, 2017-03-23:
Linux Mint 17.x is based on Ubuntu LTS 14.04 and both are supported (by the distro) until April 2019, so I suspect there are still some users with those versions out there -- although I have no indication on how many/few.

bubnikv, 2017-03-24:
Thanks @probonopd  for the clarification, though I don't currently have time to roll an older build server for Slic3r.

andershem, 2017-03-26:
@bubnikv i have more software in my computer then just Slic3r that i don't want to reconfigure all over again if i don't have to, software that works nice in mint 17. I used mint because i like the polish and it works great and its a LTS, ie. i don't want to reinstall my system every now and then, i would guess that you don't ask windows 8 user why they don't upgrade to windows 10 as a suggestion to a issue report?

bubnikv, 2017-03-27:
@andershem I understand your stance and I share your frustration, but
currently we don't have resources to downgrade the build system. You can
recompile from sources though.

probonopd, 2017-03-27:
@bubnikv since there is https://github.com/prusa3d/Slic3r/blob/master/.travis.yml - have you ever tried to build it on Travis CI?

https://travis-ci.org/getting_started

bubnikv, 2017-03-28:
@probonopd
Thanks for the hint, but as stated before, I don't currently have time to
look into that.

bubnikv, 2020-02-04:
Ubuntu 14 is no more supported.
https://www.omgubuntu.co.uk/2019/04/ubuntu-14-04-end-of-life
Closing.

closed 2020-02-04 by bubnikv