prusa3d/PrusaSlicer #184: Zero extrusion print moves and overlapping regions in first layer

DrLex0, 2017-03-13:
### Version
1.33.8-prusa3d

### Operating system type + version
Mac OS X 10.11.6

### Behavior
* When enabling the new type of supports on a [particular model](http://www.thingiverse.com/thing:112655), the first layer contains many print moves that do not extrude anything, as well as other moves that are incorrect.
* _Steps needed to reproduce the problem_
  * Slice the attached STL with the attached config. (The attached STL is a cleaned version with non-manifold triangles removed, but this makes no difference in the sliced result.) Look at the generated G-code file.
* _Expected Results_
  * Nothing weird happens.
* _Actual Results_
  * The skirt is ‘printed’ with all E0.0000 values (which is why gcode.ws shows those moves as travel moves), and the extruder also pretends to print parts of the support structure while the E coordinate is stuck at E32.47074. A different structure is then actually printed, overlapping with this non-printed area. If you look carefully, the area with zero extrusion is what should have been printed. The one that was really printed is wrong, it does not leave the requested 0.2mm Z distance with the object.

[image]

[image]

#### STL/Config (.ZIP) where problem occurs
[Config_and_STL.zip](https://github.com/prusa3d/Slic3r/files/838317/Config_and_STL.zip)

nub2, 2017-03-14:
Same skirt problem here with the [geodude](http://www.thingiverse.com/thing:205699)
Please try to enable Support material -> Syncronize with object layers
For me it helps...

bubnikv, 2017-03-24:
Thanks, I am seeing the issue.

bubnikv, 2017-03-27:
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/175

Hopefully fixed by https://github.com/prusa3d/Slic3r/commit/640698d28bc66a248199be95f70e2eed26bff6a8

bubnikv, 2017-03-30:
Fixed in 1.34.0

closed 2017-03-30 by bubnikv