prusa3d/PrusaSlicer #186: Smoothieware supports firmware retraction, but Slic3r throws the error "--use-firmware-retraction is only supported by Marlin firmware" when trying to use it

michaelnew, 2017-03-14:
### Version
version_1.33.8-114-ge7a920f

### Operating system type + version
macOS 10.12.1

### Behavior
Setting the G-code flavor to `Smoothie` and selecting `Use firmware retraction` causes Slic3r to throw the error `--use-firmware-retraction is only supported by Marlin firmware` and refuse to slice, even though Smoothie [does support](http://smoothieware.org/g10) firmware retraction.

[image]

ProfDrYoMan, 2017-03-25:
Repetier-Dev also supports firmware retraction.

lordofhyphens, 2017-03-29:
@michaelnew is this in the upstream versions as well (http://dl.slic3r.org/dev)? 

(It shouldn't, because I was the one who added the smoothieware flavor to upstream and I use firmware retraction all the time).

lordofhyphens, 2017-03-29:
@ProfDrYoMan Has there been a release of repetier firmware that has that support in it? Could you make an issue at https://github.com/alexrj/Slic3r/issues so I can remember to add it?

ProfDrYoMan, 2017-03-29:
No. There is no release of repetier firmware with this. Its in -dev.

michaelnew, 2017-03-29:
@lordofhyphens No, definitely not upstream (I just double checked to be sure). Just this fork.

bubnikv, 2017-03-30:
Fixed by cherrypicking 
https://github.com/alexrj/Slic3r/commit/99943024a9e3288cccaa6f15bc512dd95ea38216
thanks @lordofhyphens

closed 2017-03-30 by bubnikv