prusa3d/PrusaSlicer #187: [Request] General feedback and requests with are making irritating

Thirkell, 2017-03-16:
### Version
1.33.8

### Operating system type + version
Windows 10

### Behavior
Brief review of my first 2 days of using the Prusa Slicer:
Pros
Variable layer height is awesome
Maintain wall thickness is beyond epic
Price calculation saves me the mission of doing it myself for every print.
Really great work there you guys.

Cons
After slicing is done "Triangulating mesh bla bla" still stays at the bottom.
Volume of plastic is not really that useful in most cases. Knowing the meters  used is far better info to have and this unfortunately has to be worked out by after which is irritating.
Give a print time calculation. Pretty please.
Lastly: Display all this information as soon as slicing is done. Having to export the gcode before being able to see it is super irritating. 

_Is this a new feature request?_ YES

Keep up the awesome work!

bubnikv, 2017-03-23:
> After slicing is done "Triangulating mesh bla bla" still stays at the bottom.

This should be fixed in the current master.

> Volume of plastic is not really that useful in most cases. Knowing the meters used is far better info to have and this unfortunately has to be worked out by after which is irritating.

ok

> Give a print time calculation. Pretty please.

@lordofhyphens did some basic implementation of this in the @alexrj slic3r, but the estimate is very optimistic as it does not calculate with the accelerations / decelartions of the machine firmware.

> Lastly: Display all this information as soon as slicing is done. Having to export the gcode before being able to see it is super irritating.

An estimate of the material used may be possible before g-code export, but not the print time calculation, as the actual tool path planning is performed at the time of the G-code export.

lordofhyphens, 2017-03-23:
Actually it did attempt to calculate acceleration. It did not handle
retraction nor did it take into account things it doesn't know about.

bubnikv, 2017-03-24:
The correct solution is to integrate the path planner of the firmware and this is actually what Cura does. Points for Cura.

lordofhyphens, 2017-03-24:
Cura only really has to care about the firmware used by Ultimaker.
Obviously prusa3d only needs to care about its own printer's firmware.

Upstream gets used for a wide range of purposes. I am fine with being less
accurate in the time estimates (would prefer to be pessimistic than
optimistic, but that is harder) rather than integrate the motion planner
from every firmware we have flavor for.

bubnikv, 2017-03-24:
> Upstream gets used for a wide range of purposes. I am fine with being less
accurate in the time estimates (would prefer to be pessimistic than
optimistic, but that is harder) rather than integrate the motion planner
from every firmware we have flavor for.

Most of the firmwares are based on GRBL. My guess is, integrating the GRBL
planner into Slic3r would be always better than a naive estimator.

Sebastianv650, 2017-03-24:
The recent grbl version has not much todo with the version used and modified since this time, at least in Marlin. Maybe parts of the planner are still simmilar, but I would be careful.

Maybe it would be much more easy and at a sufficient quality to allow the user to enter a simple speed factor in %? At least for me, I recognised the time given with Pronterface (which isn't taking acceleration etc into account) is always off by aproximately a fixed factor. Not at the minute, but on a print taking 3h +-10min is more than enough for my needs.

lordofhyphens, 2017-03-24:
Upstream just added an estimator to libslic3r, based on my earlier pull
request. It's currently called through a post-process script.

I'll need to get the support added to the UI at some point.

"Nothing unreal exists." - Kiri-kin-tha's First Law of Metaphysics.

Thirkell, 2017-03-25:
Even a vague time estimate is helpful so plan out prints better. Repitier has a nice solution where you add in a % compensation factor for the print time, like @Sebastianv650 Suggested. It still tends to be optimistic and never takes into account retractions, accelerations or z-hop but it something and one can get prettty close by adjusting that offset your self.

bubnikv, 2017-12-05:
Closing this issue as there is a duplicate https://github.com/prusa3d/Slic3r/issues/145 still open.

closed 2017-12-05 by bubnikv