prusa3d/PrusaSlicer #188: Failed to build 8e28bd36a24d2cd29ed3502ff201701a9029e0ec

burbilog, 2017-03-17:
### Version
8e28bd36a24d2cd29ed3502ff201701a9029e0ec

### Operating system type + version
Ubuntu 14.04.5 LTS

### Behavior
Failed to compile:

~~~
cc -I/usr/lib/perl/5.18/CORE -fPIC -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -I/usr/lib/x86_64-linux-gnu/wx/include/gtk2-unicode-release-2.8 -I/usr/include/wx-2.8 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -std=c++11 -DSLIC3R_GUI -DSLIC3R_PRUS -DUNICODE -D_FILE_OFFSET_BITS=64 -D_LARGE_FILES -D__WXGTK__ -pthread -DBOOST_LOG_DYN_LINK -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/libslic3r/Format/PRUS.o src/libslic3r/Format/PRUS.cpp
src/libslic3r/Format/PRUS.cpp: In function 'bool Slic3r::load_prus(const char*, Slic3r::Model*)':
src/libslic3r/Format/PRUS.cpp:118:40: error: conversion from 'const char*' to 'const wxString' is ambiguous
     wxFFileInputStream          in(path);
                                        ^
src/libslic3r/Format/PRUS.cpp:114:6: note: candidates are:
 bool load_prus(const char *path, Model *model)
      ^
In file included from src/libslic3r/Format/PRUS.cpp:5:0:
/usr/include/wx-2.8/wx/string.h:692:3: note: wxString::wxString(wxChar, size_t) <near match>
   wxString(wxChar ch, size_t nRepeat = 1)
   ^
/usr/include/wx-2.8/wx/string.h:692:3: note:   no known conversion for argument 1 from 'const char*' to 'wxChar {aka wchar_t}'
/usr/include/wx-2.8/wx/string.h:682:3: note: wxString::wxString(int) <near match>
   wxString(int);
   ^
/usr/include/wx-2.8/wx/string.h:682:3: note:   no known conversion for argument 1 from 'const char*' to 'int'
In file included from src/libslic3r/Format/PRUS.cpp:6:0:
/usr/include/wx-2.8/wx/wfstream.h:135:5: error:   initializing argument 1 of 'wxFFileInputStream::wxFFileInputStream(const wxString&, const wxChar*)'
     wxFFileInputStream(const wxString& fileName, const wxChar *mode = wxT("rb"));
     ^
src/libslic3r/Format/PRUS.cpp:124:21: error: invalid conversion from 'const char*' to 'wxChar {aka wchar_t}' [-fpermissive]
         if (name == "scene.xml") {
                     ^
In file included from src/libslic3r/Format/PRUS.cpp:5:0:
/usr/include/wx-2.8/wx/string.h:1634:13: error:   initializing argument 2 of 'bool operator==(const wxString&, wxChar)' [-fpermissive]
 inline bool operator==(const wxString& s, wxChar c) { return s.IsSameAs(c); }
             ^
src/libslic3r/Format/PRUS.cpp:144:38: error: no matching function for call to 'wxString::EndsWith(const char [5])'
         else if (name.EndsWith(".stl") || name.EndsWith(".STL")) {
                                      ^
src/libslic3r/Format/PRUS.cpp:144:38: note: candidate is:
In file included from src/libslic3r/Format/PRUS.cpp:5:0:
/usr/include/wx-2.8/wx/string.h:1126:8: note: bool wxString::EndsWith(const wxChar*, wxString*) const
   bool EndsWith(const wxChar *suffix, wxString *rest = NULL) const;
        ^
/usr/include/wx-2.8/wx/string.h:1126:8: note:   no known conversion for argument 1 from 'const char [5]' to 'const wxChar* {aka const wchar_t*}'
src/libslic3r/Format/PRUS.cpp:144:63: error: no matching function for call to 'wxString::EndsWith(const char [5])'
         else if (name.EndsWith(".stl") || name.EndsWith(".STL")) {
                                                               ^
src/libslic3r/Format/PRUS.cpp:144:63: note: candidate is:
In file included from src/libslic3r/Format/PRUS.cpp:5:0:
/usr/include/wx-2.8/wx/string.h:1126:8: note: bool wxString::EndsWith(const wxChar*, wxString*) const
   bool EndsWith(const wxChar *suffix, wxString *rest = NULL) const;
        ^
/usr/include/wx-2.8/wx/string.h:1126:8: note:   no known conversion for argument 1 from 'const char [5]' to 'const wxChar* {aka const wchar_t*}'
src/libslic3r/Format/PRUS.cpp:146:19: error: 'wxScopedCharBuffer' does not name a type
             const wxScopedCharBuffer name_utf8 = name.ToUTF8();
                   ^
src/libslic3r/Format/PRUS.cpp:148:60: error: 'name_utf8' was not declared in this scope
             sprintf(model_name_tag, "<model name=\"%s\">", name_utf8.data());
                                                            ^
src/libslic3r/Format/PRUS.cpp:200:15: error: 'class wxZipInputStream' has no member named 'ReadAll'
       if (zip.ReadAll((void*)stl.facet_start, 50 * header.nTriangles)) {
               ^
src/libslic3r/Format/PRUS.cpp:253:87: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
                         int res_outer_loop = line_reader.next_line_scanf(" outer loop");
                                                                                       ^
src/libslic3r/Format/PRUS.cpp:255:131: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
       int res_vertex1 = line_reader.next_line_scanf(" vertex %f %f %f", &facet.vertex[0].x, &facet.vertex[0].y, &facet.vertex[0].z);
                                                                                                                                   ^
src/libslic3r/Format/PRUS.cpp:257:131: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
       int res_vertex2 = line_reader.next_line_scanf(" vertex %f %f %f", &facet.vertex[1].x, &facet.vertex[1].y, &facet.vertex[1].z);
                                                                                                                                   ^
src/libslic3r/Format/PRUS.cpp:259:131: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
       int res_vertex3 = line_reader.next_line_scanf(" vertex %f %f %f", &facet.vertex[2].x, &facet.vertex[2].y, &facet.vertex[2].z);
                                                                                                                                   ^
src/libslic3r/Format/PRUS.cpp:261:63: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
       int res_endloop = line_reader.next_line_scanf(" endloop");
                                                               ^
src/libslic3r/Format/PRUS.cpp:263:65: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
       int res_endfacet = line_reader.next_line_scanf(" endfacet");
                                                                 ^
error building src/libslic3r/Format/PRUS.o from 'src/libslic3r/Format/PRUS.cpp' at /usr/share/perl/5.18/ExtUtils/CBuilder/Base.pm line 177.
FAIL
! Installing ./xs failed. See /home/rm/.cpanm/work/1489744204.842/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting

~~~

#### STL/Config (.ZIP) where problem occurs
No config.

bubnikv, 2017-03-17:
What wxWidgets library are you using? I am working with 3.0.3-beta as it supports anti-aliased OpenGL widgets.

burbilog, 2017-03-18:
~~~ 
dragon pts/10%dpkg -l|grep -i wx                                                              [~ 11:04 Сб. 18]
ii  libalien-wxwidgets-per 0.64+dfsg-2      amd64            Perl module for locating wxWidgets binaries
ii  libwx-perl             1:0.9922-2       amd64            interface to wxWidgets cross-platform GUI toolkit
ii  libwxbase2.8-0:amd64   2.8.12.1+dfsg-2u amd64            wxBase library (runtime) - non-GUI support classe
ii  libwxbase2.8-dev       2.8.12.1+dfsg-2u amd64            wxBase library (development) - non-GUI support cl
ii  libwxbase3.0-0:amd64   3.0.2-1~ubuntu14 amd64            wxBase library (runtime) - non-GUI support classe
ii  libwxbase3.0-dev       3.0.2-1~ubuntu14 amd64            wxBase library (development) - non-GUI support cl
ii  libwxgtk-media2.8-0:am 2.8.12.1+dfsg-2u amd64            wxWidgets Cross-platform C++ GUI toolkit (GTK+ me
ii  libwxgtk-media2.8-dev  2.8.12.1+dfsg-2u amd64            wxWidgets Cross-platform C++ GUI toolkit (GTK+ me
ii  libwxgtk2.8-0:amd64    2.8.12.1+dfsg-2u amd64            wxWidgets Cross-platform C++ GUI toolkit (GTK+ ru
ii  libwxgtk2.8-dev        2.8.12.1+dfsg-2u amd64            wxWidgets Cross-platform C++ GUI toolkit (GTK+ de
ii  libwxgtk3.0-0:amd64    3.0.2-1~ubuntu14 amd64            wxWidgets Cross-platform C++ GUI toolkit (GTK+ ru
ii  libwxgtk3.0-dev        3.0.2-1~ubuntu14 amd64            wxWidgets Cross-platform C++ GUI toolkit (GTK+ de
ii  python-wxglade         0.6.8-1          all              GUI designer written in Python with wxPython
ii  python-wxgtk2.8        2.8.12.1+dfsg-2u amd64            wxWidgets Cross-platform C++ GUI toolkit (wxPytho
ii  python-wxgtk3.0        3.0.1.1+dfsg-2~u amd64            Python interface to the wxWidgets Cross-platform
ii  python-wxversion       3.0.1.1+dfsg-2~u all              API for selecting the wxPython version to use
ii  wx-common              3.0.2-1~ubuntu14 amd64            wxWidgets Cross-platform C++ GUI toolkit (common
ii  wx2.8-doc              2.8.12.1+dfsg-2u all              wxWidgets Cross-platform C++ GUI toolkit (documen
ii  wx2.8-headers          2.8.12.1+dfsg-2u amd64            wxWidgets Cross-platform C++ GUI toolkit (header
ii  wx3.0-headers          3.0.2-1~ubuntu14 amd64            wxWidgets Cross-platform C++ GUI toolkit (header
~~~

burbilog, 2017-03-20:
~~~
dragon pts/13%git bisect bad                                                                                     [~/slic3r.prusa/Slic3r 13:06 Пн. 20] 
7f7d2da5fe550a256169f9b806369833a8aafbf8 is the first bad commit
commit 7f7d2da5fe550a256169f9b806369833a8aafbf8
Author: bubnikv <bubnikv@gmail.com>
Date:   Thu Mar 16 14:02:28 2017 +0100

    Use Vertex Buffer Objects for rendering of 3D volumes if possible.

:040000 040000 8c7b6438c5547e7f204c80ead9d01a8505968262 a5e1ff5e96daf9501bc63c9fa7de62fb53a159bf M      lib
:040000 040000 eb24984c9ffacbd57fb042823d3581e048efe579 06d95fe4ac1c3cdd41233adf09b84b485f035aea M      xs
~~~

bubnikv, 2017-03-20:
What do you get when you execute

perl -e "use Alien::wxWidgets; print Alien::wxWidgets->shared_library_path"

On my windows box this gives
c:\wperl64d\site\lib\Alien\wxWidgets\msw_3_0_3_uni_cl_8\lib

so I know I am linking against wxWidgets 3.0.3.

burbilog, 2017-03-20:
~~~
dragon pts/13%perl -e "use Alien::wxWidgets; print Alien::wxWidgets->shared_library_path"                        [~/slic3r.prusa/Slic3r 17:40 Пн. 20] 
Can not use 'shared_library_path' at -e line 1.
~~~

bubnikv, 2017-03-23:
You have the wxwidgets 3.0 installed. I suppose the package libalien-wxwidgets-perl and libwx-perl likely pull the wxwidgets 2.8. I would recommend to deinstall the two and run
perl Build.PL --gui 
in the Slic3r directory. I hope this will force perl to pull the alien-wx and wx modules and compile them from sources, which in turn should link against wxwidgets 3.0.

Also the current build has this piece of code disabled by default, so you should be fine with the current master.

burbilog, 2017-03-23:
I removed these two libraries, now it fails immediately:

~~~
Configuring /home/rm/slic3r.prusa/Slic3r/xs ... Slic3r will be built with GUI support
Can't locate Alien/wxWidgets.pm in @INC (you may need to install the Alien::wxWidgets module) (@INC contains: /home/rm/perl5/lib/perl5/5.18.2/x86_64-linux-gnu-thread-multi /home/rm/perl5/lib/perl5/5.18.2 /home/rm/perl5/lib/perl5/x86_64-linux-gnu-thread-multi /home/rm/perl5/lib/perl5 /etc/perl /usr/local/lib/perl/5.18.2 /usr/local/share/perl/5.18.2 /usr/lib/perl5 /usr/share/perl5 /usr/lib/perl/5.18 /usr/share/perl/5.18 /home/rm/perl5/lib/perl5/5.18.1 /usr/local/lib/site_perl .) at Build.PL line 54.
N/A
! Configure failed for ./xs. See /home/rm/.cpanm/work/1490303297.1413/build.log for details.
The XS/C++ code failed to compile, aborting
~~~

bubnikv, 2017-03-24:
You may need to install the two modules manually:

cpan install Alien
cpan install Alien::wxWidgets
cpan install Wx

or you may follow the following:
http://www.wxperl.it/p/download.html

burbilog, 2017-03-24:
Installed these three modules. Still the same problem:
~~~
src/admesh/stlinit.c: In function 'void stl_read(stl_file*, int, int)':
src/admesh/stlinit.c:277:29: error: 'assert' was not declared in this scope
       assert(res_normal == 3);
                             ^
src/admesh/stlinit.c:273:36: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
       fscanf(stl->fp, "endsolid\n");
                                    ^
src/admesh/stlinit.c:274:40: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
       fscanf(stl->fp, "solid%*[^\n]\n");  // name might contain spaces so %*s doesn't work and it also can be e
                                        ^
error building src/admesh/stlinit.o from 'src/admesh/stlinit.c' at /usr/share/perl/5.18/ExtUtils/CBuilder/Base.pm line 177.
FAIL
! Installing ./xs failed. See /home/rm/.cpanm/work/1490391678.12668/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting
dragon pts/10%perl -e "use Alien::wxWidgets; print Alien::wxWidgets->shared_library_path"
Can not use 'shared_library_path' at -e line 1.
~~~

burbilog, 2017-03-25:
Ugh! I just realized, now it's not the same problem. It seems that something another happened during last week and now it can't build stlinit.c...

burbilog, 2017-03-29:
@bubnikv as of 4bbb1f4b63c7533029397b0b502c91ffa76badbf build fails with another message:

~~~
cc -I/usr/lib/perl/5.18/CORE -fPIC -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -I/usr/lib/x86_64-linux-gnu/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -std=c++11 -DSLIC3R_GUI -DSLIC3R_PRUS -DUNICODE -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -DBOOST_LOG_DYN_LINK -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fstack-protector -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/admesh/stlinit.o src/admesh/stlinit.c
src/admesh/stlinit.c: In function 'void stl_read(stl_file*, int, int)':
src/admesh/stlinit.c:277:29: error: 'assert' was not declared in this scope
       assert(res_normal == 3);
                             ^
src/admesh/stlinit.c:273:36: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
       fscanf(stl->fp, "endsolid\n");
                                    ^
src/admesh/stlinit.c:274:40: warning: ignoring return value of 'int fscanf(FILE*, const char*, ...)', declared with attribute warn_unused_result [-Wunused-result]
       fscanf(stl->fp, "solid%*[^\n]\n");  // name might contain spaces so %*s doesn't work and it also can be empty (just "solid")
                                        ^
error building src/admesh/stlinit.o from 'src/admesh/stlinit.c' at /usr/share/perl/5.18/ExtUtils/CBuilder/Base.pm line 177.
FAIL
! Installing ./xs failed. See /home/rm/.cpanm/work/1490800160.23177/build.log for details. Retry with --force to force install it.
The XS/C++ code failed to compile, aborting
~~~

It seems that THIS problem began at this commit:

~~~
1fb3d009320ab9b0bee2e0a5219079bc378cfca6 is the first bad commit
commit 1fb3d009320ab9b0bee2e0a5219079bc378cfca6
Author: bubnikv <bubnikv@gmail.com>
Date:   Wed Mar 22 17:57:27 2017 +0100

    Fix of parsing ASCII STLs on Visual Studio 2013 with global optimization.

:040000 040000 f9772d5f37e3407848452b9ec0387ef6fd94af3e 02b7f03204dd89fd61225d3550ff11afacd65dff M      xs
~~~

The trouble is, until this new bug is fixed, I can't check for original bug.... :-O

bubnikv, 2017-03-29:
This should fix the last error (missing assert)
https://github.com/prusa3d/Slic3r/commit/b60ae4745f0c95dd6f8a5960b9827e3d1b4e0b42

burbilog, 2017-03-29:
Yes, now it builds OK.

closed 2017-03-29 by burbilog