prusa3d/PrusaSlicer #189: OpenGL Shader Bug with smooth layer option activate

Tesla3DTore, 2017-03-17:
### Version
_Version of Slic3r 1.33.8_prusa3d_win64 from prusa website 1.7.6 package

### Operating system type + version
===================================================
GPU Caps Viewer v1.9.4
http://www.ozone3d.net/gpu_caps_viewer/
===================================================

===================================[ System / CPU ]
- CPU Name: Intel(R) Pentium(R) CPU B960 @ 2.20GHz
- CPU Core Speed: 2195 MHz
- CPU Num Cores: 8
- Family: 6 - Model: 10 - Stepping: 7
- Physical Memory Size: 16290 MB
- Operating System: Windows 7 64-bit build 7601 [Service Pack 1]
- DirectX Version: 10.0
- PhysX Version: drivers not installed

===================================[ Graphics Adapter / GPU ]
- OpenGL Renderer: Intel(R) HD Graphics
- Drivers Renderer: Intel(R) HD Graphics
- DB Renderer: Intel HD Graphics 100/200
- Device Description: 
- Adapter String: Intel(R) HD Graphics
- Vendor: Intel
- Vendor ID: 0x8086
- Device ID: 0x0106
- Sub device ID: 0x1032
- Sub vendor ID: 0x1043
- Drivers Version: 9.17.10.4229 (5-26-2015) - ig4icd64.dll
- GPU Codename: HD G100/G200
- GPU Unified Shader Processors: 12
- GPU Vertex Shader Processors: 0
- GPU Pixel Shader Processors: 0
- Video Memory Size: -1988 MB
- Video Memory Type: unknown
- BIOS String: Intel Video BIOS
- Current Display Mode: 1600x900 @ 60 Hz - 32 bpp

===================================[ OpenGL GPU Capabilities ]
- OpenGL Version: 3.1.0 - Build 9.17.10.4229
- GLSL (OpenGL Shading Language) Version: 1.40 - Intel Build 9.17.10.4229
- ARB Texture Units: 8
- Vertex Shader Texture Units: 16
- Pixel Shader Texture Units: 16
- Geometry Shader Texture Units: 0
- Max Texture Size: 8192x8192
- Max Anisotropic Filtering Value: X16.0
- Max Point Sprite Size: 255.0
- Max Dynamic Lights: 8
- Max Viewport Size: 8192x8192
- Max Vertex Uniform Components: 512
- Max Fragment Uniform Components: 1024
- Max Geometry Uniform Components: 0
- Max Varying Float: 41
- Max Vertex Bindable Uniforms: 0
- Max Fragment Bindable Uniforms: 0
- Max Geometry Bindable Uniforms: 0
- Frame Buffer Objects (FBO) Support:[yes]
- Multiple Render Targets / Max draw buffers: 8
- Pixel Buffer Objects (PBO) Support:[yes]
- S3TC Texture Compression Support:[yes]
- ATI 3Dc Texture Compression Support:[no]
- Texture Rectangle Support:[yes]
- Floating Point Textures Support:[no]
- MSAA: 4X
- OpenGL Extensions: 130 extensions (GL=116 and WGL=14)
    <li>GL_EXT_blend_minmax</li>
    <li>GL_EXT_blend_subtract</li>
[... 171 lines cut ...]
	- CL_DEVICE_PREFERRED_VECTOR_WIDTH_INT: 4
	- CL_DEVICE_PREFERRED_VECTOR_WIDTH_LONG: 2
	- CL_DEVICE_PREFERRED_VECTOR_WIDTH_FLOAT: 4
	- CL_DEVICE_PREFERRED_VECTOR_WIDTH_DOUBLE: 2
	- CL_DEVICE_EXTENSIONS: 11
	- Extensions:
		- cl_khr_fp64
		- cl_khr_icd
		- cl_khr_global_int32_base_atomics
		- cl_khr_global_int32_extended_atomics
		- cl_khr_local_int32_base_atomics
		- cl_khr_local_int32_extended_atomics
		- cl_khr_byte_addressable_store
		- cl_intel_printf
		- cl_ext_device_fission
		- cl_intel_exec_by_local_thread
		- cl_khr_gl_sharing

===================================[ Misc. ]

===================================[ Related Graphics Drivers ]
- http://www.geeks3d.com/?page_id=752

===================================[ Related Graphics Cards Reviews ]
- http://www.geeks3d.com/?cat=3

bubnikv, 2017-03-17:
I am working on the shaders part right now. The current code was written using an outdated OpenGL 1.1 + extensions as nothing else was supported by the Perl package. Now I am getting the shaders implemented in C++, therefore I am using the proper OpenGL 2.0 interface and I expect many of the issues to disappear.

bubnikv, 2017-03-17:
Also the new code renders much faster on OpenGL 2.0 capable machines, and the 3D print path preview consumes around 1/3 of RAM compared to the original code. Stay tuned.

bubnikv, 2017-03-17:
This is a duplicate of https://github.com/prusa3d/Slic3r/issues/179
Closing.

closed 2017-03-17 by bubnikv