prusa3d/PrusaSlicer #190: [request] XY size Compensation for first layer

supermerill, 2017-03-17:
Hi
Actually, i remove manually the last perimeter of my first layer to avoid the "elephant foot" problem (first layer is squished on the plate, so it's wider than what the slicer think).

I will be so much more comfortable to have a setting with a much more precise value, it's like the XY compensation but only for the first layer.

Sebastianv650, 2017-03-17:
Calibrating the z offset of your printer is the right way to do this. If your first layer is squished into the bed, your nozzle is too close to the print bed.

The easiest way to find the right value (in my opinion) is to print a single line circle or something simmilar. Measure the line height with a caliper and calculate the delta to the wanted value (your first layer height).

burbilog, 2017-03-17:
> Calibrating the z offset of your printer is the right way to do this. If your first layer is squished into the bed, your nozzle is too close to the print bed.

In a perfect world, yes. 

In reality it's often a very tough choice between adhesion and squishing....

bubnikv, 2017-03-17:
This request makes sense, only it adds another parameter to the UI and another code complexity.

Sebastianv650, 2017-03-18:
Not only in a perfect world. You can achieve the same thing with a correctly adjusted z height, setting a lower 1st layer height and a wider first layer extrusion width.
And this even has the advantage you will not damage your bed with the nozzle when you start printing at a lower layer height some day. If you do at as you want it now, for example you set a first layer height of 0.2mm but in reality you are only 0.1mm away from the bed. If you want to print at 0.1mm height some day, you start at 0mm above the bed, meaning you nozzle is scratching the surface!

Sebastianv650, 2017-03-18:
And by using the correct z calibration with wider 1st layer, you will also not have a elephant food.
Therefore I'm clearly against this "enhancement", it's only a work around for users not knowing how to to it right in my eyes..

supermerill, 2017-03-20:
note: I am talking about the little bit of plastic that is not perfectly aligned in the first layer an i have to cut with a cutter (not the multi-layer big elephant foot).

>> If you want to print at 0.1mm height some day, you start at 0mm above the bed, meaning you nozzle is scratching the surface!
Or not if your firmware has a safeguard. I'm printing at 0.2mm & 0.1mm first layer anyway. At 0.1mm, the heatbed un-flatness force me to be at a certain height to have a good enough adhesion where the bed is too low and it's very squished where it's too high. I have already tried to level the bed with screw & meshbed tuning of my firmware.

I will try to increase the first layer width even more (i was printing first layer at 0.5 vs 0.4 nozzle).

Opinion from other people:

http://support.3dverkstan.se/article/23-a-visual-ultimaker-troubleshooting-guide#elephant
"It's hard to get rid of this effect entirely without sacrificing bottom layer quality and bed adhesion"

https://forum.simplify3d.com/viewtopic.php?f=23&t=5042
"I second this. The first layer is inherently bigger by design."

bubnikv, 2017-03-20:
> troubleshooting-guide#elephant

As the day has only 24 hours, I think I should set up some voting system to
prioritize the features :-)

werewolf1479, 2017-03-20:
Honestly since it is only a couple of you guys coding at most, you should probably just determine in-house which ones you want to prioritize.  Trying to make it a democracy will just give you a headache down the road.  

That being said, Sebastian seems to have taken this suggestion personally to be so adamantly against it.  The only thing I can think of is his family was hurt by XY compensation at some point.

bubnikv, 2017-03-20:
I personally am for the feature in general as I like to rather squish the 1st layer strongly to get a good bonding to the PEI sheet. What speaks against this feature is the "feature bloat", meaning there will be new configs, new code etc. Look at the Cura engine.

werewolf1479, 2017-03-20:
I am also a squisher lol so I would use this feature as well.  Yeah feature bloat is something I deal with at my work a lot.  Our users are all internal though and I make ample use of the "advanced" tabs since that is usually a "hey, you know you are getting into the weeds at this point, user beware" kind of thing.

Sebastianv650, 2017-03-20:
Don't get me wrong, I will not sleep bad if bubnikv implement such a scale option ;) I only think that it's a waste of time because you can do the same thing with existing structures.

> Or not if your firmware has a safeguard.

This will only work if the FW has your real nozzle Z position. If you get an overall squished first layer (local ones due a not flat bed is another story of course), this means the Z offset is not calibrated and therefore your FW will not safe your bed - it will think there is some room left.

It's definitly possible to print without that squish, I'm doing it since I learned about the first layer problems on my TAZ (glass bed with PEI sheet).

The squish we are talking about to keep the print sticking to the print bed is nothing different than a layer height to line width ratio. For example 0.2mm layer height with 0.5mm line width might not stick to the bed. 
- One "solution" is to lower the z offset for example by 0.05mm - this ends up with the printer extruding the filament volume needed für a 0.2x0.5mm path, but with the nozzle sitting only 0.15mm above the bed. Now the print will stick, but you have a small elephant food on the first layer.
- But you can do the same by leaving the z offset alone. If you set the first layer height to 0.15, and combine it with a line width of 0.67mm this results in an equal squish (0.2*0.5/0.15), but now without an elephant food because the slicer knows where the nozzle is sitting around.

This are my 2 cents, now it's up to you to choose which direction you want to go.

supermerill, 2017-03-22:
> - But you can do the same by leaving the z offset alone. If you set the first layer height to 0.15, and combine it with a line width of 0.67mm this results in an equal squish (0.2*0.5/0.15), but now without an elephant food because the slicer knows where the nozzle is sitting around.

I tried this, and it work (now i have plenty of calibraiton cube :p ). But now my z dimension is ~0.1mm too tall (before it was perfect). There are a settings for z-compensation? It's the "first layer height" i think. I will make other cubes this evening.

Sebastianv650, 2017-03-22:
One of the most important things in 3D printer calibration is to get your Z-calibration perfect. This means, if you don't set a Z offset in Slic3r and you print a 0.2mm single line on the print bed for example, it should measure also 0.2mm using a caliper. Personal, I try to hit it <+-0.05mm, the rest can be done using slic3rs offset which is more easy to use than a thumb screw in my case.

If you have that done, your z dimension should be dead on. If not, the only thing that could be of is the steps/mm setting in your printer FW.

shanemgrey, 2017-03-23:
I suspect there may be more going on here that is triggering this request in the first place.  

I'm trying to print very small precise parts for a project and I'm getting what I can only figure is increased flow rate on the first layer compared to all other layers, even with identical settings for all layers.

Example:

First layer: 0.2mm height, 0.5mm width
All other layers: 0.2mm height, 0.5mm width

Print object 1x10x1

Result: 
object 1x10x1mm except the first layer which is 10.4 x 1.4 on x and y.  

Thinking maybe the first layer is not actually 0.2mm, I print just that layer and it is around 0.201mm.  Do it a few more times and get results +- 0.005

If I print 10 layers at 0.2mm each, I get a 2mm tall object =- 0.01mm.  20 layers gives a 4mm object.  It's quite accurate.  

So I'm convinced that my z height is correctly calibrated.  0.2mm produces 0.2mm height.  

But the first layer is always wider than all subsequent layers by a big margin unless I raise the nozzle physically.  But that makes the object height wrong.  

And if I change the settings in slic3r to make the width deliberately wider for the same height, it does just that.  but it draws fewer thicker lines.  And the squish out the sides is still there.

The only hack that would work properly would be to change the physical object to remove 0.2mm from the outside edge of the object for only the first layer.  And that amount would change with changing layer heights...  Not a fun hack. 

I am suspecting the firmware at this point.  It seems wrong that firmware would alter the gcode in any way.  But I can't find any other place where this might be coming from.  

I verified that the gcode for the first layer is identical to other layers when the settings are the same.  Yet the physical result of the first layer seems to be extruding about 30% more volume for the same gcode.

Being that this isn't the repo for the firmware, my comment might be misplaced as a suggestion.  

But I think it's relevant if the reason for this enhancement request is to compensate for misplaced behavior in firmware.   In my case it's the current version of firmware for the prusa i3 mk2.  And maybe there is some communication between the prusa slic3r team and the prusa firmware team?

Sebastianv650, 2017-03-23:
That's very strange, I never saw something you desicribed before. When you print it 5x10x1mm, will it also be 5.4x10.4x1? I ask because a 1.4mm wide object with 0.5mm line width might disort the perimeter line when the gap fill does something wrong. On the following layers, maybe the gap fill has more options to fill small gaps between the layers therefore the dimension is more accurate..

shanemgrey, 2017-03-23:
If I send 5x10x1mm, it will be 5.4x10.4x1.  Yes.  

I'm not 100% confident that this isn't some physical thing I'm failing to account for.  But I thought of a test that would make it definitive.  I'll print an object with a single layer height and the same one with two layers of identical settings.  Then I'll weigh the objects.  If the second is less than twice as heavy, it's clearly extruding more for the first layer for some reason.  If not, I'm stumped for how to calibrate away that 30% error.

I need to get a microgram scale first. 

My test object over a week ago was 1.5mm x 10mm x 10mm

I set it to 0.5mm width 0.2mm height.  Same for first and all other layers.

At this dimension, there was no infill.  All solid walls.

Printing the whole object resulted in 1.9mm x 10.4mm for the first layer and 1.5mm x 10mm for all other layers.  The object was 10mm tall.

Printing the first layer only (same object, cut in slicer to 0.2mm) resulted in 1.9mm x 10.4mm x 0.2mm.

All measured on a micrometer, not a caliper.

The printer could be seen to be making only one layer for this second print. 

I spent hours troubleshooting this.  My ego likes to think someone will find that the firmware is doing this and I'm not crazy.  But my gambling brain puts it's money on me being crazy.

supermerill, 2017-03-27:
Also, my first outer perimeter is not aligned with the next layer outer perimeter (in the layer preview, printing a calibration cube, same layer width). Potentially the source of my problem.

note: I have an original pruse i3 mk2, i consider that their firmware is correctly set.  
I am in the 0.05 ballpark in the center. My accuracy in measured z is ok (digital caliper, stated for 30um precision), and the squish is ok at 0.15mm layer height.

The brim width may be a bit weird also.

fiveangle, 2017-06-07:
All due respect to @Sebastianv650, the only way to calibrate your way out of the fatter first layer is to print on room temperature blue tape.  This is the Prusa Edition, so if we assume heated bed, this feature does make a lot of sense.  I think @bubnikv will have to judge based on the complexity of implementation vs the benefit.  As he said, there are only 24hrs in a day :)

Input from the communicty is great, but true majority rules voting is the surest way to ensure a race toward mediocrity.  Look at my poor country's president 😱

That said, I vote LIN_ADVANCE over this feature 😉

bubnikv, 2017-06-26:
@fiveangle > Input from the communicty is great, but true majority rules voting is the surest way to ensure a race toward mediocrity. Look at my poor country's president.

I share your frustration. Your president is as bad as mine:
https://www.youtube.com/watch?v=u2jMoOVPW8A

I have received a request from our print farm to implement exactly this feature, so it gets implemented.

Sebastianv650, 2017-06-26:
> I have received a request from our print farm to implement exactly this feature, so it gets implemented.

I think I understand now why there is this request specialy here for Prusa slic3r. While I'm quite excited about my new i3 MK2, one thing I was not able to get up to now is a stable z calirbation. It differs from one print to the next, having a tolerance slightly above 0.05mm. Doesn't sound that much, but at 0.15mm layer height that means the first layer is 33% off..
The latest i3 FW got a feature to do a temperature calibration for the inductive probe. As long as I have small test prints, I don't want to wait 2min before every print start, but does somebody here knows which tolerances can be reached with enabled temp. calibration?

For comparison, on my TAZ 5 with a simple mechanical z endstop I'm able to tune in the first layer height to a tolerance of 0.02mm, stable over moths. This way every print has a perfect 1st layer without scaling or other things, that's where I wont to go with my i3 after the LIN_ADVANCE integration..

bubnikv, 2017-06-26:
A new parameter "elefant_foot_compensation" has been added.
https://github.com/prusa3d/Slic3r/commit/8160db23ccd0cb9ffa1d03599e27a0bcd493e6b7

closed 2017-06-26 by bubnikv

fiveangle, 2017-06-26:
@Sebastianv650 - I've always struggled with first layer on the MK2 as I could never get anything that deviated from 0.2 sliced first layer height to print correctly so went searching and found some very odd discussions about it on the forums.

I took a quick look in the code last year trying to find the details of what [Jozef was talking about here](http://shop.prusa3d.com/forum/original-prusa-i3-mk2-f23/problems-printing-with-petg-after-mk2-upgrade-t1113-s10.html#p8431), but couldn't.  But my understanding at the time was the first layer height is fixed relative to 0.2mm minus the live z adjust calibration, with target of 0.15mm and setting to something else doesn't actually change it (but the [discussion was confusing](http://shop.prusa3d.com/forum/original-prusa-i3-mk2-f23/problems-printing-with-petg-after-mk2-upgrade-t1113.html#p8421), and I never found the relevant code bit that truly explained what they were talking about).

Sebastianv650, 2017-06-26:
I can't tell you what the 0.2mm are that they are refering to. My test prints today with 0.2 and 0.15mm 1st layer height both measured 0.2 and 0.15 so 0.2mm is definitly not a fixed 1st layer setting in the FW. Today, I also don't get the big z calibration differences - fingers crossed! Let's see how it develops.

fiveangle, 2017-06-26:
Good to hear !  That's an old thread so perhaps it was addressed in FW between now and then.  I'm actually running the pre-temp-calibration FW because I have no interested in waiting for 2min before every print, and solved the pinda temp drift issue with my workflow (solved in slicer start gcode by cranking the very powerful blower to high and moving head away from bed while the bed heats to pre-heat temp, which seems enough to keep consistent measurements in my environment).

Guess I'll have to kozy up to the new method when LIN_ADVANCE is integrated :)

Sebastianv650, 2017-06-26:
> Guess I'll have to kozy up to the new method when LIN_ADVANCE is integrated :)

No, the temp. calibration can be switched off in the menu. That's how I'm using it at the moment :)