prusa3d/PrusaSlicer #192: No part exported after using variable layers

ctgreybeard, 2017-03-17:
### Version
1.33.8

### Operating system type + version

10.12.4

### Behavior

When I applied variable layers the preview shows only support material, the model itself has disappeared. And if I export the GCODE and preview it in another app the model is also not there.

I had applied variable layers only at the top and bottom areas of the round part at the top.

#### STL/Config (.ZIP) where problem occurs
[Archive.zip](https://github.com/prusa3d/Slic3r/files/851855/Archive.zip)

ctgreybeard, 2017-03-17:
When I removed the part and re-added it variable layer height worked OK. I'm not sure how useful this report will be then.

bubnikv, 2017-06-06:
Do you have the background processing enabled?
Maybe it is the same issue as https://github.com/prusa3d/Slic3r/issues/293 ?

ctgreybeard, 2017-06-06:
Yes, I do actually. It is perhaps a duplicate then ...

bubnikv, 2017-06-06:
Thanks. Let's close this issue then and leave the other open.

closed 2017-06-06 by bubnikv