prusa3d/PrusaSlicer #193: Crash when deleting all objects

MartinMue, 2017-03-18:
### Version
_Version of Slic3r used goes here_
1.33.8
_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 64 bit Home

### Behavior
Slicer crashes when deleting all objects from platter.

1. add object with the add button
2. add the same object with add button
2. change layer hight via layer editing
3. slice
4. delete all object via button Delete All

_Is this a new feature request?_
NO
#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_
standard config files of this version

[image]
[image]

bubnikv, 2017-03-23:
I cannot reproduce the issue.

Could you reproduce it every time? Is it specific to the STL shown on the screenshot?

MartinMue, 2017-03-23:
It was reproduceable every time the other day, but i did not restart my PC during the testing.

Now i was only able to get it two times out of around 30. Does not matter which STL file.
I'l keep trying.

I'm not sure if this helps, this is how it looks like:

[image]

bubnikv, 2017-03-24:
Thanks, it helps. I can try to iterate it multiple times.
Danke.

bubnikv, 2017-06-06:
I think I have found the culprit. Thanks for the report.

bubnikv, 2017-06-13:
I have added some checks, which should avoid the crash you reported.
https://github.com/prusa3d/Slic3r/commit/fdb5ed1fcbce2b299bab4e9b85a719fa0a949b2a

bubnikv, 2017-06-13:
Please retest with the upcoming 1.35.3. If the issue is still there, please reopen this issue. Thank you.

closed 2017-06-13 by bubnikv