prusa3d/PrusaSlicer #194: Dropdown box in settings tabs don't show enough characters

stephanbrunker, 2017-03-20:
### Version
1.33-8-prusa3d-win64

### Operating system type + version
win10 64-bit

### Behavior
In the Print, Filament and Printer Settings tabs, the dropdown element for the presets only show the first 22 characters of the Preset name. This might be too short to describe the setting properly. Even if it is not possible to increase the width of the whole column (screen resolution issues), the least thing (and Best Practise anyway) would be to show the whole name in a tooltip. 
[image]

Cannot say if this is an issue or a feature request ...

bubnikv, 2017-03-23:
yeah, it is an inconvenience.

bubnikv, 2017-03-23:
One needs to replace wxComboBox with wxComboCtrl user interface widget as the wxComboBox does not allow the change of the text width. Now someone has to study the differences between the two widgets.

stephanbrunker, 2017-03-23:
OK, i know the WinAPI programming only in C/C++ style, so I cannot contribute ...

bubnikv, 2017-07-11:
Implemented by changing layout of the window as suggested in https://github.com/prusa3d/Slic3r/issues/377.

Implemented with 
 https://github.com/prusa3d/Slic3r/commit/a7153c67e6c548a7b923e9a59c717005c69fb551

closed 2017-07-11 by bubnikv