prusa3d/PrusaSlicer #195: Ubuntu large CPU usage and unresponsive GUI during slicing.

hexane360, 2017-03-22:
### Version
version_1.33.8-125-g6b99cbd

### Operating system type + version
Ubuntu 16.04 LTS

### Behavior
I just built Prusa Slic3r, and every time it reslices an object, it takes much longer than stock slic3r, and becomes unresponsive while slicing. I'm not 100% sure how to troubleshoot this. However, htop is showing sustained 100% CPU usage on one of my four CPU cores, simply from sliding the model along the bed a couple mm.
When not slicing, the program is very responsive.

_Is this a new feature request?_
No

#### STL/Config (.ZIP) where problem occurs
Any STL.
Configs can be found [here](https://github.com/hexane360/taz-slic3r-configs).

hexane360, 2017-03-22:
Here's some more information:

It took 2 minutes 20 seconds to slice a 20x20x10mm box (manifold with 12 facets). The GUI was unresponsive during this entire time, and single core CPU was maxed out. The "Info about 20mm-box.stl" message printed on the command line immediately, so I'm confident this time is spent in the slicing process. Partial reslicing doesn't appear to be taking effect, because it takes just as long to reslice with minimal changes.

I'll be trying to run NYTProf to see where computation is hanging up.

bubnikv, 2017-03-23:
I have just found out there is an issue with the new C++ code generating brims. The brim is too detailed, therefore the 3D path will grow excessively.

I have just fixed that through https://github.com/prusa3d/Slic3r/commit/9e0a690d2e9a8349dca42b75c5a14e39ea6cab6f

Does it fix your problem?

By the way, I hope to release an alpha soon, closing the parallelization rework covering most of the calculations.

bubnikv, 2017-03-23:
If you disable the brim, does the issue still persist?

hexane360, 2017-03-23:
Brim was disabled to begin with. Nevertheless, I am in the process of compiling the new version and seeing if it makes any difference.

hexane360, 2017-03-23:
The new version works perfectly, even with a brim. Thanks

closed 2017-03-23 by hexane360