prusa3d/PrusaSlicer #196: Parts of interface layer extends beyond supports and cannot be printed

ceryen, 2017-03-22:
### Version
1.33.8

### Operating system type + version
Windows 10

### Behavior
Slice the provided STL with the provided config.  At layer 4.85, the interface layer extends beyond the supports and cannot be printed.  [Here](http://imgur.com/a/RyXDT) is a screenshot of the interface layer

#### STL/Config (.ZIP) where problem occurs
[stl_and_config.zip](https://github.com/prusa3d/Slic3r/files/860102/stl_and_config.zip)

bubnikv, 2017-03-24:
Pasted the screen shot
[image]

bubnikv, 2017-03-24:
Yes, it is not optimal. You can work around this issue for this particular model by rotating the supports by 90 degrees.

[image]

In general there is no easy fix for this issue. One can either trim the interface layer or one may add another line of the base layer, or one may somehow automatically adjust the support angle, but a single trick will not cover all the possible cases.

bubnikv, 2017-05-20:
A similare case is https://github.com/prusa3d/Slic3r/issues/296

bubnikv, 2021-04-08:
[image]

[image]

bubnikv, 2021-04-08:
Fixed with 8fd731f
We now have a smarter sparse support zig-zag infill generator.

closed 2021-04-08 by bubnikv