prusa3d/PrusaSlicer #197: "Synchronize with object layers" does not appear to work

ceryen, 2017-03-22:
### Version
1.33.8

### Operating system type + version
Windows 10

### Behavior
Slice the provided STL with the provided config.  Note that the provided config has "Synchronize with object layers" turned on (unlike my prior bug #196).  I believe this option should cause support layers to be synchronized with object layers.  However, this does not appear to be the case.  For example, [layer 0.35](http://i.imgur.com/q0lDDBf.png) has only the object, while [layer 0.45](http://i.imgur.com/nZIarkD.png) has only the support.

#### STL/Config (.ZIP) where problem occurs
[stl_and_config.zip](https://github.com/prusa3d/Slic3r/files/860108/stl_and_config.zip)

ceryen, 2017-03-22:
It's entirely possible that I'm misinterpreting the intent of "Synchronize with object layers".  If so, please let me know!  If that is the case, I think we might need a more descriptive tool tip.

bubnikv, 2017-03-24:
This feature has not been well tested and finalized yet, it is experimental. It's main purpose is to limit the number of filament swaps on multi-material prints.

bubnikv, 2017-04-10:
Fixed with https://github.com/prusa3d/Slic3r/commit/23be6233c80353faa2871b72838f7934b94ce8db

Finalized the experimental "synchronize support layers with object layers"
feature (the support_material_synchronize_layers settings). 
The feature is now enabled for the soluble supports only
(with support_material_contact_distance > 0).

closed 2017-04-10 by bubnikv