prusa3d/PrusaSlicer #198: Contact Z distance too high for support?

ceryen, 2017-03-22:
### Version
1.33.8

### Operating system type + version
Windows 10

### Behavior
I have my Contact Z distance set to 0.1mm.  In my model, there is an overhanging piece that starts 4.8mm above the bed.  Given the Contact Z distance of 0.1mm, I would expect the supports for the overhanging piece to stop at 4.7mm.  However, the supports actually stops at 4.4mm instead, yielding an actual Contact Z distance of 0.4mm

Here is the layer containing the start of the overhanging piece.  The bottom of the layer starts at 4.8mm (layer 4.9mm minus the layer height of 0.1mm):

[image]

Here is where the support ends at layer 4.4mm:
[image]

#### STL/Config (.ZIP) where problem occurs
[stl_and_config.zip](https://github.com/prusa3d/Slic3r/files/860316/stl_and_config.zip)

bubnikv, 2017-03-24:
The current behavior is correct. The bottom most surface of the overhang is printed with a bridge flow which means it is 0.4mm thick.

[image]

closed 2017-03-24 by bubnikv

joeynuggetz, 2017-03-27:
I've noticed similar problems. Although the overhang is printed at 0.4mm thick it doesn't stick and causes errors in my projects.

bubnikv, 2017-03-27:
@joeynuggetz 
> I've noticed similar problems. Although the overhang is printed at 0.4mm thick it doesn't stick and causes errors in my projects. Problematic with circles.

What kind of problems? Does the object curl up?
What material? ABS or PLA?

joeynuggetz, 2017-03-27:
While drawing the circle it doesn't stick so it comes out looking more like a square.

joeynuggetz, 2017-03-27:
[pulley_1_fixed.zip](https://github.com/prusa3d/Slic3r/files/872879/pulley_1_fixed.zip)

joeynuggetz, 2017-03-27:
Also, at .10 layer height the problem is more severe. Last support layer is at a height of .30 and the large circle is drawn at height of .80 and that fails all the time.

bubnikv, 2017-03-27:
There is really not a good solution to this. You may decrease the support gap down to nearly zero, but this has been found not enough for ABS.

I think your issue is a duplicate of https://github.com/prusa3d/Slic3r/issues/102