prusa3d/PrusaSlicer #199: Exported gcode uses M104 before start gcode, should use M109

hoppersoft, 2017-03-24:
### Version
1.33.8

### Operating system type + version
Windows 10

### Behavior
When Slic3r generates the gcode, it sets the temperature using M104, then calls any custom start gcode.

Unfortunately, this only *starts* getting the extruder head to the desired temperature, so any homing/leveling happens prior to the head reaching that temperature. This can lead to bed adhesion issues depending on how big a difference there is between the distance when cool vs. when at-temperature. Waiting for the extruder to reach the temperature would be preferable, as it ensures that any custom gcode instructions execute under the same conditions as the instructions used for the print.

To reproduce, add a custom start gcode including "G29," then export the gcode. Inspect the gcode, and you will see that the temperature is set using M104.

Note that an M109 is included, but only *after* the start gcode has completed.

lordofhyphens, 2017-03-29:
There's a PR and issue open upstream about this. 
https://github.com/alexrj/Slic3r/issues/3268
https://github.com/alexrj/Slic3r/pull/3411

If you put M109 or M104 in your start gcode, it will inhibit the addition of the automatic temp gcode.
https://github.com/alexrj/Slic3r/wiki/Parameter-names-for-Custom-GCode

The names are the same as in the .ini file.

bubnikv, 2017-05-20:
Implemented with https://github.com/prusa3d/Slic3r/commit/a99b006b98a302436e7d965f052a7a9ab5034773

thans @lordofhyphens

closed 2017-05-20 by bubnikv